Koenkk/zigbee2mqtt #8036: Support for S-LUX-ZB

Koterak, 2021-07-13:
I tried several times to add device support for S-LUX-ZB. 

Link to the device where I bought it: https://aliexpress.com/item/1005002835450510.html

Received message from unsupported device with Zigbee model 'isltm67' and manufacturer name '_TYST11_pisltm67'
Warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

I tired to add this device with such code for tuya devices:

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            modelID: 'isltm67\u0000',
            manufacturerName: '_TYST11_pisltm67'
        },
    ],
    model: 'S-LUX-ZB',
    vendor: 'INNOPRO',
    description: 'Light Sensor',
    fromZigbee: [
        fz.battery, 
        fz.illuminance
    ],
    toZigbee: [
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        e.battery(), 
        e.illuminance(), 
        e.illuminance_lux()
    ],
};

module.exports = definition;`

And it's a bit successfull.. Because the device got recognised. But all data which I got is N/A unfortunatly. 
Screen from zigbee2mqtt:
[image]
[image]
[image]

In addition screen from logs:
[image]

Is there anything else I can do? Or can you maybe add support for this device?

Koterak, 2021-07-14:
I managed to get some data from device. Maybe this will help?
[image]

Koterak, 2021-07-14:
I added something like this to fromZigbee:

```
s_lux_zb_illuminance: {
    cluster: 'manuSpecificTuya',
    type: ['commandGetData', 'commandSetDataResponse'],
    convert: (model, msg, publish, options, meta) => {
        const dp = msg.data.dp; // First we get the data point ID
        const value = tuyaGetDataValue(msg.data.datatype, msg.data.data); // This function will take of converting the data to proper JS type

        switch (dp) {
        case tuya.dataPoints.sluxIlluminance: // DPID that we added to common
            return {illuminance_log: value.toFixed(1)}; // value is already converted to a number in JS, and we deduced that it needs to be divided by 10
        default:
            meta.logger.warn(`s_lux_zb_illuminance: NOT RECOGNIZED DP #${
                dp} with data ${JSON.stringify(msg.data)}`); // This will cause zigbee2mqtt to print similar data to what is dumped in tuya.dump.txt
    },
},
```

And in tuya.js I added 
`sluxIlluminance :4`

But I got error:
[image]

Koenkk, 2021-07-15:
Try with this external converter, this will make it completely standalone so no changes to zigbee-herdsman-converters are required:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const illuminanceConverter = {
    cluster: 'manuSpecificTuya',
    type: ['commandGetData', 'commandSetDataResponse'],
    convert: (model, msg, publish, options, meta) => {
        const dp = msg.data.dp; // First we get the data point ID
        const value = tuyaGetDataValue(msg.data.datatype, msg.data.data); // This function will take of converting the data to proper JS type

        console.log(`Received: ${dp} - ${value}`);
        switch (dp) {
        case 4: // DPID that we added to common
            return {illuminance_log: value.toFixed(1)}; // value is already converted to a number in JS, and we deduced that it needs to be divided by 10
        default:
            meta.logger.warn(`s_lux_zb_illuminance: NOT RECOGNIZED DP #${
                dp} with data ${JSON.stringify(msg.data)}`); // This will cause zigbee2mqtt to print similar data to what is dumped in tuya.dump.txt
        }
    },
};

const definition = {
    fingerprint: [
        {
            modelID: 'isltm67\u0000',
            manufacturerName: '_TYST11_pisltm67'
        },
    ],
    model: 'S-LUX-ZB',
    vendor: 'INNOPRO',
    description: 'Light Sensor',
    fromZigbee: [illuminanceConverter],
    toZigbee: [
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        e.battery(), 
        e.illuminance(), 
        e.illuminance_lux()
    ],
};

module.exports = definition;
```

Koterak, 2021-07-15:
Now I got this error:
[image]

Koenkk, 2021-07-15:
Try changing `const value = tuyaGetDataValue(msg.data.datatype, msg.data.data);` to `const value = tuya.getDataValue(msg.data.datatype, msg.data.data);`

Koterak, 2021-07-15:
Thank you very much! It's working now. And now what I will have to do to add it as homeassistant entity? How to make this data exposed?

closed 2021-07-15 by Koterak

Koenkk, 2021-07-16:
Can you provide your converter so I can add this for support out of the box?

Koterak, 2021-07-16:
I just edited it a little from code which you give me. 

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const illuminanceConverter = {
    cluster: 'manuSpecificTuya',
    type: ['commandGetData', 'commandSetDataResponse'],
    convert: (model, msg, publish, options, meta) => {
        const dp = msg.data.dp; // First we get the data point ID
        const value = tuya.getDataValue(msg.data.datatype, msg.data.data); // This function will take of converting the data to proper JS type

        console.log(`Received: ${dp} - ${value}`);
        switch (dp) {
        case 2: // DPID that we added to common
            return {illuminance_lux: value.toFixed(0)}; // value is already converted to a number in JS, and we deduced that it needs to be divided by 10
        case 4: // DPID that we added to common
            return {battery: value.toFixed(1)};
        case 1:
            return {battery_low: value.toFixed(1)};                    
        default:
            meta.logger.warn(`s_lux_zb_illuminance: NOT RECOGNIZED DP #${
                dp} with data ${JSON.stringify(msg.data)}`); // This will cause zigbee2mqtt to print similar data to what is dumped in tuya.dump.txt
        }
    },
};

const definition = {
    fingerprint: [
        {
            modelID: 'isltm67\u0000',
            manufacturerName: '_TYST11_pisltm67'
        },
    ],
    model: 'S-LUX-ZB',
    vendor: 'INNOPRO',
    description: 'Light Sensor',
    fromZigbee: [illuminanceConverter],
    toZigbee: [
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        e.battery(), 
        e.illuminance_lux(),
        e.battery_low(),
    ],
};

module.exports = definition;
```

Today I checked all the values and it's working for now.

Koenkk, 2021-07-17:
Thanks, added

hossface, 2021-11-30:
I have found another variation of this device with a different manfucturer name.
I just tweaked the above code and added as an external converter to get it working.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const illuminanceConverter = {
    cluster: 'manuSpecificTuya',
    type: ['commandGetData', 'commandSetDataResponse'],
    convert: (model, msg, publish, options, meta) => {
        const dp = msg.data.dp; // First we get the data point ID
        const value = tuya.getDataValue(msg.data.datatype, msg.data.data); // This function will take of converting the data to proper JS type

        console.log(`Received: ${dp} - ${value}`);
        switch (dp) {
        case 2: // DPID that we added to common
            return {illuminance_lux: value.toFixed(0)}; // value is already converted to a number in JS, and we deduced that it needs to be divided by 10
        case 4: // DPID that we added to common
            return {battery: value.toFixed(1)};
        case 1:
            return {battery_low: value.toFixed(1)};                    
        default:
            meta.logger.warn(`s_lux_zb_illuminance: NOT RECOGNIZED DP #${
                dp} with data ${JSON.stringify(msg.data)}`); // This will cause zigbee2mqtt to print similar data to what is dumped in tuya.dump.txt
        }
    },
};

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_pisltm67'
        },
    ],
    model: 'S-LUX-ZB',
    vendor: 'INNOPRO',
    description: 'Light Sensor',
    fromZigbee: [illuminanceConverter],
    toZigbee: [
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        e.battery(), 
        e.illuminance_lux(),
        e.battery_low(),
    ],
};

module.exports = definition;

hossface, 2021-11-30:
I have found a variation of this device with TZE200 instead of TYST11
I have tweaked the code and added as an external converter it worked for me.

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const illuminanceConverter = {
    cluster: 'manuSpecificTuya',
    type: ['commandGetData', 'commandSetDataResponse'],
    convert: (model, msg, publish, options, meta) => {
        const dp = msg.data.dp; // First we get the data point ID
        const value = tuya.getDataValue(msg.data.datatype, msg.data.data); // This function will take of converting the data to proper JS type

        console.log(`Received: ${dp} - ${value}`);
        switch (dp) {
        case 2: // DPID that we added to common
            return {illuminance_lux: value.toFixed(0)}; // value is already converted to a number in JS, and we deduced that it needs to be divided by 10
        case 4: // DPID that we added to common
            return {battery: value.toFixed(1)};
        case 1:
            return {battery_low: value.toFixed(1)};                    
        default:
            meta.logger.warn(`s_lux_zb_illuminance: NOT RECOGNIZED DP #${
                dp} with data ${JSON.stringify(msg.data)}`); // This will cause zigbee2mqtt to print similar data to what is dumped in tuya.dump.txt
        }
    },
};

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_pisltm67'
        },
    ],
    model: 'S-LUX-ZB',
    vendor: 'INNOPRO',
    description: 'Light Sensor',
    fromZigbee: [illuminanceConverter],
    toZigbee: [
    ],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        e.battery(), 
        e.illuminance_lux(),
        e.battery_low(),
    ],
};

module.exports = definition;
```