Koenkk/zigbee2mqtt #8048: GreenPower push button coupler pairing successful but no messages coming through

wb441, 2021-07-15:
## What happened

After seemingly succesfully pairing a GreenPower device, no messages are seen by Zigbee2mqtt.
Device: https://www.led-trading.de/zigbee-unterputz-modul-fuer-friends-of-hue
Basically, it is a 230V powered Friends of Hue push button coupler.

## What did you expect to happen

Similar to other Greenpower devices I have, that pushing a button results in 'recall_scene_#' actions which I can then couple through NodeRED.

## How to reproduce it (minimal and precise)

Power the device
Push B0 button, then A1 and B0 together in order to pair on Channel 11 and confirm/
This gives the following output to the log: https://hastebin.com/sijesixefe.apache 
Adding Zigbee2mqtt IN node in Nodered in order to couple it to other stuff
Pushing a button
No message/action whatsoever is seen in the log.

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version:

wb441, 2021-07-15:
Solved it.
Turns out that during the second step of pairing (Step 1 = press button for channel 7s, Step 2 = press A1 and B0 to confirm) I was pushing the wrong button. 
Turns out, if you don't follow the instructions correctly, it doesn't work.. what do you know :-)
Working brilliantly now!

closed 2021-07-15 by wb441

holstweb, 2021-10-21:
I'm running into the same problem. The device has K1, K2, K3, K4. Which are B0 and A1?

lozwei, 2022-10-24:
I know it's an older Issue but maybe I can ask you a question anyway: is your device recognized as the right device? When I pair my modules they are all detected as [PTM 215Z](https://www.zigbee2mqtt.io/devices/PTM_215Z.html#enocean-ptm%2520215z).

minke890, 2022-11-13:
> I know it's an older Issue but maybe I can ask you a question anyway: is your device recognized as the right device? When I pair my modules they are all detected as [PTM 215Z](https://www.zigbee2mqtt.io/devices/PTM_215Z.html#enocean-ptm%2520215z). Can you please tell me the ieeeAddress of your devices? It should be starting with 0x00000000427. It seems mine have the wrong but I don't know why :-(

How were you able to pair it?  I used to run this module in deconz on chanel 15, but are unable to pair it in z2m on chanel 11. Chanel 11 means  button K1 for 10sec and confirm by K1+K4 for 1sec, but it does not work for me.

wb441, 2022-11-26:
Sorry for the late reply.
I'm working on channel 11 as well, and per the instructions on the z2m website and also above, K2 first, then K1 + K4 to confirm. Can you try it this way and report back?