Koenkk/zigbee2mqtt #8049: "NO COMPONENTS" for all devices. Fresh install (docker), Conbee II Stick

markus-barta, 2021-07-15:
## What happened
Zigbee2MqttAssistant shows "NO COMPONENTS" for all devices, although they are supported based on zigbee-herdsman repo.

See: [https://pasteboard.co/KbfOpIk.png](https://pasteboard.co/KbfOpIk.png)

## What did you expect to happen
Devices should be properly detected.

## How to reproduce it (minimal and precise)
- Install docker conatiners.
- Plug in Conbee II. 
- Pair a Xiaomi Aquara temp sensor.

## Debug info
Zigbee2MQTT version: koenkk/zigbee2mqtt:latest
Adapter hardware: Conbee II
Adapter firmware version: 1.20.0

markus-barta, 2021-07-15:
Not 100% sure where to post this - Zigbee2MqttAssistant or here? Please close if inappropriate. Thank you!

Koenkk, 2021-07-15:
This looks to be an issue of zigbee2mqtt assistant. Zigbee2mqttassistant does not use the new api yet. I think it is better to migrate to the builtin frontend: https://www.zigbee2mqtt.io/information/frontend.html

closed 2021-07-15 by Koenkk

markus-barta, 2021-07-15:
alright. thanks a lot. i really missed the frontend because pulling images from docker did not give me an empty config, so I started from scratch missing the built in ui. 🤦