Koenkk/zigbee2mqtt #8057: Mercator Ikuü  Single Gane Light Dimmer (Zigbee) SSWD01

Asleep-at-the-wheel, 2021-07-16:
Hello amazing community,

I am running Home Assistant OS 6.1 using the Zigbee2mqtt addon 1.18.1-1 which I believe runs Zigbee2mqtt 1.18.1.

I have acquired a new device that has been installed and has successfully paired with my Zigbee network via the Hassio Zigbee2Mqtt addon. The device shows in the device list as unsupported and presents the following message frequently:
```
warning Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_swaamsoy'
warning Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
```
I spent a considerable amount of time yesterday and the day before trying to follow the guide in the message but found that every external converter I wrote simply stopped the addon from starting. I believe the device is a standard rebranded Tuya device like many others and would probably work just fine with same external converter as any other Tuya Zigbee no neutral dimmer switch. 

Unfortunately I tried to follow the guide for it [here ](https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html )but found that using this script I kept getting errors around `const extend = require('zigbee-herdsman-converters/lib/extend');:` because the lib/extend couldn't be found. To verify I tried the following:
`➜  ~ docker ps | grep zigbee
4d08322cf5fd   zigbee2mqtt/zigbee2mqtt-amd64:1.18.1-1                         "/init"                  15 hours ago   Up 15 hours   0.0.0.0:8485->8485/tcp, :::8485->8485/tcp                                                                                      addon_7ad98f9c_zigbee2mqtt
➜  ~ docker exec -it 4d08322cf5fd bash
bash-5.1# ls /app/node_modules/zigbee-herdsman-converters/lib/
constants.js   exposes.js     ikea.js        kelvinToXy.js  legacy.js      legrand.js     light.js       livolo.js      ota            reporting.js   store.js       tuya.js        utils.js       xiaomi.js
`
Which confirmed the error.

After all this, running out of time and simply not having the required functioning neurons to make this work, I finally gave up and came here to try and request some smart person add support for this device(s) and maybe the others in the line to Zigbee2mqtt. Is this possible given all the information I provided here?

## Information about the device + link
**The device range:**
https://www.ikuu.com.au/
**The specific device I am working with:**
https://www.ikuu.com.au/product/light-dimmer/
https://www.ikuu.com.au/downloadable-assets/product-specifications/MIDS000026Z%20Data%20Sheet.pdf

## data/database.db entry of the device
`
{"id":39,"type":"Router","ieeeAddr":"0xbc33acfffef46734","nwkAddr":55194,"manufId":4098,"manufName":"_TZE200_swaamsoy","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,3,4,5,6,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_swaamsoy","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1626404508847}`

Lastly, if this device can't be added by someone here. Could someone point me in the right direction to making this work? I know the official documentation exists, but I simply can't make it work in the hassio addon and lack any real development environment to make it all work in a fresh install.

Koenkk, 2021-07-16:
This device is already supported in zigbee2mqtt 1.20.0.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2021-07-16 by Koenkk

reidyrobot, 2021-07-27:
Sorry to comment on a closed issue but I have a very similar device.. Mercator dimmer but inline:
https://www.ikuu.com.au/product/inline-switch-with-dimmer/

 I'm running the latest dev docker image with a conbee II stick.

My system is seeing the tuya device but not recognising it:

Received message from unsupported device with Zigbee model 'TS0601' and manufacturer name '_TZE200_ojzhk75b'

Should I create a separate issue?

Amazing work Koenkk!!