Koenkk/zigbee2mqtt #8061: Schneider Electric / Clipsal / Wiser UNSUPPORT_ATTRIBUTE error

TailsK, 2021-07-17:
Hello!

## What happened
I have been setting up my Schneider Electric 41EPBDWCLMZ/354PBDMBTZ light switch.  While the device has been added to the Zigbee2MQTT dashboard, any command I try to run sees an error.  For example, toggling the light switch produces the following error:

`Publish 'get' 'state' to '0x90fd9ffffed04b57' failed: 'Error: Read 0x90fd9ffffed04b57/21 genOnOff(["onOff"], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

This is true for any command that I try to run (i.e. changing the min/max ballast levels).

`Error Publish 'set' 'ballast_minimum_level' to '0x90fd9ffffed04b57' failed: 'Error: Write 0x90fd9ffffed04b57/21 lightingBallastCfg({"minLevel":79}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

## What did you expect to happen
I expected the light to switch on or off depending on its current state.

## How to reproduce it (minimal and precise)
Under "Exposes" toggle the state between on/off.

## Debug info
Zigbee2MQTT version: 1.20.0-1
Adapter hardware: CC2531
Adapter firmware version:

closed 2021-08-25 by github-actions[bot]

YodaDaCoda, 2022-04-23:
Can this be reopened? I have the same issue with the latest software & firmware. Other devices on my network are working as expected.

Zigbee2MQTT version: 1.25.0
Adapter hardware: CC2652R (zzh!)
Adapter firmware version: 20220219

rbswift, 2022-08-06:
I had this problem initially after joining my dimmer but after restarting z2m it worked

DanielNagy, 2023-03-04:
Looks now to be fixed in issue https://github.com/Koenkk/zigbee2mqtt/issues/16822#issuecomment-1454771102