Koenkk/zigbee2mqtt #8066: Zemismart AM15 roller blinds

teunaldo, 2021-07-18:
Hi, 

I try to add the tuya Zemismart AM15 roller blinds. (https://www.zemismart.com/products/zemismart-tuya-zigbee-motor-for-25mm-tube-built-in-battery-smart-life-alexa-google-home-control)

With this solution a dongle is also in place which converts the Zigbee signal to the signal between the dongle and the roller blinds. So: CC2531 ----zigbee3.0---- Zemismart AM20 dongle ---- 433.89MHZ ---- Zemismart AM15
When I added the dongle into zigbee2mqtt it gave the error that is wasn't recognizable.

With the guide (https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html) I made it happen that I could add this device.
Now I only have the lqi as state. I don't know which information is coming out of the device. 

The problem now is.. To pair the dongle with the AM15 I need to press the "up" button in the tuya app after putting the AM15 into pairing mode.  
I also don't see the tuya.dump.txt in the zigbee2mqtt folder.
I looked at other devices and added "coverInverted" under support. This didn't help. 
Also did I have added "state" under exposes, but that also wouldn't help.
If I look in the device under clusters I see the following information:
https://gyazo.com/d37ecb915a5ebda1fe77111996faffbf

I tried to sent some messages with NodeRed to the device but it didn't react to it.
I tried to sent '{"state":"OPEN"}' to 'zigbee2mqtt/0x847127fffe0e1439/bridge/state/set'
I get this reaction in zigbee2mqtt: '2021-07-18 21:26:18: Received MQTT message on 'zigbee2mqtt/0x847127fffe0e1439/bridge/state/set' with data '{"state":"OPEN"}'
But the AM20 doesn't react to it.

The code of my _TZE200_iossyxra.js file is the following:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    // Since a lot of Tuya devices use the same modelID, but use different data points
    // it's usually necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_iossyxra'
        },
    ],
    model: 'AM20',
    vendor: 'Zemismart',
    description: 'Tube motor roller blind',
    supports: 'coverInverted',
    fromZigbee: [
        //fz.ignore_basic_report, // Add this if you are getting no converter for 'genBasic'
        fz.tuya_data_point_dump, // This is a debug converter, it will be described in the next part
    ],
    toZigbee: [
        tz.tuya_data_point_test, // Another debug converter
		tz.state
    ],
    // onEvent: tuya.setTime, // Add this if you are getting no converter for 'commandSetTimeRequest'
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genBasic']);
    },
    exposes: [
        // Here you should put all functionality that your device exposes
    ],
};
module.exports = definition;
`
I would love to add the database.db entry, but I only find 1 entry in my database.db file
If I need to add it, feel free to tell me how

botany112, 2021-08-19:
Hi 
i got the same, but i dont understand how to add. could you help?

i use zigbee2mqtt addon.
i add zigbee_shepherd_devices: true to my configration.
and put the .js in the share/zigbee2mqtt folder. 
when i restart the addon en re-add the device still get Unsupported.

closed 2021-09-27 by github-actions[bot]

snozzley, 2021-09-27:
Hey - did you get anywhere with this issue? I've got the same dongle, trying to get it going with zigbee2mqtt, and seeing same result of just lqi shown.
I have a different motor, which I believe I'm able to pair, but can't get any more info through zigbee2mqtt.  I probably don't understand it well enough though..

teunaldo, 2021-10-03:
Hi,l nope. I didn't had an reply.
I fixed it by using the Wifi-dongle and Localtuya.
Not the way I preffered, but the best option for now

Ellipsiss, 2022-05-11:
Just in case this helps anyone… I have the Zemismart AM24-ZB with [Zemismart Zigbee Dongle For AM15 Smart App Control Mini Design](https://www.zemismart.com/products/zmam02) as the RF Zigbee bridge.

I did the following

Remote

    Paired remote using the instructions in the packet
        Press and hold ‘Up’ on remote
        Simultaneously connect battery
        Blind should move slightly
        Release then press ‘UP’ again
        Blind moves - remote paired
        Set limits as per packet instructions

Zigbee

    Plug in Zigbee USB (Mine was ZM_AM02) to somewhere with power
    Press the pairing button 5-8 times in rapid succession
    Should start blinking
    Add it through ZHA how you normally would
    Once paired - press the Midde / Stop button on RF remote you paired prior. Hold for about 5 secs til blind moves up and down slightly like earlier
    Press ‘UP’ on the recently configured dongle in ZHA
    Blind moves again to confirm

Caveat - You need 1 dongle per blind. You cannot pair two blinds to one dongle. Which sucks. But whatever I guess.

Hope this was the right place and hope this helps someone. Took me ages / lots of deciphering to work out ZHA pairing.