Koenkk/zigbee2mqtt #8073: SNZB-03 Sonoff motion sensor becomes unresponsive and stuck in occupied state

kobuki, 2021-07-19:
I've been using an SNZB-03 for about 3 months. I'm only testing its reliability for now. It works for a time, then it's stuck in the occupied state for days until I discover it. It seems random but 2 things can be observed every time: 1) the LQI stays at 0 (best reception) while it's unresponsive; 2) if I move it away to another room and trigger it at least once, it starts working again for a few days or a week. When I move it, LQI jumps to around 30 and when I place it back to its usual place, it goes back to 0..2.

Can it be that z2m starts to ignore devices that have a 0 LQI for an extended period? It doesn't show an error in the web dashboard for the device when the error happens, so this is probably not the case.

I'm using z2m inside HA as service, but the sensor state is stuck in the z2m web interface, too (status JSON doesn't change), not just in HA. So I think the problem is related to z2m itself.

Zigbee2MQTT version: 1.19.1-1
Adapter hardware: CC2531 (using a long 17 dBi whip antenna)
Adapter firmware version: zStack12 0.4.29

PS.: I see now that I didn't run the latest z2m up till now. Now I updated it to 1.20.0. I'm not sure this is relevant, as I saw the same problem appearing for multiple z2m versions with the same device.

nashant, 2021-07-24:
Yeah, I've got this too. What coordinator are you using?

kobuki, 2021-07-24:
> Yeah, I've got this too. What coordinator are you using?

It's in my post.

kobuki, 2021-07-24:
A small update: z2m 1.20.0 didn't fix the issue. But I'm inclined to think it's not an issue with z2m but the device itself, as the device is not even discovered after a blackout-induced cold reboot. But it's too early to say.

All the while the Sonoff T/H sensor works perfectly 7x24, not a single loss of signal or message.

dreameradg, 2021-07-26:
Yes, I've faced with the same issue.

ec-blaster, 2021-08-31:
I have the same problem with the same device and latest version of Z2M

kobuki, 2021-09-08:
>  @github-actions github-actions bot closed this 7 hours ago 

The issue still persists, though.

Koenkk, 2021-09-08:
to further investigate this, provide a sniff starting right before you trigger the motion sensor and stop 10 seconds after this.
https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

kobuki, 2021-09-08:
ALright, will do that today evening.

kobuki, 2021-09-08:
Ah well. It requires me to have another stick and flash the sniffer firmware. I have no other stick ATM, so it needs to wait.

closed 2021-10-17 by github-actions[bot]

xrust83, 2022-02-03:
it really is. Ran into a problem in a remote room from the stick kitchen. Sonoff is weak and does not always turn on the light, aqara has a stable signal. No more desire to save on sonoff zigbee