Koenkk/zigbee2mqtt #8075: Livolo TI0001 new switch 2 gangs and TI0001 new reset switch 2 gangs don't keep pair

systemiii, 2021-07-19:
I have one Livolo TI0001 new switch 2 gangs and one Livolo TI0001 new reset switch 2 gangs.
I use a Sonoff ZBbridge with Tasmota last release as coordinator.
I can pair both switches successfully, but they lose pairing status after power cycle, so i should pair them every time.
I have followed all instructions in the website.

Koenkk, 2021-07-19:
This is the Zigbee2MQTT issue tracker, not the tasmota one, please create an issue @ the tasmota issue tracker instead.

systemiii, 2021-07-19:
Sorry, I didn't specify that I'm using Z2M Home Assistant add-on with EZSP adapter corresponding to Sonoff ZBbridge.
I thought that Tasmota was only a pass-through bridge. You confirm should I open the issue to Tasmota?

Koenkk, 2021-07-20:
Then this is the good place indeed.

systemiii, 2021-09-12:
I have found the way to fully support this two devices.
I'll open a pull request when I'm done testing them.

whitepail, 2022-01-31:
@systemiii Were you able to resolve the issue.
I have new devices, 1 gang and 4 gang and plan to implement support for them.

Please share your progress to speed up implementation

systemiii, 2022-01-31:
Yes sure, but i advise that Livolo zigbee devices has a lot of bugs because of zigbee protocol huge customization made by Livolo.

systemiii, 2022-01-31:
This Is the link to my external converter for Home Assistant addon
https://drive.google.com/file/d/11wOw2R5AuRxVmTLASznkun8G7_Gpthv4/view?usp=drivesdk

systemiii, 2022-02-03:
@whitepail my external converter was useful to help your implementation?

whitepail, 2022-02-04:
Sure, thanks a lot.
WiP, assume to finish and test it next week then will send a pull request
to Koenkk.

systemiii, 2022-02-04:
Great news. I would never be able to make a pull request due to daily family commitments. Also for me it will be much better not to have to use an external converter anymore. Thank you

systemiii, 2022-02-07:
@whitepail do you know the way to override the unpair function in the external converter? I am investigating this because the base disconnection does not currently work with any Livolo devices. thank you

whitepail, 2022-02-07:
Hi.

  I'm not sure if pairing/unpairing can be overridden.
  But if you need to do something on unpairing - there is a block of code
in converters:
        onEvent: async (type, data, device) => {
            if (type === 'stop') {
                clearInterval(globalStore.getValue(device, 'interval'));
                globalStore.clearValue(device, 'interval');
            }

As I remember it will be triggered whenever you unpair your device.

systemiii, 2022-02-08:
Thank you @whitepail, this is just what I need. I'll try it right away.

systemiii, 2022-02-10:
I have discovered the Livolo device unpair command but I cannot use it because it is necessary to use the "Write Attributes Response" (0x04) command of the ZCL frame, but only "Write" and "Write Attribute Undivided" are implemented in zigbee-herdsman.
Can you tell me how I can make the implementation request?
Should I open an issue?

razim7, 2022-02-21:
> This Is the link to my external converter for Home Assistant addon https://drive.google.com/file/d/11wOw2R5AuRxVmTLASznkun8G7_Gpthv4/view?usp=drivesdk

@systemiii  how to add it? I have 3 gang and 4 gang

systemiii, 2022-02-21:
@razim7 you can follow these instructions to add my external converter
https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device

systemiii, 2022-02-21:
If you can get my external converter working in your local environment, I will send you the update for the 4 button device, so you can test it... because I don't have any 4 button device.

I have started doing local tests to put the converter content in a pull request, but it will take some time

razim7, 2022-02-21:
@systemiii works perfectly, had to delete the devices from the database and to re-add them, otherwise they will appear as a default T100

the 3 gang working

the curtain working but in odd way, some of them open instead of close and etc... but I'll try to fix it.

the 4 gang worked with the 3 gang when I edit the database, but only 3 buttons worked, the 4th didn't

razim7, 2022-02-21:
@systemiii the 4 gang:
{
  "linkquality": 65,
  "state_left": null,
  "state_left_left": "ON",
  "state_right": null,
  "state_right_right": "ON",
  "state_top": null,
  "state_top_top": "OFF"
  "state_top_top": "ON"
}

razim7, 2022-02-21:
@systemiii you made my day!!! I was about to get rid off all the livolo and try to find something else!
now let's hope they will be officially supported

razim7, 2022-02-22:
the only issue I found out, after couple hourse the devices became N/A the only way so far to resolve it to keep "permit_join: true" otherwise they become N/A.

I'm using: usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus with Zigbee2Mqtt edge HA addon

systemiii, 2022-02-22:
@razim7 Yes, Livolo devices need to keep permit join always on.
In order to add 4 gangs switch, i need debug log of entire interview (pairing) process.

whitepail, 2022-02-22:
Hi Samuel.
I've tried to build change for 4 gang switch, but was unable to find any
message with magic header (124,210,21,216,128) to identify the switch.
attaching pairing log for new 1 gang and new 4 gang.

4 gang is 0x00124b00233fec5c
1 gang is 0x00124b00241422c5

razim7, 2022-02-23:
@systemiii here you go:

Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,13,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,15,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo3Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
[... cut ...] "state","property":"state_top","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TI0001-switch-3gang","options":[],"supports_ota":false,"vendor":"Livolo"},"friendly_name":"Livolo3Gang","ieee_address":"Livolo3Gang","status":"successful","supported":true},"type":"device_interview"}'

razim7, 2022-02-23:
@systemiii here you go:

Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,13,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,15,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,1,0,9,0,1,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from 'Livolo4Gang', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,3,100,66,7,21,11],"type":"Buffer"}' from endpoint 6 with groupID 0
[... cut ...] "state","property":"state_top","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TI0001-switch-3gang","options":[],"supports_ota":false,"vendor":"Livolo"},"friendly_name":"Livolo4Gang","ieee_address":"Livolo4Gang","status":"successful","supported":true},"type":"device_interview"}'

whitepail, 2022-02-23:
Ok, so got new switches from livolo.
Changes in firmware:
1) Switches do not send genPowercfg 124,210,21,216,128 magic after interview any more. So we have a problem how to identify device after interview and change it's model (not by hands in db).
2) Switches can have 1 to 4 gangs. State is  sent with genPowercfg with magic 122,209, switch id and msg.data[10] = 13. On/off state is in msg.data[13]; 
state_left: status & 1 ? 'ON' : 'OFF',
state_right: status & 2 ? 'ON' : 'OFF',
state_bottom-left: status & 4 ? 'ON' : 'OFF',
state_bottom-right: status & 8 ? 'ON' : 'OFF',
3) Writing state to switch is still the same, via entity.write('genPowerCfg')
const payloadOnLeft = {0x0001: {value: Buffer.from([1, 0, 0, 0, 0, 0, 0, 0]), type: 1}};
const payloadOffLeft = {0x0001: {value: Buffer.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 1}};
const payloadOnRight = {0x0001: {value: Buffer.from([2, 0, 0, 0, 0, 0, 0, 0]), type: 2}};
const payloadOffRight = {0x0001: {value: Buffer.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 2}};
const payloadOnBottomLeft = {0x0001: {value: Buffer.from([4, 0, 0, 0, 0, 0, 0, 0]), type: 4}};
const payloadOffBottomLeft = {0x0001: {value: Buffer.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 4}};
const payloadOnBottomRight = {0x0001: {value: Buffer.from([8, 0, 0, 0, 0, 0, 0, 0]), type: 8}};
const payloadOffBottomRight = {0x0001: {value: Buffer.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 8}};
4) To prevent device from unpairing, Z2M need to react to genPowercfg message with magic 122,209 and msg.data[10] = 13, sending readResponse('genPowerCfg') message (as earlier with dimmer and other 'new' devices)

Currently  I'm working on the universal 1gang to 4 gang implementation that will support all new devices by default.

systemiii, 2022-02-23:
@whitepail I don't have any 4 gang switches and bought the latest Livolo devices last December.
Are yours newer? Do you mean that Livolo has recently changed firmware?

razim7, 2022-02-23:
@systemiii mine should be the recent ones, I have several from early EA, and several from last few months, all the same, and all worked perfect with your livolo2.js, the only 2 issues: 4 gang - sent you the debug if you need more info let me know.
and the curtain - the cover not working well, besides that all is fine

Info Successfully interviewed 'Livolo4Gang', device has successfully been paired
Info Device 'Livolo4Gang' is supported, identified as: Livolo Zigbee Switch 3 gang (TI0001-switch-3gang)
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Zigbee Switch 3 gang","exposes":[{"endpoint":"left","features":[{"access":7,"description":"On/off state of the switch","endpoint":"left","name":"state","property":"state_left","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"right","features":[{"access":7,"description":"On/off state of the switch","endpoint":"right","name":"state","property":"state_right","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"endpoint":"top","features":[{"access":7,"description":"On/off state of the switch","endpoint":"top","name":"state","property":"state_top","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TI0001-switch-3gang","options":[],"supports_ota":false,"vendor":"Livolo"},"friendly_name":"Livolo4Gang","ieee_address":"Livolo4Gang","status":"successful","supported":true},"type":"device_interview"}'

razim7, 2022-02-23:
4 gang: @systemiii ouch... any other debug I can take to make it easier for you? 
Curtain: the covers not working, the open and close working.

systemiii, 2022-02-23:
Unfortunately not, the 4 gang should be paired with a Livolo gateway and then analyze the sniffed zigbee packets.
I edited my shared livolo2.js on the fly to add the 4 gang switch, based on the directions @whitepail wrote.
I haven't done any test, so make a backup of your old file before updating it.
As Z2M cannot recognize it automatically, you will need to manually change the device name to TI0001-switch-4gang in the database.db file while z2m is stopped, as suggested by @whitepail.

systemiii, 2022-02-23:
@whitepail so you will make a pull request to support all kinds of buttons, right?
Because if you confirm it then I'll stop my development, otherwise we will both do the same thing.

systemiii, 2022-02-23:
@razim7 in my external converters only open and close works for curtain switches. Slider not implemented yet.

razim7, 2022-02-23:
@systemiii when I tested the new livolo.js
I cannot start the addon:

> zigbee2mqtt@1.23.0-dev start
> node index.js
(node:2105) UnhandledPromiseRejectionWarning: SyntaxError: Unexpected token ','
(Use `node --trace-warnings ...` to show where the warning was created)
(node:2105) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:2105) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

systemiii, 2022-02-23:
Try now, please

razim7, 2022-02-23:
3 out of the 4 buttons worked, switch shown with 4 gang and 4 buttons but the last one not working:
Debug Retrieving state of '0x00124b0022e16f69' after reconnect
Debug Received Zigbee message from '0x00124b0022e16f69', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,12,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from '0x00124b0022e16f69', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,13,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from '0x00124b0022e16f69', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,15,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received MQTT message on 'zigbee2mqtt/0x00124b0022e16f69/set' with data '{"state_bottom":"OFF"}'
Debug Publishing 'set' 'state' to '0x00124b0022e16f69'
Debug Received MQTT message on 'zigbee2mqtt/0x00124b0022e16f69/set' with data '{"state_bottom":"OFF"}'
Debug Publishing 'set' 'state' to '0x00124b0022e16f69'
Debug Received MQTT message on 'zigbee2mqtt/0x00124b0022e16f69/set' with data '{"state_bottom":"OFF"}'
Debug Publishing 'set' 'state' to '0x00124b0022e16f69'
Info MQTT publish: topic 'zigbee2mqtt/0x00124b0022e16f69', payload '{"linkquality":43,"state_bottom":null,"state_bottom_bottom":"OFF","state_left":null,"state_left_left":"OFF","state_right":null,"state_right_right":"OFF","state_top":null,"state_top_top":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/0x00124b0022e16f69', payload '{"linkquality":43,"state_bottom":null,"state_bottom_bottom":"OFF","state_left":null,"state_left_left":"OFF","state_right":null,"state_right_right":"OFF","state_top":null,"state_top_top":"OFF"}'
Debug Received MQTT message on 'zigbee2mqtt/0x00124b0022e16f69/set' with data '{"state_bottom":"ON"}'
Debug Publishing 'set' 'state' to '0x00124b0022e16f69'
Info MQTT publish: topic 'zigbee2mqtt/0x00124b0022e16f69', payload '{"linkquality":43,"state_bottom":null,"state_bottom_bottom":"OFF","state_left":null,"state_left_left":"OFF","state_right":null,"state_right_right":"OFF","state_top":null,"state_top_top":"OFF"}'
Info MQTT publish: topic 'zigbee2mqtt/0x00124b0022e16f69', payload '{"linkquality":43,"state_bottom":null,"state_bottom_bottom":"ON","state_left":null,"state_left_left":"OFF","state_right":null,"state_right_right":"OFF","state_top":null,"state_top_top":"OFF"}'

when I turn it on and off manually this is what I see in the debug: 

 Received Zigbee message from '0x00124b0022e16f69', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,7,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from '0x00124b0022e16f69', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,15,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from '0x00124b0022e16f69', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,7,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from '0x00124b0022e16f69', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,15,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from '0x00124b0022e16f69', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,15,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from '0x00124b0022e16f69', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,7,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0
Debug Received Zigbee message from '0x00124b0022e16f69', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,105,111,225,34,0,75,18,0,13,1,0,15,13,11],"type":"Buffer"}' from endpoint 6 with groupID 0

whitepail, 2022-02-25:
> @whitepail so you will make a pull request to support all kinds of buttons, right? Because if you confirm it then I'll stop my development, otherwise we will both do the same thing.

I'm in progress of 'try and fail' with 4 gang switch. If I won't get any result with identification - I will remove my 4gang from the wall and will have to pair it with livolo gateway with wireshark sniffing on the table. Only have a couple of spare hours per week to do implementation and testing.

whitepail, 2022-02-25:
Regarding livolo2.js

Your implementation:
livolo_new_switch_state_4gang: {
        cluster: 'genPowerCfg',
        type: ['raw'],
        convert: (model, msg, publish, options, meta) => {
            const stateHeader = Buffer.from([122, 209]);
            if (msg.data.indexOf(stateHeader) === 0) {
                if (msg.data[10] === 7) {
                    const status = msg.data[14];
                    return {
                        state_left: status & 1 ? 'ON' : 'OFF',
                        state_right: status & 2 ? 'ON' : 'OFF',
                        state_top: status & 4 ? 'ON' : 'OFF',
			            state_bottom: status & 8 ? 'ON' : 'OFF',
                    };
                }
            }
        },
    },

I think right one should be:
livolo_new_switch_state_4gang: {
        cluster: 'genPowerCfg',
        type: ['raw'],
        convert: (model, msg, publish, options, meta) => {
            const stateHeader = Buffer.from([122, 209]);
            if (msg.data.indexOf(stateHeader) === 0) {
                if (msg.data[10] === 13) {
                    const status = msg.data[13];
                    return {
                        state_left: status & 1 ? 'ON' : 'OFF',
                        state_right: status & 2 ? 'ON' : 'OFF',
                        state_bottom-left: status & 4 ? 'ON' : 'OFF',
                        state_bottom-right: status & 8 ? 'ON' : 'OFF',
                    };
                };
                if (msg.data[10] === 7) {
                    const status = msg.data[14];
                    return {
                        state_left: status & 1 ? 'ON' : 'OFF',
                        state_right: status & 2 ? 'ON' : 'OFF',
                        state_top: status & 4 ? 'ON' : 'OFF',
			            state_bottom: status & 8 ? 'ON' : 'OFF',
                    };
                }
            }
        },
    },

In this case we will parse both statuses for old and new fw.

razim7, 2022-02-25:
@whitepail @systemiii 
I just tested livolo2.js with what @whitepail suggested, now I can see status of all the 4 buttons and I can control 3 out of 4
so it's much better, still cannot control 1 of the 4 but I can see the accurate status of it.

systemiii, 2022-02-28:
@whitepail thanks for your correction

whitepail, 2022-03-02:
Today I had enough time to pair 4 gang and 1 gang switches with new firmware to livolo gateway. 
All the traffic was sniffed using zboss, tomorrow I will share wireshark dumps here.

Findings:
1) New fw do not send any "magic" to identify the device. 
2) Livolo app assumes all the devices as 1gang switches, adding buttons to the swich as soon as you phisically push them.  My device was first identified as 1-gang switch, when I pushed second button it was converted to 2-gang switch, than to 3-gang etc...
3) problem with 4th button update is due to 'non-standard' Livolo zigbee protocol implementation. 
payloadOnBottomRight = {0x0001: {value: Buffer.from([8, 0, 0, 0, 0, 0, 0, 0]), type: 8}} is wrong construction according to zigbee-herdsman lib, whenever you set type: 8 it removes any payload and send message without payload. I will take a look at zigbee-herdsman lib to check if we can do anything with that.

I've updated main TI0001 base implementation in Z2M so it will support any versions of livolo switches with up to 4 gangs.
I will publish pull request whenever the problem with 4th buttom will be resolved.

Hope to finish it before github will be blocked for my country.

systemiii, 2022-03-03:
@whitepail 0x08 data-type need only 8 bit data, this is why payload is truncated.
You can try using a sequence of payload data like this

const payload = {0x4292: {value: 0, status: 0x06, type: 1}, 0xfff0: {value: 0, status: 0x3f, type: 1}};

whitepail, 2022-03-04:
It looks like it's ok from Livolo firmware part to put 136 instead of 8
type. From livolo part it's a mask of gangs we are modyfyind, so 136 will
modyfy two bits, 128 (virtual gang) and 8 ( gang n4).

I will check today in prod and return by evening with required modification
to code.

whitepail, 2022-03-04:
You can replace 
	    const payloadOnBottom = {0x0001: {value: Buffer.from([8, 0, 0, 0, 0, 0, 0, 0]), type: 8}};
            const payloadOffBottom = {0x0001: {value: Buffer.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 8}};
to
            const payloadOnBottom = {0x0001: {value: Buffer.from([8, 0, 0, 0, 0, 0, 0, 0]), type: 136}};
            const payloadOffBottom = {0x0001: {value: Buffer.from([0, 0, 0, 0, 0, 0, 0, 0]), type: 136}};

I can confirm that this way it works will all 4 gangs.

systemiii, 2022-03-04:
Great work @whitepail
We will look forward to your pull request.
Thank you

systemiii, 2022-03-04:
Ah, i forgot to tell you that i have found unpair command

const payload = {0x4292: {value: 0, status: 0x06, type: 1}, 0xfff0: {value: 0, status: 0x3f, type: 1}, 0xfff1: {value: 0, status: 0x01, type: 1}, 0xfff2: {value: 0, status: 0x01, type: 1}};             if (confirmation === 'ON') {                 await entity.writeResponse('genPowerCfg', 0xe9, payload,                         {                             manufacturerCode: 0x1ad2, disableDefaultResponse: true, disableResponse: true,                             reservedBits: 3, direction: 1,                         });

I have tried to put this in 'stop' event but i dont have enough time to make a try.

systemiii, 2022-03-04:
Hope to continue seeing you here on GitHub @whitepail

razim7, 2022-03-13:
@whitepail do you have updated version of livolo2.js? while we are waiting for the GA?

vladurash, 2025-01-03:
after 3 years, I have another data for new Livolo C9 zigbee.
This is what I get from z2m debug logs.

z2m: Received Zigbee message from 'Livolo_hol', type 'raw', cluster 'genPowerCfg', data '{"data":[122,209,36,161,54,35,0,75,18,0,7,1,0,1,0,11],"type":"Buffer"}' from endpoint 6 with groupID 0
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Livolo_hol', payload '{"device":{"dateCode":"20060831        ","friendlyName":"Livolo_hol","hardwareVersion":0,"ieeeAddr":"0x00124b002336a124","manufacturerID":0,"manufacturerName":"LIVOLO","model":"TI0001","networkAddress":43698,"powerSource":"Mains (single phase)","type":"EndDevice","zclVersion":0},"last_seen":"2025-01-03T19:46:36.272Z","linkquality":36,"state_bottom_left":"OFF","state_bottom_right":"OFF","state_left":"OFF","state_right":"OFF"}'

Also converted to hex returned values, trying to understand what you did, but I'm lost in space :|.

{"data":[122,209,    36,161,54,35,0,75,18,0,  7,1,0,1,0,11],"type":"Buffer"}

122=0x7a   - 0
209=0xd1   - 1
 36=0x24   - 2
161=0xa1   - 3
 54=0x36   - 4
 35=0x23   - 5
  0=0x0    - 6
 75=0x4b   - 7
 18=0x12   - 8
  0=0x0    - 9
  7=0x7    - 10
  1=0x1    - 11
  0=0x0    - 12
  1=0x1    - 13
  0=0x0    - 14
 11=0xb    - 15