Koenkk/zigbee2mqtt #8076: K4027C roller shutter

diego065, 2021-07-19:
Hello everyone!
Yesterday I connected my new bTicino devices for control the roller shutter (K4027C): the device was correctly recognized by zigbee2mqtt addon but unfortunately I cannot set the position. Only 0 and 100 works (fully opened or fully closed): any ideas on how I can fix this problem?
Thanks a lot

Koenkk, 2021-07-19:
As I found in https://github.com/Koenkk/zigbee-herdsman-converters/pull/2214 this seems to be an issue of the device firmware, so looks it cannot be fixed from z2m.

diego065, 2021-07-20:
So it’s not possible to control other than open or close? Do you think that the problem will be solved with some fw update in the future? Thanks