Koenkk/zigbee2mqtt #8078: New Home Assistant select entities send invalid values

frenck, 2021-07-19:
## What happened

Home Assistant reports warnings for `select` entities provided/created by Zigbee2MQTT.

## What did you expect to happen

No to have warnings in my logs

## How to reproduce it (minimal and precise)

Have an integration that produces an `select` entity.

## Debug info

Zigbee2MQTT version: 1.20.0-dev commit: 46e2a77
Adapter hardware: Conbee II
Adapter firmware version: 0x26660700

## Additional information

[image]

ruifung, 2021-07-20:
I'm honestly not quite sure where the 'None' is being published from?

Does Z2M publish None as a state attribute...? I've never actually seen this particular warning show up in my logs though.
I'm curious to know what the state payloads look like for those devices though. I think I remember Z2M not publishing state values for enum exposes sometimes if the state is unknown... hmm.

I can't see having None be a valid option in the select entity as a good idea, but I can't see how you could send a state attribute for something that isn't known.

frenck, 2021-07-20:
Well at this point, it defined possible values and next set a value that isn't possible according to the list provided by z2m... 🤷

frenck, 2021-07-20:
Hmm... actually, this might be a bug in the MQTT select entity @ Home Assistant...

frenck, 2021-07-21:
Fixed upstream!

closed 2021-07-21 by frenck