Koenkk/zigbee2mqtt #8079: Zigbee2MQTT not sending unique

frenck, 2021-07-19:
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.20.0-dev commit: 46e2a77
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

## Additional

[image]

Search shows it as well

Koenkk, 2021-07-20:
I think this is due to https://github.com/Koenkk/zigbee2mqtt/commit/679d2317438a3f15b2dbc912919122558f0befdf , the old topics are not cleaned up. Can you confirm this by clearing the `homeassistant/light/1/light` message?

frenck, 2021-07-20:
Yep! Missed that one, thanks @Koenkk 👍

closed 2021-07-20 by frenck

Koenkk, 2021-07-20:
@frenck thanks for confirming, modified the code to also clear it automatically now.