Koenkk/zigbee2mqtt #8080: win10 npc ci error

tiancom, 2021-07-19:
PS C:\zigbee2mqtt-master> npm ci
npm WARN prepare removing existing node_modules/ before installation

> @serialport/bindings@9.0.4 install C:\zigbee2mqtt-master\node_modules\zigbee-herdsman-converters\node_modules\zigbee-herdsman\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> serialport@9.0.6 postinstall C:\zigbee2mqtt-master\node_modules\zigbee-herdsman-converters\node_modules\zigbee-herdsman\node_modules\serialport
> node thank-you.js

Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

> @serialport/bindings@9.0.4 install C:\zigbee2mqtt-master\node_modules\zigbee-herdsman\node_modules\@serialport\bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

> serialport@9.0.6 postinstall C:\zigbee2mqtt-master\node_modules\zigbee-herdsman\node_modules\serialport
> node thank-you.js

Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

> unix-dgram@2.0.3 install C:\zigbee2mqtt-master\node_modules\unix-dgram
> node-gyp rebuild

C:\zigbee2mqtt-master\node_modules\unix-dgram>if not defined npm_config_node_gyp (node "C:\Program Files\nodejs\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )
在此解决方案中一次生成一个项目。若要启用并行生成，请添加“/m”开关。
  unix_dgram.cc
  win_delay_load_hook.cc
..\src\unix_dgram.cc(9): fatal error C1083: 无法打开包括文件: “unistd.h”: No such file or directory [C:\zigbee2mqtt-master\node
_modules\unix-dgram\build\unix_dgram.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\MSBuild\14.0\bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Windows_NT 10.0.19042
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd C:\zigbee2mqtt-master\node_modules\unix-dgram
gyp ERR! node -v v12.18.4
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
added 931 packages in 57.846s
PS C:\zigbee2mqtt-master>

tiancom, 2021-07-19:
PS C:\WINDOWS\system32> cd\
PS C:\> cd z2m
PS C:\z2m> git clone git@github.com:Koenkk/zigbee2mqtt.git
Cloning into 'zigbee2mqtt'...
git@github.com: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
PS C:\z2m>

Kung-Foo-Kamel, 2021-07-20:
I have the same issue as you have with the build error, tried both VS2017 and VS2019.

Your second issue, use this instead git clone https://github.com/Koenkk/zigbee2mqtt.git

Kung-Foo-Kamel, 2021-07-20:
I just did the build in a VM of Debian and it compiled perfectly and is now running, recommend not using Windows for this due to the build errors.

tiancom, 2021-07-20:
> I have the same issue as you have with the build error, tried both VS2017 and VS2019.

So this is not compatible with windows and can't be used at all

Kung-Foo-Kamel, 2021-07-21:
I wouldn't go as far as saying that, there is a bug when compiling on Windows that's all I know, so I moved straight to Linux and as usual, it works straight away.

Someone might chime in with a fix, but for me, I'm impatient and theirs always more than one way to skin cat. 

Compiling these projects on Windows can be a little challenging at times, If you want plugins like these to run 24/7 for reliable home automation, I recommend Linux, either on an RPI or in a VM on your Windows machine, an easy start is spin up a Virtualbox image of DietPi, most of the compiling tools and home automation stuff is ready to install in a few clicks.

MaartenVerb, 2021-07-27:
I also gave up on git cloning because of the publickey permissions denied. 
In the end I downloaded the repository as a zip, unpacked it. 

Installed nodejs "node-v14.17.1-x64.msi" minimal without all the extra stuff you can get.

Then went to the folder (in my case C:\!Programs\Zigbee2MQTT) and did the 'npm ci' command.

Then it did stuff, got a bunch of errors/warnings and it still worked. using npm start.

Consider using the experimental webinterface for further configuring, I discovered it today and it works nice;
Add 
frontend:
  port: 8080
to the C:\!Programs\Zigbee2Mqtt\data\configuration.yaml.

then on your computer go to http://localhost:8080

closed 2021-09-03 by github-actions[bot]