Koenkk/zigbee2mqtt #8082: Paulmann 501.27 cannot be paired

SharkyRawr, 2021-07-20:
Hi,
I'm trying to pair a Paulmann 501.27 E14 RGB light bulb but it doesn't seem to want to pair. I set PermitJoin and followed the reset instructions for the lamp and it starts to blink but I don't see anything happening in the zigbee2mqtt logs. Nothing at all.
How do I debug this? The RasPi with the ZigBee stick is really close (one room over, about 3 meters).

Using a Texas Instruments CC2531 ZigBee stick; USBID 0451:16a8

Any help would be much appreciated!

robertalexa, 2021-07-20:
Try setting your log level to debug to start with. You can do this via configuration and restart or via the Frontend without restart. 

Keep the logs open. Permit Join, if you use the frontend, there is a button in the top menu for it. The log should show an entry for this too.

Take your device and bring it literally next to your coordinator. Put it in pairing mode.

Now keep reading the logs, see what you get. If you do see entries relating to this, and your device still doesn't pair, post your log.

Hope this helps

SharkyRawr, 2021-07-20:
Getting the Coordinator closer to the light definitely helped, thanks!
The light wasn't supported yet so I tried writing a converter for the first time: Koenkk/zigbee-herdsman-converters#2817
and it works great!

robertalexa, 2021-07-20:
@SharkyRawr I am glad this worked. The best way to get devices paired is to get the as close as possible to the coordinator, or at least try it if pairing from a distance fails.

Feel free to close the issue if you are satisfied, to keep the repo clean.

PS: i have left a note in your PR for the converter

closed 2021-07-20 by SharkyRawr