Koenkk/zigbee2mqtt #8083: New Home Assistant select entities all enabled by default

frenck, 2021-07-20:
## What happened

Extracted this comment into this issue: <https://github.com/Koenkk/zigbee2mqtt/pull/7980#issuecomment-882789802>

Introduced in #7980, also related to #8078

[image]

These entities are not really "common". As in, they can be useful, but probably most people won't use them in their automation or dashboards in general. Therefore, it is recommended to send a disabled by default in Home Assistant. This allows a user to enable them when needed, but not overwhelm the user with entities they do not need or pollute the (historical) database unneeded.

Can we adjust this for this feature before release it to the public?

## What did you expect to happen

To have less common entities disabled by default, keeping setups nice and tidy.

## How to reproduce it (minimal and precise)

## Debug info

Zigbee2MQTT version: 1.20.0-dev commit: 46e2a77
Adapter hardware: Conbee II
Adapter firmware version: 0x26660700

ruifung, 2021-07-20:
The same could be said of the older sensor entity used to expose enums though, and that is still used to expose composite exposes.

frenck, 2021-07-20:
> sensor entity used to expose enums though

This has been corrected in #7684 and released in 1.20.0.
It will only affect new devices.

ruifung, 2021-07-20:
Well, according to that PR, that behavior is not applied to the enum / text / composite expose types though. 

Personally, I'm not entirely sure if the entire category of enum exposes can be classified as to be disabled by default.

frenck, 2021-07-20:
> In fact, when I authored that PR, I was mostly following existing behavior of those 3 expose types which are epxosed as sensor entities that were enabled by default.

Actually, you are right! 👍
I do have sensors for those already (although all manually disabled).

> Personally, I'm not entirely sure if the entire category of enum exposes can be classified as to be disabled by default.

Maybe just enum isn't specific enough to decide on exposing it with, so instead, we can add them to the lookup table and handle it that way.

PR #8087 (2021-07-20): Handle enabled by default for enum types / select|sensor entities
This PR is to get some contents in the lookup table of enum types exposing neater entities to Home Assistant by providing icons and disabling them by default if less common.

Note: this is a backward compatible change, existing entities of devices are not disabled, only new ones.

fixes #8083
closes #8084
closed 2021-07-20 by frenck