Koenkk/zigbee2mqtt #8085: Switch_type sensor is not available for Xiaomi SSM-U01 relay

MoonlightGuest, 2021-07-20:
# Bug Report
## What happened
After adding Xiaomi SSM-U01 relay Switch_type sensor is not available.
When I try to publish the topic zigbee2mqtt/Relay BR1/get with payload {"switch_type": ""} I get the next error in the zigbee2qmtt log:

Zigbee2MQTT:error 2021-07-20 15:49:02: No converter available for 'switch_type' ()

The same error when I publish the topic zigbee2mqtt/Relay BR1/set with payload {"switch_type": "momentary"}

Also there is a message in the Home Assistant log:
Template variable warning: 'dict object' has no attribute 'switch_type' when rendering '{{ value_json.switch_type }}'

## What did you expect to happen

Expose Switch_type

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.18.1-1
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20201127

Koenkk, 2021-07-20:
You are using an old z2m version.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2021-07-20 by Koenkk

paulbusse, 2022-12-07:
I experience the same problem with version 1.28.4 with an SSM-U02. What I see is that a MQTT select entity is created as follows
```json
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state",
      "value_template": "{{ value_json.state }}"
    },
    {
      "topic": "zigbee2mqtt/Switch Provisie Licht/availability",
      "value_template": "{{ value_json.state }}"
    }
  ],
  "availability_mode": "all",
  "command_topic": "zigbee2mqtt/Switch Provisie Licht/set/switch_type",
  "device": {
    "configuration_url": "https://framboos.bv.lan:8080/#/device/0x54ef4410001a83ca/info",
    "identifiers": [
      "zigbee2mqtt_0x54ef4410001a83ca"
    ],
    "manufacturer": "Xiaomi",
    "model": "Aqara single switch module T1 (without neutral). Doesn't work as a router and doesn't support power meter (SSM-U02)",
    "name": "Switch Provisie Licht"
  },
  "entity_category": "config",
  "icon": "mdi:tune",
  "name": "Switch Provisie Licht switch type",
  "options": [
    "toggle",
    "momentary"
  ],
  "state_topic": "zigbee2mqtt/Switch Provisie Licht",
  "unique_id": "0x54ef4410001a83ca_switch_type_zigbee2mqtt",
  "value_template": "{{ value_json.switch_type }}"
}
```
Not every message that appears on the `state_topic` contains the `switch_type`. Only when the value changes `switch_type` is included. This is what I see in MQTT explorer.
```json
{
  "device_temperature": 15,
  "linkquality": 94,
  "power_outage_count": 0,
  "power_outage_memory": true,
  "state": "OFF"
}
```
As there is no value for `switch_type` HA(22.11.5) produces 2 log messages:
```
WARNING (MainThread) [homeassistant.helpers.template] Template variable warning: 'dict object' has no attribute 'switch_type' when rendering '{{ value_json.switch_type }}'
ERROR (MainThread) [homeassistant.components.mqtt.select] Invalid option for select.switch_provisie_licht_switch_type: '' (valid options: ['toggle', 'momentary'])
```

Also, I found that whenever I restart Z2M the device looses its `switch_type` setting. I'm not sure it is related.