Koenkk/zigbee2mqtt #8086: Zigbee2MQTT.io Docgen Dependency On Old Version Of Ruby

danieledwardgeorgehitchcock, 2021-07-20:
## What happened
I tried to get the docgen for Zigbee2MQTT.io running on my machine using the latest stable version of Ruby (3.0). When running the bundle install command, I get the following error:

`listen-3.2.1 requires ruby version >= 2.2.7, ~> 2.2, which is incompatible with the current version, ruby 3.0.2`

The trouble is, in Ubuntu, the lowest available version of Ruby is now 2.3 stable, which I also tried and got a similar error to the above

## What did you expect to happen
Bundle install completes and allows me to execute the jekyll service

## How to reproduce it (minimal and precise)
Install the latest APT / SNAP version of Ruby on Ubuntu 20.04 LTS, run through the Zigbee2MQTT.io readme

## Debug info
N/A

Koenkk, 2021-07-20:
Updated the dependencies, can you pull and check again? (works for me now with ruby 2.6.0)

danieledwardgeorgehitchcock, 2021-07-21:
Thanks!

I am still struggling with it, but think that is now probably my machine.

I will spin it up in Docker.

Closing for now.

closed 2021-07-21 by danieledwardgeorgehitchcock