Koenkk/zigbee2mqtt #8088: EcoDim Eco-Dim.07 issue

traveltje2, 2021-07-20:
## What happened
Tried adding a new EcoDim Eco-Dim.07. Interview = ok. Device status = unsupported.
The one I just installed seems to have a different fmw version, maybe thats the problem?

[image]
[image]
[image]

## What did you expect to happen
I have 2 other EcoDim Eco-Dim.07 that work perfectly.
The eco-dim I just installed gives the message: Unsupported 

## How to reproduce it (minimal and precise)
Install eco-dim and pair it.

## Debug info
Zigbee2MQTT version: 1.18.1
Adapter hardware: zzh! CC2652R
Adapter firmware version: 
Zigbee2MQTT version
1.18.1 commit: 1.18.1
Coordinator type
zStack3x0
Coordinator revision
20201026
Frontend version
0.3.76

Koenkk, 2021-07-20:
Updating to the latest z2m version will fix it.

If you are still on 1.18.1 check first point of https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.19.0

closed 2021-07-20 by Koenkk

traveltje2, 2021-07-20:
Thank you for your help and very cool work!!!

I've installed the edge version but it still says unsupported?
Am I doing something wrong?

[image]

traveltje2, 2021-07-20:
> The first version is a slightly difference device added in [Koenkk/zigbee-herdsman-converters#2803](https://github.com/Koenkk/zigbee-herdsman-converters/issues/2803)
> 
> Currently only available in the dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Am I correct that my version is not supported yet?
https://www.ecodim.nl/eco-dim07-zigbee-pro.html

Koenkk, 2021-07-20:
Could you share your data/database.db entry?

traveltje2, 2021-07-20:
> Could you share your data/database.db entry?

Yes of course.. I'm sorry I don't know exactly what you are asking or where I can find it?
I have SQLite installed.

traveltje2, 2021-07-20:
> Could you share your data/database.db entry?

By the way I'm having issues with the Eco-Dim.07 and not with the Eco-Dim.05.

Koenkk, 2021-07-20:
It is a text file which you can find in the data folder next to your `configuration.yaml`.

traveltje2, 2021-07-20:
> It is a text file which you can find in the data folder next to your `configuration.yaml`.

I dont have a data folder but i do have a zigbee2mqtt folder:
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b00215ffe68","nwkAddr":0,"manufId":0,"epList":[242,47,13,12,110,11,10,8,6,5,4,3,2,1],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}

Koenkk, 2021-07-20:
This looks like the database.db, when the ecodim is paired there should be more in this file.

traveltje2, 2021-07-20:
> This looks like the database.db, when the ecodim is paired there should be more in this file.

{"id":8,"type":"Router","ieeeAddr":"0x0c4314fffe236913","nwkAddr":2686,"manufId":4714,"manufName":"EcoDim BV","powerSource":"Mains (single phase)","modelId":"EcoDim-Zigbee 3.0","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"EcoDim-Zigbee 3.0","manufacturerName":"EcoDim BV","powerSource":1,"zclVersion":3,"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20210228","swBuildId":"3.09"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":3,"stackVersion":6,"hwVersion":1,"dateCode":"20210228","swBuildId":"3.09","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1626813355968}

Koenkk, 2021-07-21:
Should be recognised as a EcoDim 05 now. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

traveltje2, 2021-07-21:
> Should be recognised as a EcoDim 05 now.
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

You sir... are a hero!!!! Thank you!
By the way it is recognized as EcoDim 07 (as it should).

traveltje2, 2021-07-29:
Sorry for the late coffee :)

aykin, 2022-09-03:
Hello @Koenkk . It looks like this change has broken the support for EcoDim-Zigbee 3.0 which is no longer detected as a supported device. I'm looking into the following commits if it gets fixed but otherwise I will make a new PR to mostly revert this

Koenkk, 2022-09-06:
@aykin 

- can you provide the data/database.db entry of that device?
- should it be detected as `Eco-Dim.05` or `Eco-Dim.07/Eco-Dim.10`?