Koenkk/zigbee2mqtt #8090: aqara TVOC Air Quality Monitor (AAQS-S01)

das399igste, 2021-07-20:
Hello,
I am trying to add a new device from Xiaomi, the TVOC Air Quality Monitor (AAQS-S01).

I was able to pair the device and created an external converter file
AAQS-S01.js
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.airmonitor.acn01'],
    model: 'AAQS-S01',
    vendor: 'Xiaomi',
    description: 'Aqara TVCO Air Quality Monitor',
    fromZigbee: [fz.xiaomi_battery, fz.WSDCGQ01LM_WSDCGQ11LM_interval, fz.xiaomi_temperature, fz.humidity],
    toZigbee: [],
    exposes: [e.battery(), e.temperature(), e.humidity(), e.battery_voltage()],
};

module.exports = definition;
`
But, the device is not reporting anything.

I don't see any values under device details either.

I found the following in the log file:

`MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Aqara TVCO Air Quality Monitor","exposes":[{"access":1,"description":"Remaining battery in %","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Measured temperature value","name":"temperature","property":"temperature","type":"numeric","unit":"\u00b0C"},{"access":1,"description":"Measured relative humidity","name":"humidity","property":"humidity","type":"numeric","unit":"%"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"AAQS-S01","supports_ota":false,"vendor":"Xiaomi"},"friendly_name":"0x54ef44100010687b","ieee_address":"0x54ef44100010687b","status":"successful","supported":true},"type":"device_interview"}'`

What am I doing wrong or who can help me to create the configuration?

Best wishes
Harald

dasfuu, 2021-07-20:
See this issue :)
https://github.com/Koenkk/zigbee2mqtt/issues/6872

Koenkk, 2021-07-21:
Already added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/2815 . Available in the dev branch now: https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

closed 2021-07-21 by Koenkk

danieledwardgeorgehitchcock, 2021-07-21:
When I looked into it, I found that what you had used was a designation number not the model number. If you look closely at the back of the box just below top left, you should see the value that I have used (I may be wrong though)?

The value that I have used also aligns more the numbering convention that Xiaomi Aqara have used with their other sensors - see the Zigbee2MQTT wiki to confirm.

my device handler should work with your device anyway as Zigbee2MQTT only uses that field for documentation purposes.

Please let me know if this isn’t the case

das399igste, 2021-07-21:
Thanks, it works fine with the settings.

The model stated on the package is AAQS-S01.
[image]

As SKU but AS029GLW02. 
[image]