Koenkk/zigbee2mqtt #8093: Groups not exposed to Home Assistant when devices in group are unreachable

frenck, 2021-07-21:
## What happened

I've defined a group, exposed to Home Assistant which all automatically discovers now, so that is awesome!

However, one of my kids touched a switch! (the horror)... Which made a group of bulb powerless.
Not a big deal, but I restarted Home Assistant in between.

Resulting in:

[image]

[image]

## What did you expect to happen

With this current logic, automations would start to fail in various ways and the dashboards show big yellow warnings as the specified entity doesn't exist anymore.

I expected the entities in Home Assistant to be there (in an unavailable state), as that would show unavailable on dashboards and can be taken into account when automating.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.20.0-dev commit: 321b347
Adapter hardware: Conbee II
Adapter firmware version: 0x26660700

Koenkk, 2021-07-21:
Devices marked as unavailable does not influence the Home Assistant group discovery.

Given that the the group shows: 1.19.1-dev while you are running 1.20.0 indicates that this is an outdated group where all entities have been undiscovered from and HA did not delete it.

frenck, 2021-07-21:
🤦 

[image]

There is another one that is correct.

I really need to stop using my actual production machine for all the testing and experiments.

Thanks @Koenkk, sorry for the spam.

closed 2021-07-21 by frenck