Koenkk/zigbee2mqtt #8096: The homeassistant_legacy_entity_attributes option delete the last_seen attribute

Enzokot, 2021-07-21:
## What happened
If I set the option `homeassistant_legacy_entity_attributes` to `false` then the attribute `last_seen` deleted from all sensors
## What did you expect to happen
The option `homeassistant_legacy_entity_attributes` don't delete the attribute `last_seen` 
## How to reproduce it (minimal and precise)
Set  `"homeassistant_legacy_entity_attributes": false` and `"last_seen": "epoch"`
## Debug info
Zigbee2MQTT version: 1.20.0
Adapter hardware: CC26X2R1
Adapter firmware version: zStack3x0 20210218

Koenkk, 2021-07-21:
`homeassistant_legacy_entity_attributes` will indeed disable all attributes on the discovered HA entities (including `last_seen`) so that is expected.

Enzokot, 2021-07-21:
Great option, thanks!

closed 2021-07-21 by Enzokot