Koenkk/zigbee2mqtt #8107: Allow Herdsman Converters To Be Created With Multiple Models / Brands

danieledwardgeorgehitchcock, 2021-07-22:
## Is your feature request related to a problem? Please describe
It has become clear that there are several Zigbee devices now coming to market that are either the same device that has been branded i.e. Sylvania / Lightify / LEDvance or even the same device being released with different model numbers by the same manufacturer for different territories i.e. Aqara VOCKQJK11LM / AAQS-S01

## Describe the solution you'd like
It would be beneficial if when creating the converters / documentation that we could also specify these different brands / models so that when people search for device compatibility without attempting to join the device to a Zigbee network, they can see it is supported.

## Describe alternatives you've considered
Creating documentation for each model / brand of the device and submitting to the Zigbee2MQTT.io repo

## Additional context
Proof of the confusion that this can cause can be seen here #6872

Koenkk, 2021-07-22:
This can already be done via the `whiteLabel` attribute, e.g.:
- https://github.com/Koenkk/zigbee-herdsman-converters/blob/5a7a61e08b3a63d85158641e2c4be2334ec441b4/devices/tuya.js#L82
- https://www.zigbee2mqtt.io/information/supported_devices.html#mumubiz

danieledwardgeorgehitchcock, 2021-07-22:
Thanks!

I have tweaked my converter to take account!

closed 2021-07-22 by danieledwardgeorgehitchcock