Koenkk/zigbee2mqtt #8108: AQARA WXCJKG11LM strange behavior !!!

Docteur-No, 2021-07-22:
The AQARA OPPLE WXCJKG11LM has a strange behavior !!! it's not a Zigbee2MQTT problem, it is a WXCJKG11LM problem.

I use it to pilot my roller shutters and to be able to close or open partialy, I choose to use hold to move and release to stop.
But, when I hold, the WXCJKG11LM automatically send a release after 5 or 10 seconds randomly ?!?! so it is unsusable.

I tried to use "legacy : false" no command works, have to go back
I tried to use {"operation_mode": "command"} no command works, have to go back

Is it known ?
Did AQARA correct this bug ?

kisseler, 2021-08-04:
I had the same problem several months ago. I updated zigbee2mqtt to the latest version und made sure the interview is completed.

That behaviour accured, when the device is connected, but interview was not completed or failed.

Docteur-No, 2021-08-05:
I am sorry it’s really the Aqara Opple which generate a release 
Spying with MQTTbox

closed 2021-09-12 by github-actions[bot]