Koenkk/zigbee2mqtt #8110: After trying to join router to network, zigbee2mqtt crash and require re-flash of firmware.

Anon0ne, 2021-07-22:
## What happened
Issues with CC2531, reflashed latest source routing firmware. Then I flashed a router with firmware, attempted to join network and got an error.
## What did you expect to happen
Normally it would join network and work fine.
## How to reproduce it (minimal and precise)
Flash latest source routing to CC2531, flash routing firmware to second CC2531, try to join CC2531 with network via holding S2 for 5 seconds. Error persists, and the coordinator no longer will work.
## Debug info
Zigbee2MQTT version: docker:latest-dev
Adapter hardware: CC2531
Adapter firmware version: latest

```
Zigbee2MQTT:info  2021-07-22 19:14:42: Starting zigbee-herdsman (0.13.119)

Zigbee2MQTT:debug 2021-07-22 19:14:42: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'

Zigbee2MQTT:debug 2021-07-22 19:14:48: Saving state to file /app/data/state.json

Zigbee2MQTT:error 2021-07-22 19:14:48: Not connected to MQTT server!

Zigbee2MQTT:error 2021-07-22 19:14:48: Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline

Zigbee2MQTT:info  2021-07-22 19:14:48: Disconnecting from MQTT server

(node:6) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'end' of undefined

    at MQTT.disconnect (/app/lib/mqtt.js:93:21)

    at Controller.stop (/app/lib/controller.js:204:9)

    at stop (/app/index.js:95:5)

    at process.handleQuit (/app/index.js:101:9)

(Use `node --trace-warnings ...` to show where the warning was created)

(node:6) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)

(node:6) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

Zigbee2MQTT:error 2021-07-22 19:14:50: Error while starting zigbee-herdsman

Zigbee2MQTT:error 2021-07-22 19:14:50: Failed to start zigbee

Zigbee2MQTT:error 2021-07-22 19:14:50: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions

Zigbee2MQTT:error 2021-07-22 19:14:50: Exiting...

Zigbee2MQTT:error 2021-07-22 19:14:50: Error: SRSP - ZDO - extFindGroup after 6000ms

    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35)

    at listOnTimeout (internal/timers.js:557:17)

    at processTimers (internal/timers.js:500:7)
```

Anon0ne, 2021-07-22:
This also crashed all my devices and Home Assistant now shows all entities as unavailable.
```

Zigbee2MQTT:debug 2021-07-22 19:29:08: Received MQTT message on 'zigbee2mqtt/bridge/groups' with data '[]'

Zigbee2MQTT:debug 2021-07-22 19:29:08: Received MQTT message on 'zigbee2mqtt/bridge/extensions' with data '[]'

Zigbee2MQTT:debug 2021-07-22 19:29:13: Received MQTT message on 'zigbee2mqtt/bridge/config/devices/get' with data ''
```

Anon0ne, 2021-07-22:
There could be more to this, as now after I reflash the device and reboot zigbee2mqtt; it reboots the first time and connects. Then all devices return raw data. I reboot again, attempting a fresh start; the device no longer connects. Not sure what has happened. I've attempted reflash of latest source routing several times with failures.

Koenkk, 2021-07-24:
Can you provide the herdsman debug log of this?

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

Anon0ne, 2021-07-26:
I will get one tomorrow. I flashed again after fighting the adapter for some time after trying to join router again; I kept the router offline for several days and tried it again yesterday. I enabled the router, and that same error popped up again. Zigbee2mqtt was unresponsive, and I tried to replug the stick and reboot the docker container.

It failed again, requiring another re-flash of firmware. It's quite irritating, but I will try to get some further logs for you.

closed 2021-09-02 by github-actions[bot]