Koenkk/zigbee2mqtt #8117: No support for MI motion sensor RTCGQ002LM???

waldi2909, 2021-07-23:
Hello, I tried to add a new motion detector in "zigbee2mqtt". to do this, you have to press the reset button for approx. 8 seconds. Then the device starts to flash blue. after "zigbee2mqtt" didn't find anything in the "Home Assistant", I tried to find a solution under "https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html". Unfortunately, the device does not appear in the log file either. Then I set "permit_join" to true in the "configuration.yaml". But that also had no effect. The homserver was also restarted several times. I just can't find the solution. Can you help me please?

## Information about the device + link
https://www.amazon.de/gp/product/B08SQKQSX2/ref=ppx_yo_dt_b_asin_image_o00_s00?ie=UTF8&psc=1
Manufacturer: MI; 
Battery: CR2450; 
CMIIT ID: 2020DP12044; 
Connection via bluetooth and Sigbee

## data/database.db entry of the device
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b001cce0938","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1024,"inClusterList":[25],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":49246,"epId":2,"devId":2080,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{},"lastSeen":null}
{"id":2,"type":"Router","ieeeAddr":"0x5c0272fffe8aaf6b","nwkAddr":62239,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 WS opal 1000lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64636],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 WS opal 1000lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":2,"appVersion":32,"stackVersion":98,"hwVersion":1,"dateCode":"20190308","swBuildId":"2.0.022"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":32,"stackVersion":98,"hwVersion":1,"dateCode":"20190308","swBuildId":"2.0.022","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1627050687865}
{"id":3,"type":"EndDevice","ieeeAddr":"0xb4e3f9fffe5aaf0b","nwkAddr":6377,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI on/off switch","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,1,3,9,32,4096,64636],"outClusterList":[3,4,6,8,25,258,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI on/off switch","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":3,"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190723","swBuildId":"2.2.010"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":87}}},"binds":[{"cluster":6,"groupID":901,"type":"group"},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001cce0938","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190723","swBuildId":"2.2.010","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1627050687472}
{"id":4,"type":"Group","groupID":901,"members":[],"meta":{}}
{"id":5,"type":"EndDevice","ieeeAddr":"0x804b50fffef5f397","nwkAddr":18848,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI remote control","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,1,3,32,4096,64636],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI remote control","manufacturerName":"IKEA of Sweden","powerSource":3,"zclVersion":3,"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190401","swBuildId":"2.3.014"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":100}}},"binds":[{"cluster":6,"groupID":901,"type":"group"},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001cce0938","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":33,"stackVersion":98,"hwVersion":1,"dateCode":"20190401","swBuildId":"2.3.014","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1627030909945}
{"id":6,"type":"EndDevice","ieeeAddr":"0x00124b0009727a1c","nwkAddr":4877,"manufId":0,"manufName":"Bitron Video","powerSource":"Battery","modelId":"AV2010/22A","epList":[1,2,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"AV2010/22A","manufacturerName":"Bitron Video","powerSource":3,"zclVersion":1,"appVersion":20,"hwVersion":0,"dateCode":"20160815        "}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b001cce0938","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":262,"inClusterList":[0],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"hwVersion":0}}},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":1026,"inClusterList":[],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":20,"hwVersion":0,"dateCode":"20160815        ","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1627055131045}
{"id":7,"type":"Router","ieeeAddr":"0x5c0272fffe8a929c","nwkAddr":20827,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 WS opal 1000lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64636],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 WS opal 1000lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":2,"appVersion":32,"stackVersion":98,"hwVersion":1,"dateCode":"20190308","swBuildId":"2.0.022"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":250,"colorTempPhysicalMax":454}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":32,"stackVersion":98,"hwVersion":1,"dateCode":"20190308","swBuildId":"2.0.022","zclVersion":2,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1626945490598}
[... cut ...] e":"20190308","swBuildId":"2.0.022"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":87}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b001cce0938","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":32,"stackVersion":98,"hwVersion":1,"dateCode":"20190308","swBuildId":"2.0.022","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1627055221974}

McGiverGim, 2021-07-23:
This seems a Bluetooth motion sensor, not a ZigBee one?

waldi2909, 2021-07-24:
oh man ... i'm an idiot ... you're right, it's a bluetooth based sensor. I have to see how I can get the bluetooth to work. and which app can integrate the sensor. Thanks for your answer.

McGiverGim, 2021-07-24:
Try with this https://github.com/custom-components/ble_monitor

Koenkk, 2021-07-25:
I will close this issue, since it is a Bluetooth device it is not relevant for Zigbee2MQTT.

closed 2021-07-25 by Koenkk