Koenkk/zigbee2mqtt #8125: HA MQTT integration shows only one entity per device

nickcheung1000, 2021-07-25:
# Bug Report
## What happened
zigbee2mqtt and HA MQTT integration works. Devices appear in HA MQTT integration. However, there is only one entity created for the device when there are multiple attributes are exposed. The attributes are in the device state but not created as device entities in HA. I have tried to search for a similar issue but could not find any. Any suggestion would be highly appreciated. Thanks. 

[image]
[image]

## What did you expect to happen
Device attributes are created as entities. 

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2MQTT version: 1.20.0-1 
Adapter hardware: CC1352P-2
Adapter firmware version: 
Coordinator type: zStack3x0
Coordinator revision: 20210120
Frontend version: 0.4.29

robertalexa, 2021-08-01:
I think this is expected behaviour. Those are attributes of your entity and they show up in the attributes. You can still use the entity attributes in Node Red or HA dashboard cards, have a look at this thread for example https://community.home-assistant.io/t/how-to-use-attributes/139154/14

I hope I am not wrong about this being expected behaviour, if anyone knows more please correct me.

nickcheung1000, 2021-08-03:
Thanks, Robert. 
I upgraded the core to version core-2021.7.4 last night, and now the integration is working as expected. 
[image]

closed 2021-08-03 by nickcheung1000