Koenkk/zigbee2mqtt #8126: Xiaomi LLKZMK11LM turned on when restarting z2m

Jeoffreybauvin, 2021-07-25:
## What happened

When I restart z2m (1.20.0), my Xiaomi LLKZMK11LM  is turning on... I need to manually set it off...

## What did you expect to happen

Not turning on when restarting z2m

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.20.0
Adapter hardware: CC1352P-2
Adapter firmware version: zStack3x0

Koenkk, 2021-07-25:
Can you provide the debug log when this happens? 
See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

Jeoffreybauvin, 2021-07-26:
Yes : 
[logs.txt](https://github.com/Koenkk/zigbee2mqtt/files/6881347/logs.txt)

I did grep for "relay". 

I tested a few things :  

- Turn off the relay
-  Stop z2m
- Delete the mqtt topic related to my relay
- Start z2m
- The relay is turning on at startup :(

Koenkk, 2021-07-27:
As you can see in the log, zigbee2mqtt receives a request to turn on the switch indicating this is not a problem of zigbee2mqtt:

```
juil. 26 23:31:13 domotic npm[8313]: Zigbee2MQTT:debug 2021-07-26 23:31:13: Received MQTT message on 'zigbee2mqtt/xiaomi_relay_garage_etabli_lampe/l1/set' with data 'ON'
juil. 26 23:31:13 domotic npm[8313]: Zigbee2MQTT:debug 2021-07-26 23:31:13: Publishing 'set' 'state' to 'xiaomi_relay_garage_etabli_lampe'
```

Jeoffreybauvin, 2021-07-27:
I don't understand :  it seems Home Assistant is sending the "ON" value , but I'm only using auto discovery from z2m, nothing more.

I don't have any scene nor automation in HA...

Jeoffreybauvin, 2021-07-27:
I'm trying to declare the relay without auto discovery : 

```
- name: xiaomi_relay_garage_etabli_lampe_l1
  platform: "mqtt"
  state_topic: "zigbee2mqtt/xiaomi_relay_garage_etabli_lampe"
  availability_topic: "zigbee2mqtt/bridge/state"
  payload_off: "OFF"
  payload_on: "ON"
  value_template: "{{ value_json.state_l1 }}"
  command_topic: "zigbee2mqtt/xiaomi_relay_garage_etabli_lampe/l1/set"

- name: xiaomi_relay_garage_etabli_lampe_l2
  platform: "mqtt"
  state_topic: "zigbee2mqtt/xiaomi_relay_garage_etabli_lampe"
  availability_topic: "zigbee2mqtt/bridge/state"
  payload_off: "OFF"
  payload_on: "ON"
  value_template: "{{ value_json.state_l2 }}"
  command_topic: "zigbee2mqtt/xiaomi_relay_garage_etabli_lampe/l2/set"
```

HA is still sending ON at every startup of z2m... I can't figure out why. It's always l1, not l2.

DubhAd, 2021-07-28:
Are you sure the problem isn't a retained message on the topic telling the light to turn on? That would match the description of your problem. 

Tools like _MQTT Explorer_ can help you check what's going on with your broker.

Jeoffreybauvin, 2021-07-28:
I tried to set retain to false for my device, but no luck :(. 

I already use MQTT Explorer, and here are the stephs I'm following : 

- HA is started
- stop z2m
- delete the topics related to my device
- start z2m
- My device is turning on, I can see the l1=>SET=>ON in MQTT Explorer...

DubhAd, 2021-07-28:
I'm not talking about retain in the device settings, I'm talking about _a retained message held on the broker_.

Jeoffreybauvin, 2021-07-28:
Oh sorry. But if I delete all the topics related to my device before restarting z2m, there can't be no retained message held on the broker, right ?

DubhAd, 2021-07-29:
Depends on how you're deleting those messages

closed 2021-09-05 by github-actions[bot]