Koenkk/zigbee2mqtt #8127: Are "TuYa TS0222" supported?

Bagunda, 2021-07-25:
In this page https://www.zigbee2mqtt.io/devices/TS0222.html write that it supported.
But:
```
Jul 25 15:32:59 RanchoRPI npm[742]: Zigbee2MQTT:info  2021-07-25 15:32:59: Device '0x84fd27fffec8f526' joined
Jul 25 15:32:59 RanchoRPI npm[742]: Zigbee2MQTT:info  2021-07-25 15:32:59: Starting interview of '0x84fd27fffec8f526'
Jul 25 15:32:59 RanchoRPI npm[742]: Zigbee2MQTT:info  2021-07-25 15:32:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x84fd27fffec8f526"},"type":"device_connected"}'
Jul 25 15:32:59 RanchoRPI npm[742]: Zigbee2MQTT:info  2021-07-25 15:32:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x84fd27fffec8f526"},"type":"pairing"}'
Jul 25 15:33:00 RanchoRPI npm[742]: Zigbee2MQTT:info  2021-07-25 15:33:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x84fd27fffec8f526"},"type":"device_announced"}'
Jul 25 15:33:07 RanchoRPI npm[742]: Zigbee2MQTT:info  2021-07-25 15:33:07: Successfully interviewed '0x84fd27fffec8f526', device has successfully been paired
Jul 25 15:33:07 RanchoRPI npm[742]: Zigbee2MQTT:warn  2021-07-25 15:33:07: Device '0x84fd27fffec8f526' with Zigbee model 'TS0222' and manufacturer name '_TYZB01_4mdqxxnn' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Jul 25 15:33:07 RanchoRPI npm[742]: Zigbee2MQTT:info  2021-07-25 15:33:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x84fd27fffec8f526","supported":false},"type":"pairing"}'

Jul 25 15:52:31 RanchoRPI npm[28060]: Zigbee2MQTT:info  2021-07-25 15:52:31: Starting Zigbee2MQTT version 1.20.0 (commit #41b67fdd)
Jul 25 15:52:31 RanchoRPI npm[28060]: Zigbee2MQTT:info  2021-07-25 15:52:31: Starting zigbee-herdsman (0.13.117)
Jul 25 15:52:34 RanchoRPI npm[28060]: Zigbee2MQTT:info  2021-07-25 15:52:34: zigbee-herdsman started (resumed)
Jul 25 15:52:34 RanchoRPI npm[28060]: Zigbee2MQTT:info  2021-07-25 15:52:34: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
```

Koenkk, 2021-07-25:
After updating Zigbee2MQTT to the latest version, did you do a `npm ci` in the zigbee2mqtt directory?

Bagunda, 2021-07-30:
I did it. And now I did.

```
[00:11:31] root@RanchoRPI:/opt/zigbee2mqtt# npm ci
npm WARN prepare removing existing node_modules/ before installation

> @serialport/bindings@9.0.4 install /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=12.20.2 runtime=node arch=arm libc= platform=linux)
make: Entering directory '/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Leaving directory '/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'

> serialport@9.0.6 postinstall /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/node_modules/zigbee-herdsman/node_modules/serialport
> node thank-you.js

Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

> @serialport/bindings@9.0.4 install /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=12.20.2 runtime=node arch=arm libc= platform=linux)
make: Entering directory '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Leaving directory '/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build'

> serialport@9.0.6 postinstall /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/serialport
> node thank-you.js

Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

> unix-dgram@2.0.3 install /opt/zigbee2mqtt/node_modules/unix-dgram
> node-gyp rebuild

make: Entering directory '/opt/zigbee2mqtt/node_modules/unix-dgram/build'
  CXX(target) Release/obj.target/unix_dgram/src/unix_dgram.o
  SOLINK_MODULE(target) Release/obj.target/unix_dgram.node
  COPY Release/unix_dgram.node
make: Leaving directory '/opt/zigbee2mqtt/node_modules/unix-dgram/build'
added 935 packages in 96.056s
```

closed 2021-07-31 by Koenkk