Koenkk/zigbee2mqtt #8130: Zigbee2MQTT pinning CPU to 100% and freeze the network

golddragon007, 2021-07-25:
## What happened
After a time things stops working, the device states actions doesn't being sent to mqtt. Even the UI doesn't want to load. And node index.js is pinning a CPU to 100%. Restarting the  In the log I see these:
```
2021-07-25T20:43:23.238Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,1,0,211,222,1,1,0,120,0,232,141,137,0,0,7,24,64,10,32,0,32,31,96,201,28] - 185
2021-07-25T20:43:23.238Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":1,"srcaddr":57043,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":120,"securityuse":0,"timestamp":9014760,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[24,64,10,32,0,32,31]}}
2021-07-25T20:43:23.239Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":64,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":32,"dataType":32,"attrData":31}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":57043,"endpoint":1,"linkquality":120,"groupID":0}'
2021-07-25T20:43:23.241Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,4,11,152,76,1,1,0,69,0,98,158,137,0,0,8,8,43,10,8,5,33,34,0,152,76,29,212,254,28,68,129,0,0,4,11,20,136,1,1,0,51,0,38,209,137,0,0,8,8,61,10,11,5,41,80,0,20,136,29,198,254,28,68,129,0,0,4,11,152,76,1,1,0,63,0,34,252,137,0,0,8,8,45,10,11,5,41,46,0,152,76,29,141,254,28,68,129,0,0,4,11,20,136,1,1,0,57,0,183,34,139,0,0,8,8,62,10,5,5,33,241,8,20,136,29,0,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,47,91,139,0,0,8,8,46,10,8,5,33,35,0,152,76,29,93,254,28,68,129,0,0,4,11,109,223,1,1,0,75,0,228,0,140,0,0,8,8,53,10,5,5,33,5,9,109,223,29,250,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,17,3,140,0,0,8,8,47,10,5,5,33,12,9,152,76,29,22,254,28,68,129,0,0,4,11,20,136,1,1,0,51,0,216,10,141,0,0,8,8,63,10,8,5,33,174,0,20,136,29,16,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,216,140,141,0,0,8,8,47,10,5,5,33,12,9,152,76,29,81,254,33,68,129,0,0,4,11,109,223,1,1,0,78,0,162,10,142,0,0,13,8,54,10,8,5,33,30,5,11,5,41,133,8,109,223,29,58,254,28,68,129,0,0,4,11,20,136,1,1,0,48,0,120,160,142,0,0,8,8,64,10,11,5,41,82,0,20,136,29,146,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,68,211,142,0,0,8,8,48,10,11,5,41,45,0,152,76,29,160,254,28,68,129,0,0,4,11,20,136,1,1,0,51,0,100,242,143,0,0,8,8,65,10,5,5,33,240,8,20,136,29,115,254,28,68,129,0,0,4,11,20,136,1,1,0,48,0,250,42,144,0,0,8,8,64,10,11,5,41,82,0,20,136,29,132,254,28,68,129,0,0,4,11,152,76,1,1,0,63,0,154,44,144,0,0,8,8,49,10,8,5,33,34,0,152,76,29,231,254,28,68,129,0,0,4,11,109,223,1,1,0,75,0,246,208,144,0,0,8,8,55,10,5,5,33,3,9,109,223,29,32,254,28,68,129,0,0,4,11,152,76,1,1,0,69,0,24,212,144,0,0,8,8,50,10,5,5,33,11,9,152,76,29,201,254,28,68,129,0,0,4,11,20,136,1,1,0,54,0,81,218,145,0,0,8,8,66,10,8,5,33,176,0,20,136,29,51,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,82,94,146,0,0,8,8,50,10,5,5,33,11,9,152,76,29,12,254,33,68,129,0,0,4,11,109,223,1,1,0,75,0,108,219,146,0,0,13,8,56,10,8,5,33,50,5,11,5,41,131,8,109,223,29,24,254,28,68,129,0,0,4,11,20,136,1,1,0,51,0,14,112,147,0,0,8,8,67,10,11,5,41,85,0,20,136,29,46,254,28,68,129,0,0,4,11,152,76,1,1,0,69,0,96,164,147,0,0,8,8,51,10,11,5,41,44,0,152,76,29,235,254,28,68,129,0,0,4,11,20,136,1,1,0,45,0,16,194,148,0,0,8,8,68,10,5,5,33,239,8,20,136,29,40,254,28,68,129,0,0,4,11,152,76,1,1,0,69,0,23,254,148,0,0,8,8,52,10,8,5,33,35,0,152,76,29,194,254,28,68,129,0,0,4,11,152,76,1,1,0,72,0,231,165,149,0,0,8,8,53,10,5,5,33,9,9,152,76,29,74,254,28,68,129,0,0,4,11,109,223,1,1,0,78,0,217,193,149,0,0,8,8,57,10,5,5,33,3,9,109,223,29,16,254,5,69,196,248,219,1,23,60,141,254,29,68,129,0,0,0,5,248,219,1,1,0,69,0,240,244,149,0,0,9,25,163,0,1,0,0,0,0,0,23,60,28,175,254,28,68,129,0,0,4,11,20,136,1,1,0,48,0,91,170,150,0,0,8,8,69,10,8,5,33,174,0,20,136,29,81,254,5,69,196,204,93,1,181,235,74,254,28,68,129,0,0,0,4,204,93,1,1,0,105,0,45,125,151,0,0,8,24,237,10,0,0,33,136,56,181,235,28,198,254,33,68,129,0,0,4,11,109,223,1,1,0,75,0,132,171,151,0,0,13,8,58,10,8,5,33,192,5,11,5,41,152,9,109,223,29,111,254,5,69,196,141,46,1,143,181,28,254,28,68,129,0,0,0,4,141,46,1,1,0,63,0,127,27,152,0,0,8,24,157,10,0,0,33,0,0,143,181,28,61,254,28,68,129,0,0,4,11,20,136,1,1,0,48,0,54,63,152,0,0,8,8,70,10,11,5,41,85,0,20,136,29,84,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,228,117,152,0,0,8,8,54,10,11,5,41,46,0,152,76,29,181,254,33,68,129,0,0,4,11,109,223,1,1,0,75,0,250,53,153,0,0,13,8,58,10,8,5,33,192,5,11,5,41,152,9,109,223,29,129,254,28,68,129,0,0,4,11,20,136,1,1,0,48,0,153,146,153,0,0,8,8,71,10,5,5,33,242,8,20,136,29,255,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,97,207,153,0,0,8,8,55,10,8,5,33,35,0,152,76,29,140,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,255,255,153,0,0,8,8,54,10,11,5,41,46,0,152,76,29,37,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,214,118,154,0,0,8,8,56,10,5,5,33,11,9,152,76,29,162,254,28,68,129,0,0,4,11,109,223,1,1,0,75,0,249,145,154,0,0,8,8,59,10,5,5,33,2,9,109,223,29,105,254,28,68,129,0,0,4,11,20,136,1,1,0,48,0,26,134,155,0,0,8,8,72,10,8,5,33,175,0,20,136,29,61,254,33,68,129,0,0,4,11,109,223,1,1,0,78,0,190,123,156,0,0,13,8,60,10,8,5,33,48,5,11,5,41,177,8,109,223,29,85,254,28,68,129,0,0,4,11,20,136,1,1,0,51,0,224,14,157,0,0,8,8,73,10,11,5,41,83,0,20,136,29,188,254,28,68,129,0,0,4,11,152,76,1,1,0,69,0,31,71,157,0,0,8,8,57,10,11,5,41,46,0,152,76,29,113,254,28,68,129,0,0,4,11,20,136,1,1,0,51,0,34,98,158,0,0,8,8,74,10,5,5,33,242,8,20,136,29,189,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,83,160,158,0,0,8,8,58,10,8,5,33,35,0,152,76,29,219,254,28,68,129,0,0,4,11,152,76,1,1,0,69,0,84,72,159,0,0,8,8,59,10,5,5,33,13,9,152,76,29,25,254,28,68,129,0,0,4,11,109,223,1,1,0,75,0,93,98,159,0,0,8,8,61,10,5,5,33,3,9,109,223,29,60,254,28,68,129,0,0,0,4,128,197,1,1,0,75,0,22,58,160,0,0,8,24,113,10,0,0,33,33,78,30,75,28,51,254,28,68,129,0,0,4,11,20,136,1,1,0,51,0,234,85,160,0,0,8,8,75,10,8,5,33,174,0,20,136,29,36,254,28,68,129,0,0,4,11,152,76,1,1,0,69,0,30,210,160,0,0,8,8,59,10,5,5,33,13,9,152,76,29,246,254,33,68,129,0,0,4,11,109,223,1,1,0,75,0,145,75,161,0,0,13,8,62,10,8,5,33,32,5,11,5,41,157,8,109,223,29,76,254,28,68,129,0,0,4,11,20,136,1,1,0,48,0,105,222,161,0,0,8,8,76,10,11,5,41,80,0,20,136,29,220,254,28,68,129,0,0,4,11,20,136,1,1,0,48,0,170,223,161,0,0,8,8,75,10,8,5,33,174,0,20,136,29,236,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,29,24,162,0,0,8,8,60,10,11,5,41,46,0,152,76,29,17,254,28,68,129,0,0,4,11,20,136,1,1,0,51,0,83,48,163,0,0,8,8,77,10,5,5,33,242,8,20,136,29,164,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,190,113,163,0,0,8,8,61,10,8,5,33,35,0,152,76,29,221,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,196,25,164,0,0,8,8,62,10,5,5,33,14,9,152,76,29,226,254,28,68,129,0,0,4,11,109,223,1,1,0,78,0,203,52,164,0,0,8,8,63,10,5,5,33,3,9,109,223,29,192,254,28,68,129,0,0,4,11,20,136,1,1,0,51,0,18,37,165,0,0,8,8,78,10,8,5,33,176,0,20,136,29,178,254,5,69,196,204,93,1,181,235,74,254,28,68,129,0,0,0,4,204,93,1,1,0,102,0,224,192,165,0,0,8,24,253,10,0,0,33,195,11,181,235,28,227,254,33,68,129,0,0,4,11,109,223,1,1,0,75,0,87,28,166,0,0,13,8,64,10,8,5,33,32,5,11,5,41,155,8,109,223,29,162,254,28,68,129,0,0,4,11,20,136,1,1,0,51,0,232,180,166,0,0,8,8,78,10,8,5,33,176,0,20,136,29,218,254,28,68,129,0,0,4,11,20,136,1,1,0,51,0,187,181,166,0,0,8,8,79,10,11,5,41,81,0,20,136,29,99,254,5,69,196,216,132,1,181,235,135,254,27,68,129,0,0,1,0,216,132,1,1,0,102,0,108,201,166,0,0,7,24,93,10,32,0,32,31,181,235,28,243,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,33,233,166,0,0,8,8,63,10,11,5,41,47,0,152,76,29,218,254,28,68,129,0,0,4,11,20,136,1,1,0,54,0,218,255,167,0,0,8,8,80,10,5,5,33,242,8,20,136,29,254,254,28,68,129,0,0,4,11,20,136,1,1,0,51,0,21,63,168,0,0,8,8,79,10,11,5,41,81,0,20,136,29,73,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,89,67,168,0,0,8,8,64,10,8,5,33,35,0,152,76,29,126,254,28,68,129,0,0,4,11,152,76,1,1,0,66,0,17,235,168,0,0,8,8,65,10,5,5,33,14,9,152,76,29,182,254,28,68,129,0,0,4,11,109,223,1,1,0,78,0,184,2,169,0,0,8,8,65,10,5,5,33,3,9,109,223,29,246,254,28,68,129,0,0,4,11,20,136,1,1,0,60,0,195,200,169,0,0,8,8,79,10,11,5,41,81,0,20,136,29,102,254,28,68,129,0,0,4,11,152,76,1,1,0,69,0,182,205,169,0,0,8,8,64,10,8,5,33,35,0,152,76,29,25,254,28,68,129,0,0,4,11,20,136,1,1,0,57,0,155,133,170,0,0,8,8,81,10,8,5,33,175,0,20,136,29,158,254,33,68,129,0,0,4,11,109,223,1,1,0,75,0,130,236,170,0,0,13,8,66,10,8,5,33,41,5,11,5,41,153,8,109,2 [... cut ...]
2021-07-25T20:43:25.748Z zigbee-herdsman:controller:endpoint DefaultResponse 0x086bd7fffeee0278/1 2820(10, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
```
## What did you expect to happen
Stable.
## How to reproduce it (minimal and precise)
-
## Debug info
Zigbee2MQTT version:
Adapter hardware: cc2652rb
Adapter firmware version:

Koenkk, 2021-07-26:
- Make sure to update to the latest z2m first (1.20.0).
- If it still happens then provide the herdsman debug log when z2m freezes.

See https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

golddragon007, 2021-07-26:
Ok, I've just installed it as I've fully missed the new repository...

golddragon007, 2021-07-26:
I still have this issue, and it seems over time it gets worse, there's no exact time, i.e. in the logs at 16:00 there was some lag in the system already, and around 14:45 it was pinned at 100%. Also around 16:00, it was already pinning to 100% for a few seconds.
[zigbee-blocked.zip](https://github.com/Koenkk/zigbee2mqtt/files/6880418/zigbee-blocked.zip)
I also noticed that the MQTT seems to drop out?

golddragon007, 2021-07-27:
I don't know how much it relates to these messages, but I have a lot of them in the log file:
```
[31mZigbee2MQTT:error[39m 2021-07-26 16:17:32: Failed to configure 'kitchen_radio_left_outlet', attempt 2 (Error: Bind 0x086bd7fffeee2329/1 genOnOff from '0x00124b0021cc09e4/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))
```
and
```
2021-07-26T14:17:32.492Z zigbee-herdsman:controller:endpoint DefaultResponse 0x086bd7fffeee031d/1 2820(10, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (SREQ '--> AF - dataRequest - {"dstaddr":57197,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":32,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,14,11,10,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
2021-07-26T14:17:32.492Z zigbee-herdsman:controller:device:error Default response to 0x086bd7fffeee031d failed
```

Koenkk, 2021-07-27:
What devices are: `0x086bd7fffeee2c0c`, `0x086bd7fffeee031d`, `0x086bd7fffeee0278` and `0x086bd7fffeedfee4`?

golddragon007, 2021-07-27:
These: https://www.zigbee2mqtt.io/devices/170-33505.html#niko-170-33505
I have currently several that doesn't want to configure, but before I've restored my instance it worked well (at least the configuring part).

Koenkk, 2021-07-28:
They are still spamming the network, you have to play a bit with the change value (I don't know exactly how it relates to the units). You can also configure this from the frontend (so no need to restart z2m everytime to try).

Also the `onEvent` does not seem to be correct since the default responses are still send, try with this  `onEvent`: https://github.com/Koenkk/zigbee-herdsman-converters/blob/9dfecb20e38ea057e5240f51baaaf10a620fd6c7/devices/ewelink.js#L49

golddragon007, 2021-07-28:
Because I continuously trying to adapt that code as half of my system based on zigbee and not funny when it's broken :)
```
Zigbee2MQTT:info  2021-07-28 16:44:15: MQTT publish: topic 'zigbee2mqtt/bedroom1_desk_outlet', payload '{"current":0.52,"linkquality":96,"power":59.97,"state":"ON","voltage":232.1}'
2021-07-28T14:44:15.994Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-07-28T14:44:15.994Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-07-28T14:44:15.994Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-07-28T14:44:15.995Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-07-28T14:44:15.995Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-28T14:44:16.047Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,61,251]
2021-07-28T14:44:16.047Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,61,251]
2021-07-28T14:44:16.047Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,61] - 251
2021-07-28T14:44:16.048Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":61}
2021-07-28T14:44:16.049Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-28T14:44:17.002Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,4,11,182,34,1,1,0,102,0,201,146,129,0,0,8,8,16,10,8,5,33,218,1,182,34,29,154]
2021-07-28T14:44:17.003Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,4,11,182,34,1,1,0,102,0,201,146,129,0,0,8,8,16,10,8,5,33,218,1,182,34,29,154]
2021-07-28T14:44:17.003Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,4,11,182,34,1,1,0,102,0,201,146,129,0,0,8,8,16,10,8,5,33,218,1,182,34,29] - 154
2021-07-28T14:44:17.005Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":2820,"srcaddr":8886,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":102,"securityuse":0,"timestamp":8491721,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[8,16,10,8,5,33,218,1]}}
2021-07-28T14:44:17.009Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":16,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1288,"dataType":33,"attrData":474}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":8886,"endpoint":1,"linkquality":102,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}'
2021-07-28T14:44:17.013Z zigbee-herdsman:controller:endpoint DefaultResponse 0x14b457fffe21e263/1 2820(10, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-07-28T14:44:17.014Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x14b457fffe21e263:8886/1 (0,0,1)
2021-07-28T14:44:17.015Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":8886,"destendpoint":1,"srcendpoint":1,"clusterid":2820,"transid":62,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,16,11,10,0]}}
2021-07-28T14:44:17.016Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,182,34,1,1,4,11,62,0,30,5,24,16,11,10,0,157]
2021-07-28T14:44:17.017Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```
This is one of the current longs, I see there what I've set, but still, there's a lot of devices that don't connect to the network (blinking red). Also, do you know why this can happen? because the USB 2 port doesn't have enough bandwidth or it's just can't serve this amount of message with this chip?

Koenkk, 2021-07-28:
Bandwith of de usb port is definitely not the issue (zigbee is 250kbps). Because of this spamming the whole network can become congested, it's not only about the coordinator but also other devices on the network.

First reduce the spamming of the outlets by changing the rep change. If you click apply here the values are overwritten until the `configure()` is executed again. `configure()` is executed again when the content of the `configure()` method changes or when you click the yellow reconfigure button in the frontend.

golddragon007, 2021-07-31:
It seems to me that this Zigbee network is sometimes really strange. I had to switch off the main switch because some of the devices did not want to join back to the coordinator, now it seems I have a stable network with the settings from that PR for 22 Niko devices. Another strange thing is that now I can't reach the Zigbee2MQTT UI from the home assistant as I getting "502 gateway error" message...

Koenkk, 2021-07-31:
@golddragon007 can you reach the frontend via: `http://ipaddress:8080`? If yes then it's looks to be a configuration error on your side.

golddragon007, 2021-07-31:
After I've restarted the whole device it's started to work again... HA or z2m restart wasn't enough for it.

Koenkk, 2021-07-31:
Great, can this issue be closed?

closed 2021-08-03 by golddragon007