Koenkk/zigbee2mqtt #8131: Allow RGB controllers to be used as a multi-channel light device

cmiguelcabral, 2021-07-25:
## Is your feature request related to a problem? Please describe
No

## Describe the solution you'd like
I would like to have an option like Tasmota's SetOption68 (Multi-channel PWM instead of a single light).

The idea is to have an RGB controller that instead of exposing a single RGB light entity, exposes three different PWM light entities on Home Assistant to be used as three independent lights.
This can also be extended for RGB-CCT controllers that could expose 4 or 5 entities depending on the number of channels they have.

## Describe alternatives you've considered
Right now I'm working around setting only the channels RGB in a way that I can control each light. but that requires me to know the level of each other channels before setting the one I want.

## Additional context
[image]
[image]

Koenkk, 2021-07-26:
This is not possible with Zigbee since we don't have access to the low-level hardware. A Zigbee command just contains the color to set, the device itself is then responsible for interpreting this.

cmiguelcabral, 2021-07-26:
Possible it is...it depends the way you want to do it.
You could do this on zigbee2mqtt side. Expose 3 lights to the user, but control the light like an RGB light. Using zigbee2mqtt as an abstraction layer between the 3 lights on HA and the rgb light on the hardware.

cmiguelcabral, 2021-07-26:
In a way, if the user increases or decreases the brightness of one light, z2m, would increase/decrease the corresponding color channel.

Koenkk, 2021-07-26:
> z2m, would increase/decrease the corresponding color channel.

That's something that should be possible indeed (but has quite an impact on the code)

cmiguelcabral, 2021-07-26:
Yeah, maybe.
I checked out the project yesterday. I want to contribute to this feature.
But first I need to get acquainted with the code, so it will take some days before I have an idea how to implement it.

cmiguelcabral, 2021-07-26:
...how to implement it, and if it is doable...

Koenkk, 2021-07-26:
The HA discovery of lights is done here: https://github.com/Koenkk/zigbee2mqtt/blob/41b67fdd07792a6c6569341d980ec60c1456c2d7/lib/extension/homeassistant.js#L81

All bulbs can already be controlled with rgb, e.g. `zigbee2mqtt/my_bulb/set` payload `{"color": {"r": 200, "g": 100, "b": 50}}`, if you discover each channel as a seprate light it should be doable.

cmiguelcabral, 2021-07-26:
Thank's for pointing me the way.
I'll do some trials during this week, I hope I can come up with something acceptable for a PR :-)

wizu, 2021-08-17:
@cmiguelcabral any news on that? Could you share your HA workaround?

cmiguelcabral, 2021-08-17:
> @cmiguelcabral any news on that? Could you share your HA workaround?

Hi, wizu
Unfortunately, I did not work on this as I was having some stability problems using zigbee2mqtt so I swapped to ZHA.
Right now, I cannot afford to have the lights not turning on/off and the shutters also not responding, as I risk the wife cutting my head off... :-D
In the first week of September, I'll be on vacation and I'll work on this as then I can afford to have those stability problems and try to troubleshoot them.

closed 2021-09-25 by github-actions[bot]