Koenkk/zigbee2mqtt #8132: move stick to other device. Same pan_id and network_key, but still not (all) devices are recognized

bartplessers, 2021-07-25:
Hi,

I wanted to move my zzh-stick to another server.
I'm running zigbee2mqtt + hass in docker. Latest firmware and versions.

Procedure
- stopped zigbee2mqtt addon on SERVER01
- copied zigbee2mqtt config from SERVER01 to SERVER02
- copied whole directory `\\SERVER01\usr\share\hassio\share\zigbee2mqtt` to `\\SERVER02\usr\share\hassio\share\zigbee2mqtt`
- contents of this directory:
  - configuration.yaml
  - coordinator_backup.json
  - database.db
  - devices.yaml
  - groups.yaml
  - state.json
- unplugged stick from SERVER01
- plugged stick into SERVER02
- started zigbee2mqtt addon on SERVER02

and waited a while.

However 
- most of end-devices are connected to the zigbee network and I can see their MQTT messages
- but I noticed that most of my bulbs where not responsive
- seems that are some errors in the log (attached)
  - Data request failed with error: 'MAC channel access failure' (225)
  - Data request failed with error: 'No network route' (205)
  - ...

According to https://www.zigbee2mqtt.io/information/FAQ.html#doesnt-require-repairing, this should be a supported scenario

For now: I reverted the scenario: 
- SERVER02: stopped zigbee2mqtt
- unplugged stick on SERVER02
- plugged stick on SERVER01
- started zigbee23qtt on SERVER01
everythings works fine again.

what's happening here?
Why did the coordinator only recognizes end devices and no routers on the zigbee network?
How can I move my stick to another device without the need to repair all my devices again?

thanx in advance,
Bart

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/6874663/log.txt)

Koenkk, 2021-07-26:
Looks that the move went well, however: 
- `MAC channel access failure' (225)` indicates that the adapter cannot broadcast commands due to intereference
  - Make sure the adapter is positioned at least +-30 cm from the computer
  - Make sure there is no wifi interference: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html
- `No network route' (205)` when you remove the adapter from power, it loses its routes (since those are stored in the RAM). It will need to rebuild those, this happens automatically when sending a command to a device, however, I expect that due to the above errors commands fail to send and no routes are rebuilt. (so `MAC channel access failure' (225)` appears to be the root issue here).

bartplessers, 2021-07-26:
ok, I will try to redo it this evening.

> Make sure the adapter is positioned at least +-30 cm from the computer
it's connected by use of an extension cable of 2m

> Make sure there is no wifi interference: https://www.zigbee2mqtt.io/how_tos/how_to_improve_network_range_and_stability.html
This should be the best as possible. However, my access point was +/- 1m from zigbee stick.

The only thing I can do is moving the stick a little further. Will see.

thanx for response.
grtz
B

closed 2021-09-02 by github-actions[bot]