Koenkk/zigbee2mqtt #8134: Feibit Universal IR remote

Faustino-ai, 2021-07-26:
I tried to follow the guide to support new devices, however the guide provides little to no information on how to add support with devices that require `toZigbee` implementation. Anyhow, I tried to follow the other IR remote (HS2IRC) that is already supported on z2m (although its broken after version 1.17.1), I realized there are some distinct differences between these two devices in the way they implement this. What I've done so far is to setup a CC2531 to sniff the zigbee traffic when I use the official app as well as the gateway to control it. Attached is the picture of the sniffed traffic when I use the app the control the TV. The button is the power on/off button. 
[image]

The `Write Attribute` part really made it very different as compared to the HS2IRC which uses a manufacturer specific cluster. It seems like Feibit uses the `Write Attribute` to send the IR control commands.

I've also attached the API document that I received from the manufacturer. Please help me with this. Thank you.
[IR blaster APIinterface V1.00.pdf](https://github.com/Koenkk/zigbee2mqtt/files/6876724/IR.blaster.APIinterface.V1.00.pdf)

## Information about the device + link
.. 
Feibit IR blaster
http://www.feibit.com/en/zhinengwangguanxilie/products-560.html
## data/database.db entry of the device
..
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d000209fb5b","nwkAddr":10215,"manufId":4151,"manufName":"Feibit","powerSource":"Battery","modelId":"FNB56-ZIR04FB1.3","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"manufacturerName":"Feibit","modelId":"FNB56-ZIR04FB1.3","powerSource":3,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20181224","swBuildId":"1000-0001"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20181224","swBuildId":"1000-0001","zclVersion":2,"interviewCompleted":false,"meta":{},"lastSeen":1627284087946}

Faustino-ai, 2021-08-06:
@Koenkk Can you show me where should I change for the ZCL (write attribute)? Thank you.

Koenkk, 2021-08-06:
You can find an example here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/6a5a29f65204a0b739860513ccd07b5b2698e5be/converters/toZigbee.js#L1757

Faustino-ai, 2021-08-07:
Ok I will try to figure out the rest. Thanks for the tip

Faustino-ai, 2021-08-11:
> You can find an example here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/6a5a29f65204a0b739860513ccd07b5b2698e5be/converters/toZigbee.js#L1757

I tried to refer to the example and managed to get pretty close (I think). However it is still not working. Attached is the sniffed data that I sent out using z2m. Compared to the original one, I saw that the frame control field from z2m is `0x10` where the original one is `0x00` . Also, the string from the original is very different from what I sent from z2m. I hard coded the string for now to test if it works (unfortunately it doesn't). Can you give me another direction to continue?
[image]

Faustino-ai, 2021-08-16:
Ok I've managed to control the IR remote by hard coding the payload. Now the issue is how do I set make a change such that the value is taken from the mqtt json payload? I tried to find an example but I couldn't find anything with `write attribute` . This is what I've done so far. 
https://pastebin.com/nUcj8DYb

It works but obviously due to the hardcoded part, I can only emulate one button press. My question is how do I make my `deviceType` and  `keyStroke` as shown in the code to be taking inputs from the mqtt publish json?

Koenkk, 2021-08-16:
You can do something like:

```js
let deviceType = value.type;
let keyStroke = value.key;
```

and send to `zigbee2mqtt/my_device/set` with payload `{"send_command": {"key": 3, "type": 2}}`

Faustino-ai, 2021-08-17:
> You can do something like:
> 
> ```js
> let deviceType = value.type;
> let keyStroke = value.key;
> ```
> 
> and send to `zigbee2mqtt/my_device/set` with payload `{"send_command": {"key": 3, "type": 2}}`

Thank you so much! That did the trick!!

closed 2021-08-17 by Faustino-ai