Koenkk/zigbee2mqtt #8135: Error: All table entries need to be the same length

0sorkon, 2021-07-26:
## What happened
Raspberry pi 3b+cc2652p . Official image on microSD. Zigbee22mqll installed according to instruction on https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html
after 'npm start' i receive error "All table entries need to be the same length", pastebin of debug output below:
https://pastebin.com/38q4zQwD
configuration.yaml:
https://pastebin.com/8YWFAN68
It is not problem with power supply, tested on different and without "Under-voltage detected!" 
## What did you expect to happen
Everything must be ok, considering that this stick working well on x86 and ubuntu + zigbee2mqtt + majordomo
## How to reproduce it (minimal and precise)
Take RPI 3b+cc2652, Install Zigbee2mqll 
## Debug info
Log debug output: https://pastebin.com/38q4zQwD
configuration.yaml: https://pastebin.com/8YWFAN68
Zigbee2MQTT version: zigbee2mqtt@1.20.0
Adapter hardware: CC2652P
Adapter firmware version: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20210319,"transportrev":2},"type":"zStack3x0"}'

Koenkk, 2021-07-26:
@castorw what does this error mean?

castorw, 2021-07-26:
This could point to possible NV memory corruption. This error indicates that entries of various length were entered into table creation function - which does not make sense if proper data were introduced.

@0sorkon:
* Is this a new setup or are you trying to recover a previously working network?
* Did you try to erase the adapter, flash it and try again?
* Before doing the above, could you please provide NV memory dump from the adapter? (use this script: https://github.com/zigpy/zigpy-znp/blob/dev/zigpy_znp/tools/nvram_read.py)

0sorkon, 2021-07-26:
1. Yes. It is fresh install, Nothing else then zigbee2mqtt is working on this rpi
2. No
3. 
[output.log](https://github.com/Koenkk/zigbee2mqtt/files/6878320/output.log)

castorw, 2021-07-26:
@0sorkon Yep, your NV seems to be corrupt indeed.
[image]

Please flash recommended firmware as per zigbee2mqtt website. If you are not able to do so at least try resetting the NVRAM - eg. using this script (https://github.com/zigpy/zigpy-znp/blob/dev/zigpy_znp/tools/nvram_reset.py).

Hope this helps.

0sorkon, 2021-07-26:
But in the same time, i just plugging in stick to x86 pc with ubuntu and older z2m version and everything work well
https://pastebin.com/XYz6gZrq

0sorkon, 2021-07-26:
awesome! After resetting nvram, z2m starting correctly! thank you very much!

castorw, 2021-07-26:
@0sorkon Yes its possible you were able to start it because the previous codebase (prior to my PR https://github.com/Koenkk/zigbee-herdsman/pull/303) didn't perform these checks when starting up. Several processes are now more strict and therefore can provide more reliable execution.

I am glad to hear it worked out okay for you. I believe we can close this now.

closed 2021-07-27 by 0sorkon