Koenkk/zigbee2mqtt #8136: "ZB-L04C" Support, device ist not paired

M0P, 2021-07-26:
# Bug Report
## What happened
Device: https://zigbee.blakadder.com/Zemismart_ZB-L04C.html

After enabling pairing mode the lights on the device flashes and turn off, as if pairing was successfull.

debug logs from zigbee_herdsman_debug: 
`2021-07-26T12:49:59.151Z zigbee-herdsman:deconz:adapter PERMIT_JOIN - 254 seconds
2021-07-26T12:49:59.153Z zigbee-herdsman:deconz:adapter zclFrame to all - zclFrame.payload:
2021-07-26T12:49:59.153Z zigbee-herdsman:deconz:adapter { options: 11, commisioningWindow: 254 }
2021-07-26T12:49:59.154Z zigbee-herdsman:deconz:adapter sendZclFrameToAll - message send
2021-07-26T12:49:59.155Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xfffd EP:242 SeqNr. 127 request id: 20
2021-07-26T12:49:59.156Z zigbee-herdsman:deconz:driver send write parameter request from queue. seqNr: 126 paramId: 33 param: 254
2021-07-26T12:49:59.159Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 20 status: 0
2021-07-26T12:49:59.159Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-07-26T12:49:59.165Z zigbee-herdsman:deconz:frameParser write parameter response - parameter id: 33 - status: 0
2021-07-26T12:49:59.201Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2021-07-26T12:49:59.201Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-07-26T12:49:59.204Z zigbee-herdsman:deconz:driver query aps data confirm
2021-07-26T12:49:59.209Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 128
2021-07-26T12:49:59.211Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xfffd request id: 20 confirm status: 0
2021-07-26T12:49:59.212Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
Zigbee2MQTT:info  2021-07-26 14:49:59: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"wwvtn-3"}'
2021-07-26T12:50:01.421Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2021-07-26T12:50:01.421Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2021-07-26T12:50:01.425Z zigbee-herdsman:deconz:driver query aps data indication
2021-07-26T12:50:01.430Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 129
2021-07-26T12:50:01.433Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 129 srcAddr: 0x292f destAddr: 0x0 profile id: 0x104 cluster id: 0x702 lqi: 255
2021-07-26T12:50:01.434Z zigbee-herdsman:deconz:frameParser response payload: 8,87,10,0,0,37,218,3,0,0,0,0
2021-07-26T12:50:01.447Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":87,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":37,"attrData":[0,986]}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":10543,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2021-07-26T12:50:01.455Z zigbee-herdsman:controller:endpoint DefaultResponse 0x60a423fffeae6716/1 1794(10, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-07-26T12:50:01.456Z zigbee-herdsman:deconz:adapter zclFrame.payload:
2021-07-26T12:50:01.456Z zigbee-herdsman:deconz:adapter { cmdId: 10, statusCode: 0 }
2021-07-26T12:50:01.457Z zigbee-herdsman:deconz:adapter no response expected
2021-07-26T12:50:01.457Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-07-26T12:50:01.467Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x292f EP:1 SeqNr. 130 request id: 21
2021-07-26T12:50:01.471Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 21 status: 0
2021-07-26T12:50:01.472Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-07-26T12:50:01.480Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2021-07-26T12:50:01.480Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-07-26T12:50:01.485Z zigbee-herdsman:deconz:driver query aps data confirm
2021-07-26T12:50:01.490Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 131
2021-07-26T12:50:01.493Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x292f request id: 21 confirm status: 0
2021-07-26T12:50:01.493Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2021-07-26T12:50:01.494Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 87`

## What did you expect to happen
Deivce gets successfully paired

## Debug Info
Zigbee2MQTT version: 1.20.0-1 - official zigbee2mqtt addon for home assistant on home assistant os
Adapter hardware: CONBEE II
Adapter firmware version: deCONZ_ConBeeII_0x266e0700

robertalexa, 2021-08-01:
To make it easier to read, can you wrap the code in 3 ticks each side rather than just 1? This is all on 1 line making it really hard to read

robertalexa, 2021-08-01:
Also can you please check your database.db file and look for this device? If you find it can you please paste it in here? Thanks

M0P, 2021-08-01:
> 
> 
> To make it easier to read, can you wrap the code in 3 ticks each side rather than just 1? This is all on 1 line making it really hard to read

Thanks, didn't knew the syntax. I fixed it.

I've checked the database and couldn't find anything.

Yesterday I had a chat with the guys from deconz regarding this device, and we did some test on another system running the deconz firmware. I got the feedback that they could find a problem within pairing. Although I didn't got any further information, so I don't know which kind of fix is needed, and where/when it'll be implemented (if needed)

M0P, 2021-08-07:
The device was a bit picky about the battery - using fresh batteries solved the pairing problem.

The device now reports as a TS0044 (should this get corrected?) and so far all 12 actions seems to work .

I wonder is there a better way to display this as only one service_action entity? At the moment I can't really see at the homeassistant logbook which button was pressed.

closed 2021-09-15 by github-actions[bot]