Koenkk/zigbee2mqtt #8137: TypeError: this.zigbee.getClients(...).map(...).flat is not a function

froogl, 2021-07-26:
## What happened

When binding or unbinding a cluster through the UI the following error message is logged:

```
Zigbee2MQTT:info  2021-07-26 16:28:16: Successfully unbound cluster 'genLevelCtrl' from 'sw_test_1' to 'g_bj_test'
Zigbee2MQTT:error 2021-07-26 16:28:16: Failed to call 'Bind' 'onMQTTMessage' (TypeError: this.zigbee.getClients(...).map(...).flat is not a function
    at Bind.disableUnnecessaryReportings (/opt/zigbee2mqtt/zigbee2mqtt/lib/extension/bind.js:336:87)
    at Bind.onMQTTMessage (/opt/zigbee2mqtt/zigbee2mqtt/lib/extension/bind.js:261:32))
```

## What did you expect to happen

Binding/unbinding should not cause any errors.

## How to reproduce it (minimal and precise)

Bind or unbind a cluster in the frontend.

## Debug info

Zigbee2MQTT version: 1.20.0
Adapter hardware: CC2652RB
Adapter firmware version: :20210120
Host OS: Debian Buster on Cubietruck
NodeJS Version: v10.24.0

Koenkk, 2021-07-27:
Should be fixed now. 
Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

froogl, 2021-07-27:
I have just cherry-picked your commit to stable and it's working. Thanks for the quick fix. :+1:

closed 2021-07-27 by froogl