Koenkk/zigbee2mqtt #8144: UNSUPPORTABLE_ATTRIBUTE error when configuring Sinope Themostat

3gyptian, 2021-07-27:
Upgraded from 1.18.2-dev to latest Dev branch (1.20-0-dev) and now seeing unsupportable attribute errors for each of my TH1123ZB .. somewhat similar to what happened when upgrading to 1.18-2: (https://github.com/Koenkk/zigbee2mqtt/issues/5441)

Here's what is being logged for each device after the upgrade:

```
info  2021-07-27 00:11:39: MQTT publish: topic 'zigbee2mqtt/dining_thermostat', payload '{"energy":null,"keypad_lockout":null,"linkquality":7,"local_temperature":23.81,"occupied_heating_setpoint":19,"operation":"idle","pi_heating_demand":0,"power":null,"system_mode":"off"}'
info  2021-07-27 00:11:39: Configuring 'dining_thermostat'
error 2021-07-27 00:11:42: Failed to configure 'dining_thermostat', attempt 2 (Error: Read 0x500b9140000108ee/1 seMetering(["multiplier","divisor"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:196:23)
    at Endpoint.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:251:26)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58))
```

I then tried swapping out the TH1123ZB section of sinope.js with what was provided in #5441 (and worked on 1.18-2-dev) but it then log it as an  unsupported device (Router).

Zigbee2MQTT version: 1.20.0-dev
Adapter hardware: Comidox CC2531 Sniffer USB Dongle
Adapter firmware version: CC2531_DEFAULT_20190608

Any advice greatly appreciated!

Koenkk, 2021-07-27:
Should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

3gyptian, 2021-07-28:
Thank you!  That seems to have done the trick.  At least on the HA side things seems to be working ok.   However fyi I'm still seeing the unreportable attribute  error in the log:

```
error 2021-07-28 11:39:24: Failed to configure 'livingroom_thermostat', attempt 1 (Error: ConfigureReporting 0x500b9140000107d8/1 hvacThermostat([{"attribute":"systemMode","minimumReportInterval":1,"maximumReportInterval":0,"reportableChange":null}], {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNREPORTABLE_ATTRIBUTE')
    at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:293:23)
    at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:627:22)
    at Object.thermostatSystemMode (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/lib/reporting.js:116:9)
    at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/sinope.js:35:13)
    at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.js:137:13)
    at Configure.onZigbeeStarted (/opt/zigbee2mqtt/lib/extension/configure.js:105:17)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:408:58))
info  2021-07-28 11:39:24: Configuring 'bedroom_thermostat'
```

Is this anything to be concerned about?

thanks

3gyptian, 2021-07-28:
Sorry .. HA is actually not receiving updates for some of the elements of the thermostat after the update - linkquality, local_temperature and thermostat_power seem to be not coming through.  Along with the error posted above.

Thanks again for your assistance with this.

Koenkk, 2021-07-29:
Error should be fixed now, can you check again?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

3gyptian, 2021-07-30:
YES!  That seems to have fixed the problem.  Thank you for the awesome support for this! 

 I'm still also seeing these 'extra keys not allowed' errors in HA which I'm just starting to look into resolving.  I'm thinking it might be old data in the HA database it's complaining about?   I might purge the DB to see if they go away. 

```
2021-07-29 23:20:18 ERROR (MainThread) [homeassistant.util.logging] Exception in async_discover when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}], 'device': {'identifiers': ['zigbee2mqtt_0x500b9140000108ee'], 'manufacturer': 'Sinope', 'model': 'Zigbee line volt thermostat (TH1123ZB)', 'name': 'dining_thermostat', 'sw_version': 'Zigbee2MQTT 1.20.0-dev'}, 'enabled_by_default': False, 'icon': 'mdi:signal', 'json_attributes_topic': 'zigbee2mqtt/dining_thermostat', 'name': 'dining_thermostat_linkquality', 'state_class': 'measurement', 'state_topic': 'zigbee2mqtt/dining_thermostat', 'unique_id': '0x500b9140000108ee_linkquality_zigbee2mqtt', 'unit_of_measurement': 'lqi', 'value_template': '{{ value_json.linkquality }}', 'platform': 'mqtt'},)
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.8/site-packages/homeassistant/components/mqtt/mixins.py", line 160, in async_discover
    config = schema(discovery_payload)
  File "/srv/homeassistant/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/srv/homeassistant/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/srv/homeassistant/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: extra keys not allowed @ data['state_class']
```

All of my zigbee thermostat seem to be working fine now would appreciate any suggestions on this.  Happy to open another issue on it as well.

thanks again!

Koenkk, 2021-07-30:
Are you running the latest HA version?

3gyptian, 2021-07-30:
I am now!!  Upgraded to 2021.7.4 this morning and all is well. Thanks again man!!!

closed 2021-07-30 by 3gyptian