Koenkk/zigbee2mqtt #8145: `npm run build` during application startup on dev branch

andir, 2021-07-27:
## What happened

I updated my distribution packaging of zigbee2mqtt to the latest master version that includes the new TypeScript code. Everything worked like with all the previous versions except during runtime `zigbee2mqtt` wanted to call `npm run build` which it can't because the directory is not writable by the users the services is running as and also `npm` is not available to the user.

## What did you expect to happen

`zigbee2mqtt` should not fail during startup while trying to compile typescript. That seems like it is targeted towards a `git pull && ./start.js` kind of workflow and to ease migration of existing installations?

## How to reproduce it (minimal and precise)

Download a tarball of the current version in `dev`. `npm install` the dependencies and build the application via `npm run build`. Ensure that `npm` is no longer on your path. Start zigbee2mqtt. It will now trip on the missing `dist/.hash` file and try to do an initial build of the application even thought that isn't required.

## Expected solution

It would be great if the `npm run build` step could be disabled somehow. An environment variable would be great for that.

## Debug info
Zigbee2MQTT version: 35c2ffc0e5e45e819d85cec9ba77a23d76e681f4
Adapter hardware: Conbee II
Adapter firmware version: N/A

Koenkk, 2021-07-27:
Should be fixed now.

andir, 2021-08-02:
Thanks! Works for me.

closed 2021-08-02 by andir