Koenkk/zigbee2mqtt #8146: Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'

danielbrunt57, 2021-07-27:
## What happened
HA version core-2021.7.4 is reporting this warning:
Template variable warning: 'dict object' has no attribute 'click' when rendering '{{ value_json.click }}'
from my Xiaomi Button:
```
Friendly name Button 1
Last seen N/A
Device type EndDevice
Zigbee Model lumi.sensor_switch
Description MiJia wireless switch
Support status supported

IEEE Address 0x00158d000572c90f
Network address 0x7EAC
Manufacturer Xiaomi
Model WXKG01LM
Power Battery 100%
Interview completed True
```

## What did you expect to happen
No warnings.

## How to reproduce it (minimal and precise)
Pair the button with Zigbee2MQTT

## Debug info
Zigbee2MQTT version: 1.20.0-1
Adapter hardware: CC2652R
Adapter firmware version: Z-Stack_3.x.0

danielbrunt57, 2021-07-27:
As per https://community.home-assistant.io/t/help-needed-with-automation-and-json-template-warning/297258/7, I've modified /app/lib/extension/homeassitant.js line 15...
from:
```
value_template: '{{ value_json.click }}',
```
to:
```
value_template: '{{ value_json.click | default}}',
```
and the problem seems to have been fixed.

Koenkk, 2021-07-27:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/7824

closed 2021-07-27 by Koenkk