Koenkk/zigbee2mqtt #8154: Tuya TS0601_air_quality_sensor too many sensor publishes

MaartenVerb, 2021-07-27:
I use the tuya air quality sensor. It is a slightly different, round [version](https://nl.aliexpress.com/item/1005002511043244.html), but the thumbprints are matching. 

(i had some trouble setting it up, thinking it was not supported, but I think I've reversed all the custom configurations and now it is working) "manufName":"_TZE200_yvx5lh6k"

The device is working great, routing and all. But it seems to be spamming my network with updates about it's sensors.
Temperature, humidity, voc, etc etc.

This is not really a problem, but the Home Assistant database is quickly filling up. 

I've tried to find a way to limit the amount of spamming with 'debounce'. It seems to work a bit, but not sufficient. Still about every 3 seconds updated sensor data is published.

I use the zigbee2mqtt on windows.
[image]

MaartenVerb, 2021-07-30:
Hmm. I found out the formaldehyde on the sensor is in home assistant as text instead of a number. 

I guess the whole device is not working/configured well.

Thrasher2020, 2021-08-02:
I am seeing this issue as well (as well as the formaldehyde showing as text)

Koenkk, 2021-08-02:
@savikko @vben could you investigate this? Does it also happen with the original gateway?

savikko, 2021-08-05:
I do not have access to original gateway, but it seems that this round model shouts out too often (I would say something between 1 to 4 times per sec).

When debugging with `DEBUG=zigbee-herdsman*` and grepped with `grep -E 'incomingMsg|publish'` output is:
```
Zigbee2MQTT:info  2021-08-05 15:41:19: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe8293b0', payload '{"co2":362,"formaldehyd":2,"humidity":45.1,"linkquality":30,"temperature":23.9,"voc":8}'
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":48234,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":36,"securityuse":0,"timestamp":10069912,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,118,1,0,2,2,2,0,4,0,0,1,103]}} +60ms
Zigbee2MQTT:info  2021-08-05 15:41:19: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe8293b0', payload '{"co2":359,"formaldehyd":2,"humidity":45.1,"linkquality":36,"temperature":23.9,"voc":8}'
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":48234,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":33,"securityuse":0,"timestamp":10097752,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,119,1,0,0,18,2,0,4,0,0,0,237]}} +411ms
Zigbee2MQTT:info  2021-08-05 15:41:19: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe8293b0', payload '{"co2":359,"formaldehyd":2,"humidity":45.1,"linkquality":33,"temperature":23.7,"voc":8}'
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":48234,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":36,"securityuse":0,"timestamp":10103182,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,120,1,0,0,19,2,0,4,0,0,1,197]}} +51ms
Zigbee2MQTT:info  2021-08-05 15:41:19: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe8293b0', payload '{"co2":359,"formaldehyd":2,"humidity":45.3,"linkquality":36,"temperature":23.7,"voc":8}'
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":48234,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":33,"securityuse":0,"timestamp":10254397,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,121,1,0,0,18,2,0,4,0,0,0,239]}} +2s
Zigbee2MQTT:info  2021-08-05 15:41:22: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe8293b0', payload '{"co2":359,"formaldehyd":2,"humidity":45.3,"linkquality":33,"temperature":23.9,"voc":8}'
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":48234,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":36,"securityuse":0,"timestamp":10260358,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,122,1,0,0,19,2,0,4,0,0,1,194]}} +64ms
Zigbee2MQTT:info  2021-08-05 15:41:22: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe8293b0', payload '{"co2":359,"formaldehyd":2,"humidity":45,"linkquality":36,"temperature":23.9,"voc":8}'
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":48234,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":33,"securityuse":0,"timestamp":10278164,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,123,1,0,1,22,2,0,4,0,0,0,2]}} +247ms
Zigbee2MQTT:info  2021-08-05 15:41:22: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe8293b0', payload '{"co2":359,"formaldehyd":2,"humidity":45,"linkquality":33,"temperature":23.9,"voc":8}'
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":48234,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":36,"securityuse":0,"timestamp":10283726,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,124,1,0,1,21,2,0,4,0,0,0,6]}} +52ms
Zigbee2MQTT:info  2021-08-05 15:41:22: MQTT publish: topic 'zigbee2mqtt/0x0c4314fffe8293b0', payload '{"co2":359,"formaldehyd":2,"humidity":45,"linkquality":36,"temperature":23.9,"voc":6}'
```

So it really seems to just spam the network.

Of course this could be rate-limited from zigbee2mqtt but I think it really is not good thing that this device communicates so often (read: stay away from this device).

GijsvanDulmen, 2021-08-07:
Had the same experience. Probably we need to figure out how to influence the reporting intervals. I don't know how... but if anybody has a suggestion... we could check this out.

MattWestb, 2021-08-08:
Its default response and APS Ack being sent to the device an / or is it repenting the same tsn with every attribute reported ??

closed 2021-09-16 by github-actions[bot]

VikeDragon, 2021-11-19:
+ Same problems with T0601_air_quality_sensor Zigbee! It's spam Zigbee network. Please help reduce transfer data from it.

sgtwilko, 2021-11-28:
What's the best way to work out how to send to this device to slow it down?

vk496, 2021-12-07:
Exactly the same problem here. Maybe starting with limiting MQTT messages from Z2M to HA?

MaartenVerb, 2021-12-07:
I've added this to my config\zigbee2mqtt\devices.yaml;

```
'0x84fd27fffec83098':
  friendly_name: AirQuality woonkamer
  debounce: 1
```

That seems to calm it down enough, my database is not growing too fast anymore.
I use it as router and for constant temperature measuring and turning on/off heating. It works well for that. But Air quality measurement is useless;
[image]

it never goes up or down, just spikes sometimes without reason.

vk496, 2021-12-07:
Hello @MaartenVerb , thanks for sharing. It's exactly what I was looking for!

In the [documentation](https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#common-device-options) don't say explicitly it is seconds, so setting up a value of 10 in `debounce` and **restarting** works like a charm

netweaver1970, 2021-12-22:
If I breath on it, the CO2 shoots up (as expected but not as high as it should be) but also the voc and formaldehyd shoot up, more or less in the same pattern. As human breath can also contain formaldehyd and voc's that's also acceptable. The absolute values are less reliable without a proper calibrated sensor. And I'm sure the CO2 is actually just an eCO2, a calculated value. My CO2 level in my small office is hovering around 365 ppm, rather strange if clean air has 400 ppm these days...  The graphs below are before the debouncing (only installed yesterday).  I do have problems in that room with buring, watery eyes (formaldehyd exposure) when my 3d printers is on. I blame it on insulated panels, using spray glue to attach the insulating layers to the panels. My printer will have to move to the garage or so ...
[image]

uowis, 2022-01-02:
same problem is it possible to set interval min résponse ?

andrejbestuzhev, 2022-01-26:
Can confirm previous solution: setting debounce to 1 (and restart zigbee2mqtt service!)  reduces messages to ~1 per second.

uowis, 2022-02-10:
CO2 Value is completly false : it stay ever under 380, it's impossible in interior, 
This sensor is so cheap.

andrejbestuzhev, 2022-02-10:
@uowis In my case it stays about 364 in usual time, but everytime i smoke nearby... You can see ;)
[image]

MaartenVerb, 2022-02-22:
One last comment, and everybody already now knows this should not be bought, I will add one more reason; one pin was not soldered in my device. That is why co2 etc. was not working
[image]

vk496, 2022-02-22:
I don't know if the problem is with the integration in Z2M or the device itself, but it would be nice to have seen a warning/message to avoid this device in Z2M devices list.

Do you contemplate this @Koenkk? I would definitely not bought it if I saw some kind of warn like that.

andrejbestuzhev, 2022-02-23:
Does anyone know any alternative?

vk496, 2022-02-23:
> Does anyone know any alternative?

I saw some pretty cool in IKEA, but they are not zigbee :(

Sesshoumaru-sama, 2022-07-02:
Thanks, the debounce helped - at least on HAs end.

VikeDragon, 2022-08-05:
But 'debounce' did not solve the problem... Only reduced it a little, but the network is clogged with messages from the sensor.
Maybe there is another solution?

kevincaradant, 2022-10-19:
Hi guys

Any news on this since August? New firmware maybe? 
Mine is in the drawer because it kill 2 times my zigbee network with spamming messages. I'm waiting a good news to use it again

Thanks

savikko, 2022-10-19:
This is something which needs to be solved within the device. I would be surprised if that happens.
Keep your device in the drawer for few years and then throw it away. Probably feels better than throwing it away immediately.

kevincaradant, 2023-02-04:
Hey today I was hesitating to throw it... Reading your last comment, that give me a smile because it's exactly that 😂.
Trash or no trash. My guilty feeling is over now 😁

kevincaradant, 2023-02-04:
I don't have tuya gateway, I want to say thank you if someone could check it :)

Haha I don't like keep things that I don't want / Can't use them anymore. 
I'm waiting a little bit more but if no body can tell me that an update already exist, I will take irrevocable decision :P

VikeDragon, 2023-02-05:
> I don't have tuya gateway, I want to say thank you if someone could check it :)
> 
> Haha I don't like keep things that I don't want / Can't use them anymore. 
> I'm waiting a little bit more but if no body can tell me that an update already exist, I will take irrevocable decision :P

There is a tuya gateway. I checked it again. There are no updates, even in the Taya application, changes are transmitted very often. In z2m, debounce does not help and the device clogs Zigbee networks. The device is worthy of a basket.[image]

[image]

kevincaradant, 2023-02-05:
@VikeDragon 
First thank you to have checked it ;)
Very disapointing of Tuya they didn't fix it because it's very easy to them to fix it ... just less messages by seconds :(

schauveau, 2024-04-14:
I do not own that device but it should not be that hard to create en external converter that limits the number of messages sent by Z2M.  `debounce` is not really useful here because it requires that the device stops sending data for at least a few seconds.   What needs to be done is : store the datapoints in a cache and flush them after a given delay. Probably 10 lines of code.

I could give it a try but I would not be able to test it. Let me know if you are interested.

roumano, 2024-04-14:
> I do not own that device but it should not be that hard to create en external converter that limits the number of messages sent by Z2M. `debounce` is not really useful here because it requires that [...]

Yes, i'm interested and i will be able to test

savikko, 2024-04-14:
> I do not own that device but it should not be that hard to create en external converter that limits the number of messages sent by Z2M. `debounce` is not really useful here because it requires that [...]

That still does not solve the issue that this device will flood your zigbee network. Throw your device away and buy better one.

schauveau, 2024-04-14:
I am quite new to zigbee but I doubt that a few messages every seconds can significantly flood the zigbee network which is supposed to have ` data rates varying from around 20 kbit/s for Sub-1GHz bands to around 250 kbit/s for channels on the 2.4 GHz band range`.  I have an energy meter that is sending 14 datapoints every 3 seconds and everything seems fine.  Most of the problems probably comes from the MQTT side (encoding to json in Z2M, send, receive, decode from json in Home Assistant, ...).

schauveau, 2024-04-14:
I have a first version in a gist [here](https://gist.github.com/schauveau/08fca627b37edf3ffdc5c3d7478b0897)

savikko, 2024-04-14:
> I am quite new to zigbee but I doubt that a few messages every seconds can significantly flood the zigbee network

Yes and no. But still, this sensor is one of the worst I have ever seen in zigbee space, and there are other which perform much better in every thinkable way. I see no reason why this device should be ever used.

roumano, 2025-08-28:
> > > I am quite new to zigbee but I doubt that a few messages every seconds can significantly flood the zigbee network
> > 
> > 
> > Yes and no. But still, this sensor is one of the worst I have ever seen in zigbee space, and there are other which perform much better in every thinkable way. I see no reason why this device should be ever used.
> 
> Do you have any air quality sensor recomendations?

It's not zigbee, but i think the https://apolloautomation.com/products/air-1 is a good choice (based on esphome, openhardware) and fully compatible with home assistant without cloud