Koenkk/zigbee2mqtt #8159: After roughly a week zigbee2mqtt stops working (and needs a restart)

Piscator74, 2021-07-27:
## What happened
After roughly a week zigbee2mqtt stops working working and on stderr I get continuously the following UnhandledPromiseRejectionWarning. The zigbee network stops functioning. As I run it inside a docker container, the docker container keeps running and keeps spamming these error messages on stderr.

```
(node:7) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 4)
(node:7) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 5)
(node:7) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 6)
(node:7) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:37:35)
    at listOnTimeout (internal/timers.js:557:17)
    at processTimers (internal/timers.js:500:7)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled 
```

I think I have now seen this for the 3rd or 4th time and I think it is roughly after a week.
This time I checked and it was about 6 to 7 days. It happened with the current version (1.20.0) but also with previous versions (1.19.1)

I am running version version:
* Zigbee2MQTT version 1.20.0 (commit #41b67fd)
* zigbee-herdsman (0.13.117)
But i had the same problem with previous versions.

## How to reproduce it (minimal and precise)
Happens after roughly a week.

## Debug info
Adapter hardware:: Slaesh's CC2652RB stick 
Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20201026,"transportrev":2},"type":"zStack3x0"}'

Any suggestions how I can find the cause of this problem or is it really time based.
Note that I am running roughly 40 devices on the zigbee network.

Koenkk, 2021-07-28:
Make sure to use the latest firmware. The firmware you are using has a known stability issue after a long uptime. 

https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin

Piscator74, 2021-07-30:
Thanks for the pointer. I thought I already checked the FW version, but apparently not.

algo4711, 2021-08-23:
Facing the same problem, probably my CC2652RB is not working after around 1,5 days "online". Unfortunatelly the debug logging states not an error. Stick is updated to the latest firmware. Zigbee2MQTT: 1.18.1-1, Running Zigbee2MQTT as an addon in Hassio, core-2021.8.8, supervisor-2021.06.8
Re-Attach the Stick help, so I would assume that the Stick had an issue?!?
Any ideas? Is there a change to enable some more logging? Debus is active: zigbee_herdsman_debug: true 
But see only in the logs were the Stick quite working that no messages come through.

closed 2021-09-30 by github-actions[bot]