Koenkk/zigbee2mqtt #8167: Hive SLRb1 standalone - no control

stevetz, 2021-07-28:
## What happened
Paired a Hive SLR1b heating receiver successfully to z2m on HA, created entities etc all OK. I seem to be able to set the `system_mode` OK, but cannot set the `occupied_heating_setpoint` - which goes back to 1C as soon as I try through the z2m frontend or HA or direct through mqtt, I get this in the logs:
```
info MQTT publish: topic 'zigbee2mqtt/boiler', payload '{"linkquality":54,"local_temperature":21,"occupied_heating_setpoint":27,"running_state":"idle","system_mode":"heat"}'
info MQTT publish: topic 'zigbee2mqtt/boiler', payload '{"linkquality":54,"local_temperature":21,"occupied_heating_setpoint":1,"running_state":"idle","system_mode":"heat"}'
```
It looks like it sets the temp i picked (27C here) then immediately sets it back to 1C?

Hoping that setting this, to be above the `local_temperature` when `system_mode` is `heat` (or `auto`?) will change the `running_state` to heat - and flip the relay on the receiver itself.

I'll add that I don't use the hive thermostat - I wanted to skip this extra device and use some tablets and automate it mostly. The status light flashes slowly orange for a few minutes before becoming solid.

## What did you expect to happen
See above, expecting setting heat above current temp would open the relay & turn the heating on!

## How to reproduce it (minimal and precise)
pair slr1b, try to set heating.

## Debug info
Zigbee2MQTT version: 1.20.0_1 HA addon
Adapter hardware: zzh CC2652R
Adapter firmware version: z stack 3.x.x (can't remember exactly which)

stevetz, 2021-07-28:
This issue doesn't appear to be the same as another open one: https://github.com/Koenkk/zigbee2mqtt/issues/7903

stevetz, 2021-07-29:
Additionally, after leaving it on overnight - I think the `local_temperature` Is also not correct - it hasn’t changed at all.

robertalexa, 2021-08-01:
@stevetz have you also paired the actual thermostat remote? e.g. the SLT3 or SLT3b? That is the device that actually handles the local temperature and passes it to the receiver. Then the receiver exposes it.

stevetz, 2021-08-01:
Ah I don’t, I have it on its own…others on the ha community forum have the slr2b working standalone direct with z2m…I had hoped the 1b would also work in this way?

robertalexa, 2021-08-01:
@stevetz The device would work standalone in the sense of you can control the desired temperature, but you would need something to also read your local temperature, and an automation flow that tells the receiver how to behave depending on that.

With regards to the temperature going back to 1C, can you confirm what happens when you change system_mode to heat, give it a minute, and then change the temperature to 28C (just an example)?

PS: i am not sure of the device behaviour without the thermostat, so I am just guessing here in an attempt to help. In parallel i would suggest you go and try to find more information about the standalone setup where you found the source, maybe someone will have some more info too.

robertalexa, 2021-08-03:
I have tested this with my SLR1 and the behaviour is the same. Temperature gets automatically set to 1C and that is when the system_mode is off.

So simply set system_mode to heat, then refresh all the attributes using either get commands or by pressing the blue icons next to them (in frontend) - this should take the value to the last used one while heating. I am not sure however what the behaviour is without the actual thermostat. But try that anyway and then choose the temperature and see what happens. 

PS: I am currently working on making slight improvements to the whole Hive thermostats and receivers range, you can keep an eye on it here https://github.com/Koenkk/zigbee2mqtt/issues/8184

stevetz, 2021-08-03:
Thanks so much for the help @robertalexa.

Now the receiver has started allowing a temperature to be set. I don't know if this was through setting system_mode to heat or just because it stayed on for a long time... I left it overnight and now it seems to let me set the temp through the z2m dashboard or lovelace.

It has also started publishing a few other topics, this is the current state:
```
{
    "linkquality": 90,
    "local_temperature": 21,
    "occupied_heating_setpoint": 25,
    "running_state": "idle",
    "system_mode": "heat",
    "temperature_setpoint_hold": true,
    "temperature_setpoint_hold_duration": 65535
}
```

The local_temperature still reports at 21C - which would tie into you saying this actually comes from the thermostat... So it doesn't appear I can toggle the boiler relay, even when system_mode is heat and when a temp is set (eg 28C)...

I've not had a response on the HA forums yet from the two people who suggest theirs worked without a thermostat.

If the receiver doesn't have a thermometer (I assume it doesn't) - I wonder if I can publish a topic from another sensor in HA? Or simply control the running_state from z2m?

robertalexa, 2021-08-03:
I am currently trying to work through some of the improvements. Let me do this and then i will try some tests at my end.

It might be that multiple commands will need to be sent to the receiver. At the minute my thermostat is very temperamental moaning that it doesn;t have signal, making my life hard.

I shall get back to you with my findings as soon as i can.

stevetz, 2021-08-03:
Thanks so much! Happy to help with anything I can on my end.

closed 2021-09-11 by github-actions[bot]

imnotverygood, 2021-10-30:
@robertalexa I am trying to use the SLR1 standalone, I imagine it would require much the same work as the SLRb1? All I have is a WPT1 so not particularly helpful, however please let me know if I can be of any help at all!

robertalexa, 2021-10-30:
Device is already supported in the latest version of z2m.

Here is a thread where you can find any information you would need:
https://github.com/Koenkk/zigbee2mqtt/issues/8184

Also here is the device documentation explaining all the commands:
https://www.zigbee2mqtt.io/devices/SLR1.html

imnotverygood, 2021-10-30:
I have gone through the documentation and I am able to pair with the SLR1, however as I do not have a hive thermostat the `local_temperature` value is useless to me. 
Similarly to stevetz, I am trying to run this without a hive thermostat and instead use my own. The only two ways I can see this to be possible would be to be able to control the `running_state` directly or be able to update the `local_temperature` value from another source. 

I hope I am making sense

robertalexa, 2021-10-30:
Pair it with your WPT1 and everything works... Without a thermostat connected to the receiver, like the device was built to operate, what you are trying to achieve is impossible, at least to my knowledge. There is no way around it.

imnotverygood, 2021-10-30:
Sadly not possible to pair the WPT1 whilst HA is paired with the SLR1. 

Thank you for clearing this up for me though :)

robertalexa, 2021-10-30:
why not? I am not sure if i a missing some info, but the pairing process is described here.
https://www.jrplumbingandheatingltd.co.uk/faq/how-to-re-pair-your-hive/

There are also others using the WPT1 and as far as I know they were successful.

@steveuk you are using your SLR1 with the WPT1 right?

imnotverygood, 2021-10-31:
The [WPT1](https://www.zigbee2mqtt.io/devices/WPT1.html) does not have support for a hive hub and only supports a direct connection with the SLR1.

From what I have gathered, the SLR1 has two kinds of pairing modes, one that will pair to a hive hub, which is the one required when pairing with HA and another for direct pairing with the older style hive thermostats like the WPT1. 

I hope I am wrong! But I have been bashing my head against this on and off for months.

robertalexa, 2021-10-31:
The WPT will get paired with the SLR1 during the pairing process. The SLR1 will expose the local_temperature from WPT1 to z2m. The remote is just....well useless when it comes to z2m. So it should all be fine

imnotverygood, 2021-10-31:
How would I pair both the WPT1 and z2m to the SLR1? Because I have not found any way of doing this.

robertalexa, 2021-10-31:
https://www.jrplumbingandheatingltd.co.uk/faq/how-to-re-pair-your-hive/

This is the pairing process. It will automatically pair with the SLR1 during the pairing process. Just follow the steps here

imnotverygood, 2021-10-31:
Yes, it's a similar process to pair the WPT1 with the SLR1. 

However as I have been saying only the z2m or WPT1 can be connected to the SLR1 at one time. As the WPT1 doesn't support the hive app or the hive hub, the SLR1 has to be put into the direct connection mode when pairing the WPT1, which will not allow both z2m and the WPT1 to be connected simultaniously.

If in the hive hub mode, I can connect z2m to the SLR1 but the WPT1 will never connect as it does not support that means of connection like the other hive thermostats do. What you are telling me to do and what is mentioned within the zigbee2mqtt device documentation does not work with the WPT1 for these reasons, however will work with the other hive thermostats.

robertalexa, 2021-10-31:
Ohhhh i understand now, I was not aware of that limitation. My apologies and thanks for clarifying.

That being said, the only real solution I can suggest is to try and sell your WPT on ebay and get a different thermostat from ebay as well. Sorry I couldn't be of more help

imnotverygood, 2021-10-31:
Not a problem at all. This is the conclusion I have sadly come to, also.

I'm gonna make one last shot at reverse engineering the WPT1 to try and spoof it's pairing and reporting before giving up however. I will update if I have any kind of success.