Koenkk/zigbee2mqtt #8170: TS021 plug image update

KixMan28, 2021-07-29:
## Is your feature request related to a problem? Please describe
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Hi! Could you please update/change the image for the TS0121 plug. The current device image is barely visible. 

## Describe the solution you'd like
A clear and concise description of what you want to happen.

Any of the following images would be better, in my opinion.

[image]
[image]

## Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

## Additional context
Add any other context or screenshots about the feature request here.

Koenkk, 2021-07-29:
Done

closed 2021-07-29 by Koenkk

KixMan28, 2021-07-29:
Thank you for your quick reply. I appreciate your on this integration.