Koenkk/zigbee2mqtt #8178: tamper shows up as [object Object]

evilpie, 2021-07-29:
## What happened

- `tamper` shows up as `[object Object]` in the "Exposes" tab in the device view

## What did you expect to happen

- Something else like `true` / `false`, similar to `battery_low

## How to reproduce it (minimal and precise)

- Installed Home Assistant add-on
- Went to the Zigbee2MQTT dashboard
- Clicked on my ZS110050078 door/window sensor
- Selected "Exposes"

## Debug info
Zigbee2MQTT version: 1.20.0
Adapter hardware: CC2531
Adapter firmware version: ?

nurikk, 2021-07-30:
Hi!
Can you attach complete state.json file from `Settings -> Tools -> Download state` ?

evilpie, 2021-07-30:
[state.json.zip](https://github.com/Koenkk/zigbee2mqtt/files/6907085/state.json.zip)

nurikk, 2021-07-30:
Fixed, thanks

closed 2021-07-31 by Koenkk