Koenkk/zigbee2mqtt #8180: LED Driver TuYa TS0503B RGB work only color_temp, color_xy got error in logs

shirou93, 2021-07-30:
When i try change color_xy in my LED Driver TuYa TS0503B RGB i have errors in logs.
Color temp working and when i try change color_xy device blink and don't responding some times. Strange because month ago color_xy working.

Zigbee2MQTT version: 1.20.0-1
Adapter hardware: CC2652RB
Adapter firmware version: CC2652RB_coordinator_20210120.zip

'Zigbee2MQTT:error 2021-07-30 13:23:20: Publish 'set' 'color' to '0x60a423fffefbbffd' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":38797,"colory":22479}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 199 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:23:43: Publish 'set' 'color' to '0x60a423fffefbbffd' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":13173,"colory":43712}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 200 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:24:06: Publish 'set' 'color' to '0x60a423fffefbbffd' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":36831,"colory":20316}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 207 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:24:29: Publish 'set' 'color' to '0x60a423fffefbbffd' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":29687,"colory":28311}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 208 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:24:52: Publish 'set' 'color' to '0x60a423fffefbbffd' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":41418,"colory":19988}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 209 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:25:15: Publish 'set' 'color' to '0x60a423fffefbbffd' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":37683,"colory":18022}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 210 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:25:38: Publish 'set' 'color' to '0x60a423fffefbbffd' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":32178,"colory":25559}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 211 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:26:01: Publish 'set' 'color' to '0x60a423fffefbbffd' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":41418,"colory":19988}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 212 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:26:24: Publish 'set' 'color' to '0x60a423fffefbbffd' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":32178,"colory":25559}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 213 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:30:22: Publish 'set' 'color' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":13173,"colory":43712}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 4 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:30:47: Publish 'set' 'color' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":32178,"colory":25559}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 7 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:32:20: Publish 'set' 'color' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":9896,"colory":22479}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 105 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:32:43: Publish 'set' 'color' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":12648,"colory":38862}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 106 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:33:06: Publish 'set' 'color' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":19071,"colory":41484}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 107 - 768 - 11 after 10000ms)'
Zigbee2MQTT:error 2021-07-30 13:33:29: Publish 'set' 'color' to 'Kuchnia LED' failed: 'Error: Command 0x60a423fffefbbffd/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":38797,"colory":22479}, {"sendWhenActive":false,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 16031 - 1 - 108 - 768 - 11 after 10000ms)''

shirou93, 2021-07-30:
@sjorge  can you check this? Thx :)

sjorge, 2021-07-30:
Looks like it's all timeouts, so either the device doesn't support it and is just ignore it or the requests are not reaching the device.

shirou93, 2021-07-30:
i try different colors and this colors working.
[image]

sjorge, 2021-07-31:
Given the log only shows timeouts there is not really much to go on, as far as z2m is concerned the device just never responded. Someone who also has the device might be of more help.

shirou93, 2021-07-31:
Today when i try turn on light, device left from network and switch to pair mode. I think the my device is propably broken.

shirou93, 2021-08-05:
My device stop working actual is dead. I close this issue. Thanks for checking.

closed 2021-08-05 by shirou93