Koenkk/zigbee2mqtt #8184: Hive SLR1

steveuk23, 2021-07-30:
Hi @Koenkk @robertalexa 
I'm trying to add a Unsupported device to zigbee2mqtt (i use the edge addon btw)
The device is listed as SLR1 and i was hoping it was basically the same as the SLR1b that is supported.
i found this hive.js and edited it to suit my device and added the below to my zigbee2mqtt configuration.yaml
```
external_converters:
  - hive.js
```
But every time i restart the add on it fails to reload until i delete the line again,
I'm not great with all this so am just guessing as i go along but have read about 
adding it to devices.js but not sure how or where that file is.
My hive.js code is like below

```
const exposes = require('../lib/exposes');
const fz = {...require('../converters/fromZigbee'), legacy: require('../lib/legacy').fromZigbee};
const tz = require('../converters/toZigbee');
const globalStore = require('../lib/store');
const constants = require('../lib/constants');
const reporting = require('../lib/reporting');
const extend = require('../lib/extend');
const e = exposes.presets;
const ea = exposes.access;

module.exports = [
    {
        zigbeeModel: ['SLR1'],
        model: 'SLR1',
        vendor: 'Hive',
        description: 'Heating thermostat',
        fromZigbee: [fz.legacy.thermostat_att_report, fz.legacy.thermostat_weekly_schedule_rsp],
        toZigbee: [tz.thermostat_local_temperature, tz.thermostat_system_mode, tz.thermostat_running_state,
            tz.thermostat_occupied_heating_setpoint, tz.thermostat_control_sequence_of_operation, tz.thermostat_weekly_schedule,
            tz.thermostat_clear_weekly_schedule, tz.thermostat_temperature_setpoint_hold, tz.thermostat_temperature_setpoint_hold_duration],
        exposes: [exposes.climate().withSetpoint('occupied_heating_setpoint', 7, 30, 1).withLocalTemperature()
            .withSystemMode(['off', 'auto', 'heat']).withRunningState(['idle', 'heat']).withPiHeatingDemand()],
        meta: {disableDefaultResponse: true},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(5);
            const binds = ['genBasic', 'genIdentify', 'genAlarms', 'genTime', 'hvacThermostat'];
            await reporting.bind(endpoint, coordinatorEndpoint, binds);
            await reporting.thermostatTemperature(endpoint, {min: 0, max: constants.repInterval.HOUR, change: 1});
            await reporting.thermostatRunningState(endpoint);
            await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
            await reporting.thermostatTemperatureSetpointHold(endpoint);
            await reporting.thermostatTemperatureSetpointHoldDuration(endpoint);
        },
    },
];

```
[image]

robertalexa, 2021-07-30:
@steveuk23 Thanks for emailing me, but as explained to you i much prefer to try and help you here for the benefit of everyone else. Hope you can understand :)

Can you please try the following, as per the documentation:
- in the same location as your configuration.yaml file create a new file called SLR1.js

Paste the following code
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['SLR1'],
    model: 'SLR1',
    vendor: 'Hive',
    description: 'Heating thermostat',
    fromZigbee: [fz.thermostat_att_report, fz.thermostat_weekly_schedule_rsp],
    toZigbee: [tz.thermostat_local_temperature, tz.thermostat_system_mode, tz.thermostat_running_state,
        tz.thermostat_occupied_heating_setpoint, tz.thermostat_control_sequence_of_operation, tz.thermostat_weekly_schedule,
        tz.thermostat_clear_weekly_schedule, tz.thermostat_temperature_setpoint_hold, tz.thermostat_temperature_setpoint_hold_duration],
    exposes: [exposes.climate().withSetpoint('occupied_heating_setpoint', 7, 30, 1).withLocalTemperature()
        .withSystemMode(['off', 'auto', 'heat']).withRunningState(['idle', 'heat']).withPiHeatingDemand()],
    meta: {disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(5);
        const binds = ['genBasic', 'genIdentify', 'genAlarms', 'genTime', 'hvacThermostat'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.thermostatTemperature(endpoint, {min: 0, max: constants.repInterval.HOUR, change: 1});
        await reporting.thermostatRunningState(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatTemperatureSetpointHold(endpoint);
        await reporting.thermostatTemperatureSetpointHoldDuration(endpoint);
    },
};

module.exports = definition;
```
- make sure the log level is set to debug so you can check the logs if needed.
- restart z2m
- try the device.

Let me know how it goes, if things do not work, can you please paste your logs in here.

Rob

steveuk23, 2021-07-30:
> @steveuk23 Thanks for emailing me, but as explained to you i much prefer to try and help you here for the benefit of everyone else. Hope you can understand :)
>
> Can you please try the following, as [...]

No i understand what you mean about doing it through here for others :)
I added the code to a SLR1.js file and placed it in the same folder.
But i'm not 100% sure how i add it as a external convertor so i added it in the configuration section of the add on like below
```
data_path: /share/zigbee2mqtt
external_converters:
  - SLR1.js
```
is that correct ?
But again the add on fails to restart when i press restart this is the error i'm getting in the log

```
Error: Cannot find module 'zigbee-herdsman-converters/lib/extend'
Require stack:
- /app/lib/util/utils.js
- /app/lib/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:880:15)
    at Function.Module._load (internal/modules/cjs/loader.js:725:27)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at /app/lib/util/externally-loaded.js:5:16
    at Script.runInContext (vm.js:143:18)
    at Script.runInNewContext (vm.js:148:17)
    at Object.runInNewContext (vm.js:303:38)
    at loadModuleFromText (/app/lib/util/utils.js:162:8)
    at loadModuleFromFile (/app/lib/util/utils.js:168:12)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:2569) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:2569) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

robertalexa, 2021-07-30:
@steveuk23 Yes the configuration.yaml bit is correct.

Just to clarify:
- if you remove the external converter from your configuration.yaml does it start properly?
- do you have any other zigbee devices and are they working?
- is you z2m instance the HA addon or a separate installation?

steveuk23, 2021-07-30:
@robertalexa  Yes it starts back up no problems.

i have about 30 zigbee devices working no problems.
I'm actually using the Zigbee2mqtt EDGE addon within HA as it supported another device a while ago but i ended up staying on it to save the hassle of moving back to the standard zigbee2mqtt.
I just noticed it says 501 bad gateway where the dashboard should load.

robertalexa, 2021-07-30:
It would say 501 bad gateway while the addon properly starts. Give is a minute and it should load.

From the code i gave you, try removing this line and try again
```
const extend = require('zigbee-herdsman-converters/lib/extend');
```

With regards to the Edge addon, to my knowledge you can keep both of them installed, as long as one of them is stopped. You can always backup you device data before swapping the addons. My personal opinion on this is to stay on the stable release, this was you don't risk things breaking as they are still in development stage (the reason why the stable one gets released once every beginning of a month). And should you need to test things out, either swap to edge (which personally i don't do) or have a completely separate instance away from your "production" HA. Buy a second cheap adapter and either run it straight on you laptop or in a virtual box if you want to run it in HA environment.

Let me know how the above goes
Rob

steveuk23, 2021-07-30:
@robertalexa That did it.
Once i deleted that line it restarted fine and shows as a supported device.
I just need to work out the next step now as i just added a basic climate card which seems to turn the receiver on and off but obviously doen't show any current temp etc so i need to work that out.
But thanks for the help least it seems like i have the device added now to have a mess around.
i'm going to move back to the standard zigbee2mqqt add on i just wasn't sure how i did it lol
Thanks again

steveuk23, 2021-07-30:
@robertalexa 
I may have spoken too soon as i am getting a error in the zigbee2mqtt addon
And when i turn on the heat through the linked thermostat the heat led turns on the receiver but the climate card doesn't update and vise versa when i adjust the climate card it doesn't affect the receiver it stays as it is
```
Zigbee2MQTT:error 2021-07-30 18:42:21: Failed to configure 'Hive receiver', attempt 2 (ReferenceError: constants is not defined
    at Object.configure (/app/lib/util/externally-loaded.js:24:71)
    at async Configure.configure (/app/lib/extension/configure.js:131:13))
```

robertalexa, 2021-07-30:
@steveuk23 hey. I am currently not at home anymore to look at code but for now i would suggest you stay away from ha.

While testing stick with the z2m frontend and run all the commands from there. Click on your device then exposes and chekc what is there.

In a different tab open up your logs and keep an eye on them.

Start running commands from physical thermostat and see what the logs output and also if the frontend reflecta the change.

If you get an error make a note of the action you did and then restart the addon. There is a chance that once you get an error it could be a fatal one and nothing will work after it. So always restart it between an error and trying different commands.

Also open the documentation and have a look at slr1b and what commands you can send via mqtt, you can try sending thise commands yourself and aee what you get.

You have a go at that and document as much as possible, keep everything posted here and i will contribute when i get around my computer.

If we nail this down i can submit a pull request and get this device supported for everyone

steveuk23, 2021-07-30:
> @steveuk23 hey. I am currently not at home anymore to look at code but for now i would suggest you stay away from ha.
>
> While testing stick with the z2m frontend and run all the commands from there. [...]

Thanks for that.
I've run out of time now but will try that tomorrow and see what shows up

steveuk23, 2021-07-31:
Hi @robertalexa 
I've been testing a few buttons on the Z2M frontend like you said.
i have 5 options available:

- occupied_heating_setpoint         - This seems to work ok the heat turns on and off when the sliders adjusted

- local_temperature                       - This just displays N/A

- running_state                              - This just displays N/A

- pi_heating_demand                    - This just displays N/A

- system_mode       **heat**         - Set to heat gives no errors but doesn't seem to do anything
```
Zigbee2MQTT:debug 2021-07-31 10:36:16: Received MQTT message on 'zigbee2mqtt/Hive receiver/set' with data '{"system_mode":"heat"}'
Zigbee2MQTT:debug 2021-07-31 10:36:16: Publishing 'set' 'system_mode' to 'Hive receiver'
Zigbee2MQTT:info  2021-07-31 10:36:16: MQTT publish: topic 'zigbee2mqtt/Hive receiver', payload '{"linkquality":81,"system_mode":"heat"}'

```
- system_mode     **auto**        - Same no errors but cant see anything happen
```
Zigbee2MQTT:debug 2021-07-31 10:41:24: Received MQTT message on 'zigbee2mqtt/Hive receiver/set' with data '{"system_mode":"auto"}'
Zigbee2MQTT:debug 2021-07-31 10:41:24: Publishing 'set' 'system_mode' to 'Hive receiver'
Zigbee2MQTT:info  2021-07-31 10:41:24: MQTT publish: topic 'zigbee2mqtt/Hive receiver', payload '{"linkquality":78,"system_mode":"auto"}'
```
- system_mode     **off**        - Same no errors but cant see anything happen
```
Zigbee2MQTT:debug 2021-07-31 10:44:04: Received MQTT message on 'zigbee2mqtt/Hive receiver/set' with data '{"system_mode":"off"}'
Zigbee2MQTT:debug 2021-07-31 10:44:04: Publishing 'set' 'system_mode' to 'Hive receiver'
Zigbee2MQTT:info  2021-07-31 10:44:04: MQTT publish: topic 'zigbee2mqtt/Hive receiver', payload '{"linkquality":28,"system_mode":"off"}'

```
and then on the refresh icon next to each option i get this error
```
Zigbee2MQTT:error 2021-07-31 10:45:30: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /app/lib/extension/receive.js:117:42
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/app/lib/extension/receive.js:116:65)
    at Controller.callExtensionMethod (/app/lib/controller.js:380:44))

```
And lastly i'm getting this error randomly appearing in the log
```
Zigbee2MQTT:error 2021-07-30 18:42:21: Failed to configure 'Hive receiver', attempt 3 (ReferenceError: constants is not defined
    at Object.configure (/app/lib/util/externally-loaded.js:24:71)
    at async Configure.configure (/app/lib/extension/configure.js:131:13))

```
Hopefully you can make sense of this as it's beyond me 
Thanks
[image]

robertalexa, 2021-07-31:
Hi @steveuk23 

Did you all the files you have manully modified on your own? What I am trying to say - apart from the external converter, are you sure there is no other code that you have either added or removed while trying to support this device on your own?

Rob

steveuk23, 2021-07-31:
@robertalexa 
The only thing I modified was the default name of the receiver in the devices.yaml in the Z2M folder to "Hive Receiver" 
Everything else I did was just modifying the js file of the convertor.

robertalexa, 2021-07-31:
Could you please go here https://www.zigbee2mqtt.io/information/debug.html
And follow the instructions under: Home Assistant OS/Supervised addon

That will turn on the debuging for zigbee herdsman and hopefully will give us some more clues.

Could you please have another play like you did above and post back logs, making sure they are long enough to include the command and everything that was written as a result of it. (hopefully that makes sense)

Please do so for system_mode, the little refresh icons (might be worth you spliting these logs by each command in your reply so it is easy to follow).

Also, should you see the "Failed to configure" error come up again, try to paste it here again with more logging.

I had a look at other zigbee software solutions and it seems that no one has the SLT1 receiver supported, now i do not know if it is because the device is "basic" or no one actually worked with it. Also Hive is not publishing any specs for their devices so the only solution is to sniff the traffic between the receiver and the hub to learn exactly what is what.

But lets see where that takes us, maybe we can get to the bottom of it, maybe not.

@Koenkk and @sjorge any thoughts, you guys are a hell lot more familiar with the codebase than i am.

steveuk23, 2021-07-31:
@robertalexa 
Thanks again i've enabled that logging and have it copied down i'll post it over a couple of posts.

steveuk23, 2021-07-31:
**occupied_heating setpoint refresh**
```
Zigbee2MQTT:debug 2021-07-31 18:59:55: Received MQTT message on 'zigbee2mqtt/Hive receiver/get' with data '{"occupied_heating_setpoint":""}'
Zigbee2MQTT:debug 2021-07-31 18:59:55: Publishing get 'get' 'occupied_heating_setpoint' to 'Hive receiver'
2021-07-31T17:59:55.256Z zigbee-herdsman:controller:endpoint Read 0x001e5e0902193948/5 hvacThermostat(["occupiedHeatingSetpoint"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-07-31T17:59:55.257Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001e5e0902193948:62986/5 (0,0,1)
2021-07-31T17:59:55.258Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":62986,"destendpoint":5,"srcendpoint":1,"clusterid":513,"transid":53,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,12,0,18,0]}}
2021-07-31T17:59:55.259Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,10,246,5,1,1,2,53,0,30,5,16,12,0,18,0,241]
2021-07-31T17:59:55.270Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-07-31T17:59:55.270Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-07-31T17:59:55.270Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-07-31T17:59:55.270Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-07-31T17:59:55.270Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-31T17:59:55.275Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,53,243]
2021-07-31T17:59:55.275Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,53,243]
2021-07-31T17:59:55.276Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,53] - 243
2021-07-31T17:59:55.276Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":53}
2021-07-31T17:59:55.276Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-31T17:59:55.292Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,10,246,1,217,0,160]
2021-07-31T17:59:55.292Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,10,246,1,217,0,160]
2021-07-31T17:59:55.293Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [10,246,1,217,0] - 160
2021-07-31T17:59:55.293Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":62986,"relaycount":1,"relaylist":[217]}
2021-07-31T17:59:55.293Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-31T17:59:55.348Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,29,68,129,0,0,1,2,10,246,5,1,0,84,0,77,144,20,0,0,9,8,12,1,18,0,0,41,208,7,217,0,28,155]
2021-07-31T17:59:55.348Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,29,68,129,0,0,1,2,10,246,5,1,0,84,0,77,144,20,0,0,9,8,12,1,18,0,0,41,208,7,217,0,28,155]
2021-07-31T17:59:55.348Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 29 - 2 - 4 - 129 - [0,0,1,2,10,246,5,1,0,84,0,77,144,20,0,0,9,8,12,1,18,0,0,41,208,7,217,0,28] - 155
2021-07-31T17:59:55.349Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":513,"srcaddr":62986,"srcendpoint":5,"dstendpoint":1,"wasbroadcast":0,"linkquality":84,"securityuse":0,"timestamp":1347661,"transseqnumber":0,"len":9,"data":{"type":"Buffer","data":[8,12,1,18,0,0,41,208,7]}}
2021-07-31T17:59:55.352Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":12,"manufacturerCode":null,"commandIdentifier":1},"Payload":[{"attrId":18,"status":0,"dataType":41,"attrData":2000}],"Command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}},"address":62986,"endpoint":5,"linkquality":84,"groupID":0}'
Zigbee2MQTT:debug 2021-07-31 18:59:55: Received Zigbee message from 'Hive receiver', type 'readResponse', cluster 'hvacThermostat', data '{"occupiedHeatingSetpoint":2000}' from endpoint 5 with groupID 0
2021-07-31T17:59:55.358Z zigbee-herdsman:controller:endpoint DefaultResponse 0x001e5e0902193948/5 513(1, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-07-31T17:59:55.359Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
Zigbee2MQTT:error 2021-07-31 18:59:55: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /app/lib/extension/receive.js:117:42
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/app/lib/extension/receive.js:116:65)
    at Controller.callExtensionMethod (/app/lib/controller.js:380:44))
2021-07-31T17:59:55.363Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001e5e0902193948:62986/5 (0,0,1)
[... cut ...]
2021-07-31T18:10:16.824Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,90,156]
2021-07-31T18:10:16.824Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,90,156]
2021-07-31T18:10:16.824Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,90] - 156
2021-07-31T18:10:16.825Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":90}
2021-07-31T18:10:16.825Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

steveuk23, 2021-07-31:
**system_mode TURNED OFF**
```
Zigbee2MQTT:debug 2021-07-31 19:11:23: Received MQTT message on 'zigbee2mqtt/Hive receiver/set' with data '{"system_mode":"off"}'
Zigbee2MQTT:debug 2021-07-31 19:11:23: Publishing 'set' 'system_mode' to 'Hive receiver'
2021-07-31T18:11:23.932Z zigbee-herdsman:controller:endpoint Write 0x001e5e0902193948/5 hvacThermostat({"systemMode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-07-31T18:11:23.933Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001e5e0902193948:62986/5 (0,0,1)
2021-07-31T18:11:23.934Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":62986,"destendpoint":5,"srcendpoint":1,"clusterid":513,"transid":93,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[16,22,2,28,0,48,0]}}
2021-07-31T18:11:23.935Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,10,246,5,1,1,2,93,0,30,7,16,22,2,28,0,48,0,163]
2021-07-31T18:11:23.946Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-07-31T18:11:23.946Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-07-31T18:11:23.946Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-07-31T18:11:23.946Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-07-31T18:11:23.947Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-31T18:11:23.953Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,93,155]
2021-07-31T18:11:23.953Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,93,155]
2021-07-31T18:11:23.953Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,93] - 155
2021-07-31T18:11:23.954Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":93}
2021-07-31T18:11:23.954Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-31T18:11:23.972Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,10,246,1,217,0,160]
2021-07-31T18:11:23.972Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,10,246,1,217,0,160]
2021-07-31T18:11:23.973Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [10,246,1,217,0] - 160
2021-07-31T18:11:23.973Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":62986,"relaycount":1,"relaylist":[217]}
2021-07-31T18:11:23.974Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-31T18:11:24.025Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,68,129,0,0,1,2,10,246,5,1,0,86,0,73,103,53,0,0,4,8,22,4,0,217,0,28,176]
2021-07-31T18:11:24.026Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,1,2,10,246,5,1,0,86,0,73,103,53,0,0,4,8,22,4,0,217,0,28,176]
2021-07-31T18:11:24.026Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,1,2,10,246,5,1,0,86,0,73,103,53,0,0,4,8,22,4,0,217,0,28] - 176
2021-07-31T18:11:24.027Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":513,"srcaddr":62986,"srcendpoint":5,"dstendpoint":1,"wasbroadcast":0,"linkquality":86,"securityuse":0,"timestamp":3499849,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[8,22,4,0]}}
2021-07-31T18:11:24.029Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":22,"manufacturerCode":null,"commandIdentifier":4},"Payload":[{"status":0}],"Command":{"ID":4,"name":"writeRsp","parameters":[{"name":"status","type":32},{"name":"attrId","type":33,"conditions":[{"type":"statusNotEquals","value":0}]}]}},"address":62986,"endpoint":5,"linkquality":86,"groupID":0}'
2021-07-31T18:11:24.031Z zigbee-herdsman:controller:endpoint DefaultResponse 0x001e5e0902193948/5 513(4, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-07-31T18:11:24.032Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-31T18:11:24.032Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001e5e0902193948:62986/5 (0,0,1)
2021-07-31T18:11:24.034Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":62986,"destendpoint":5,"srcendpoint":1,"clusterid":513,"transid":94,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,22,11,4,0]}}
2021-07-31T18:11:24.034Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,10,246,5,1,1,2,94,0,30,5,24,22,11,4,0,149]
Zigbee2MQTT:info  2021-07-31 19:11:24: MQTT publish: topic 'zigbee2mqtt/Hive receiver', payload '{"linkquality":86,"system_mode":"off"}'
2021-07-31T18:11:24.046Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-07-31T18:11:24.046Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-07-31T18:11:24.046Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-07-31T18:11:24.047Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
[... cut ...]
2021-07-31T18:14:07.664Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,103,161]
2021-07-31T18:14:07.664Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,103,161]
2021-07-31T18:14:07.664Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,103] - 161
2021-07-31T18:14:07.665Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":103}
2021-07-31T18:14:07.665Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
```

steveuk23, 2021-07-31:
This is the only thing i can see that works
**occupied_heating_setpoint slider to to max 30**
```
Zigbee2MQTT:debug 2021-07-31 19:14:58: Received MQTT message on 'zigbee2mqtt/Hive receiver/set' with data '{"occupied_heating_setpoint":30}'
Zigbee2MQTT:debug 2021-07-31 19:14:59: Publishing 'set' 'occupied_heating_setpoint' to 'Hive receiver'
2021-07-31T18:14:59.012Z zigbee-herdsman:controller:endpoint Write 0x001e5e0902193948/5 hvacThermostat({"occupiedHeatingSetpoint":3000}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-07-31T18:14:59.014Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001e5e0902193948:62986/5 (0,0,1)
2021-07-31T18:14:59.015Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":62986,"destendpoint":5,"srcendpoint":1,"clusterid":513,"transid":104,"options":0,"radius":30,"len":8,"data":{"type":"Buffer","data":[16,27,2,18,0,41,184,11]}}
2021-07-31T18:14:59.016Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,18,36,1,10,246,5,1,1,2,104,0,30,8,16,27,2,18,0,41,184,11,51]
2021-07-31T18:14:59.026Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-07-31T18:14:59.027Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-07-31T18:14:59.027Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-07-31T18:14:59.027Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2021-07-31T18:14:59.027Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-31T18:14:59.032Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,104,174]
2021-07-31T18:14:59.033Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,104,174]
2021-07-31T18:14:59.033Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,104] - 174
2021-07-31T18:14:59.033Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":104}
2021-07-31T18:14:59.033Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-31T18:14:59.050Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,10,246,0,126]
2021-07-31T18:14:59.050Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,10,246,0,126]
2021-07-31T18:14:59.050Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [10,246,0] - 126
2021-07-31T18:14:59.052Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":62986,"relaycount":0,"relaylist":[]}
2021-07-31T18:14:59.052Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-31T18:14:59.101Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,24,68,129,0,0,1,2,10,246,5,1,0,76,0,208,168,63,0,0,4,8,27,4,0,10,246,29,223]
2021-07-31T18:14:59.103Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,24,68,129,0,0,1,2,10,246,5,1,0,76,0,208,168,63,0,0,4,8,27,4,0,10,246,29,223]
2021-07-31T18:14:59.104Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 24 - 2 - 4 - 129 - [0,0,1,2,10,246,5,1,0,76,0,208,168,63,0,0,4,8,27,4,0,10,246,29] - 223
2021-07-31T18:14:59.105Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":513,"srcaddr":62986,"srcendpoint":5,"dstendpoint":1,"wasbroadcast":0,"linkquality":76,"securityuse":0,"timestamp":4171984,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[8,27,4,0]}}
2021-07-31T18:14:59.108Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":27,"manufacturerCode":null,"commandIdentifier":4},"Payload":[{"status":0}],"Command":{"ID":4,"name":"writeRsp","parameters":[{"name":"status","type":32},{"name":"attrId","type":33,"conditions":[{"type":"statusNotEquals","value":0}]}]}},"address":62986,"endpoint":5,"linkquality":76,"groupID":0}'
2021-07-31T18:14:59.110Z zigbee-herdsman:controller:endpoint DefaultResponse 0x001e5e0902193948/5 513(4, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false})
2021-07-31T18:14:59.111Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2021-07-31T18:14:59.112Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001e5e0902193948:62986/5 (0,0,1)
2021-07-31T18:14:59.113Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":62986,"destendpoint":5,"srcendpoint":1,"clusterid":513,"transid":105,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,27,11,4,0]}}
2021-07-31T18:14:59.114Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,15,36,1,10,246,5,1,1,2,105,0,30,5,24,27,11,4,0,175]
2021-07-31T18:14:59.126Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2021-07-31T18:14:59.126Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2021-07-31T18:14:59.127Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2021-07-31T18:14:59.127Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
[... cut ...]
2021-07-31T18:16:05.584Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":62,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":1288,"dataType":33,"attrData":863}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":217,"endpoint":1,"linkquality":84,"groupID":0}'
```

steveuk23, 2021-08-01:
@robertalexa The receivers not actually connected to the boiler at the moment i didn't want to connect it until i had it working so it's just connected to a plug.
I've changed your code and i've not noticed any constant errors so far, But the same errors are still there when i refresh any of the refresh icons.
You're right the system_modes are actually working it's just taking time to sync to the linked thermostat.
Thanks for all the help with this i'm further on then when i was before ha ha
Could i ask how it detects the current temperature is that sent over from the linked Hive thermostat ? as i noticed all the Hive thermostats only expose the link quality.
I'm guessing it's not displaying the statues of the receiver exposes because of this error when refreshing ?

steveuk23, 2021-08-01:
i've just been reading [https://github.com/Koenkk/zigbee2mqtt/issues/7926](This) post where @Koenkk replies saying he's missing 2 convertors.
Going off that am i right in thinking i'm missing `fz.thermostat_att_report` and `fz.thermostat_weekly_schedule_rsp` ?

robertalexa, 2021-08-01:
Morning @steveuk23  :)

Glad to hear about system_mode

With regards to local temperature, the thermostat is the one that will measure it and compare it to the temperature that you set, the receiver (SLR1) will just "do the work", but yes, it might appear confusing for the receiver to report the temperature and the thermostat to only act as a remote. In short, the thermostat reports the local temp to the receiver and the receiver "declares/exposes" it basically.

Your URL text is correct but the actual link is broken - but i read the thread and yes I expect it to be something along those lines, thus me asking if for sure you have not removed any other code. Technically all the endpoints should be available.

Can you please replace this line `fromZigbee: [fz.thermostat_att_report, fz.thermostat_weekly_schedule_rsp],` with `fromZigbee: [fz.legacy.thermostat_att_report, fz.legacy.thermostat_weekly_schedule_rsp],`

That might be what is causing this. Give it a try and please let me know how it went.

Thanks,
Rob

steveuk23, 2021-08-01:
Sorry morning @robertalexa i wasn't sure which time zone you was in ha ha
I have just replaced that line of code but the Z2M dashboard fails to load back up until i replace it back.
I just want to to double check when you say i might have removed code, where exactly should i look to make sure ?

robertalexa, 2021-08-01:
@steveuk23 that makes absolutely no sense. When you replace that line do you get an error in the log?

If by you can't access the dashboard as in you get 501 Bad Gateway, remember you have to give it a minute for the addon to properly start.

Also make sure that you have copied and pasted correctly and not created any syntax errors.

As to code removed, that is very hard for me to advise, you are the only one that knows what files you edited :)

steveuk23, 2021-08-01:
@robertalexa 
Yeah it's the 502 Bad gateway.
i remembered what you said about waiting and have given it nearly 5 minutes but still nothing.
I'm getting this error in the logs:
```
> zigbee2mqtt@1.18.1 start /app
> node index.js
(node:3138) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'thermostat_att_report' of undefined
(Use `node --trace-warnings ...` to show where the warning was created)
(node:3138) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:3138) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
[13:43:23] INFO: Zigbee Herdsman debug logging enabled
[13:43:23] INFO: Handing over control to Zigbee2mqtt Core ...
> zigbee2mqtt@1.18.1 start /app
> node index.js
```

robertalexa, 2021-08-01:
Thanks for the logs.

Try changing that line again and on top of that also replace this line
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');`

with this
`const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};`

It's my bad for that. I forgot i didn't include the legacy in my original code.

I have tried this in my z2m instance and it boots up correctly.

Let me know how it goes.

steveuk23, 2021-08-01:
@robertalexa 
Yes... that seems to have done it :)
All the refreshes work and the system_modes etc and it all syncs to HA and to the thermostat
Only thing i can't see is the pi_heating_demand but i'm not sure what that's meant to show ?
Thanks so much for all you're help it really is appreciated.
If this was you would you leave the heating schedule to the one set in the thermostat or maybe create one with node red or a automation etc ?
On another note do you happen to know if i can change the default image of the device from the red Z,  i do remember seeing a option somewhere but can't find it now.
[image]

robertalexa, 2021-08-01:
Hi @steveuk23 

I am glad it is now working. So yeah, it basically operates like the SLR1b so it will be an easy fix.

My bad for the problem caused by the "legacy". I should have just used it in the code.

pi_heating_demand will technically show how much the valve should be open and it is decided based on the difference between local_temperature and occupied_heating_setpoint. In principle this should update itself once an hour, based on the reporting settings. I do not know however if this actually works or not. Can you please set your temperature to the highest, and leave it running for say an hour or 2 and see if pi_heating_demand gets updated? That will give us a definitive answer.

As mentioned to you before, i have the SLR1b but never found the time to try and get it set up in my z2m, but I plan to do it really soon, and also fix everything that is missing, cause there are some missing features in z2m.

When i get around it, will you please be willing to do some testing for me, as everything new that i will do to the slr1b i will do to slr1, so that people like you benefit from it too. I am also inclined to try to "blindly" improve the SLR2 and SLR2b and i think i can find a couple of people to test for me.

On this topic, could you please tell me something? What is the lowest and highest temperature that you can set on the SLR1 using the physical thermostat? Cause i bet it is different than what you can do in the Frontend.

With regards to the device image, I will add the image when i submit my PR to support your device (which i will do as soon as we both agree things behave as they should, so maybe even today.) Can you confirm if your receiver looks like this? https://www.zigbee2mqtt.io/devices/SLR1b.html

As to the heating schedule, i would personally not do it through the device itself. This is exactly what i have chosen to do for my TRVs. That is also because on the TRVs, if you take out the batteries and put them back in the schedule gets lost. I wouldn't be surprised if the same happens to the thermostat. Instead, I simply change the desired temperature using Node-Red flows. I think system_mode auto actually uses the internal scheduler. Off is obviously off and heat is manual control. So basically I would do Off when you want it off, and "heat" for everything else. Then using node red you decide if it is OFF at night, HEAT 18C when you are away and HEAT 22C when you are home :) But I would test what is the behaviour of OFF, i would expect it to set the temperature around 6 degrees, that is to have frost protection. If it doesn't do that, then i would just stick to HEAT instead and manually setting the temp to 9C

Sorry for the long post, hopefully this answers your questions. If you clarify the things above for me then i can go ahead and submit the PR and get your device officially supported.

robertalexa, 2021-08-01:
@steveuk23 Try replacing 3600 with 600 and reload the addon. That would change the reporting of heating demand from hourly to only 10 minutes, which would be better anyway. See if that value gets populated.

steveuk23, 2021-08-01:
@robertalexa 
Hopefully ill answer all the questions ha ha

- Yes it's the same as that image
- The temp dial goes to a maximum 32 degrees ( sorry can't find my degrees symbol on the laptop )
- the minimum temp dial goes down to 5 degrees and shows the frost symbol at 7 degrees 
- I've set it to max 32 now and will check what happens later.

Of course i'm more then happy to help you with testing.
I do have a working hive attached to my boiler and i think the receiver i have attached is the SLR2b as it's the dual channel but i don't need the dual channel only single and i saw this thermostat and receiver going for £6 for both on eBay and thought they would be great to work on and when i've got it working swap the dual channel to the single channel and go local but i didn't realise i had bought a older version receiver.
I am going away tomorrow on a Mon-Fri break so won't be able to help test things for you till afterwards
I've updated that code to 600 now but not sure if it's updated yet , i'm sure it would

robertalexa, 2021-08-01:
@steveuk23 
Thanks for coming back to me.

- i will sort out the image as part of my PR, so no need for you to do anything about it
- I indeed expected the temperature to be 5 to 32 rather than 7-30 like it is in z2m. I will update that to match the device. Don't know if you knew, but you can change the frost protect temperature from the thermostat. In my case I have set it to 9C. I am also using 9C as my minimum temp in all my automations because Google Home supports minimum 9, so i wanted things to match :)
- keep an eye on that and see if you get any value.

I know you are probably looking forward to swapping the 2 receivers, but would you be kind to help me test both SLR1 and SLR2b before you do so? This way i can hopefully improve all these devices in one swift movement, and it also means i do not need to find someone else to test it for me. Of course that is when you come back. Please? :)

All this work we did on this device made me really want to get on with my SLR1b, which i now have on my desk. If my research is correct, i should be able to make some changes to it while you are away, then we can look at the other receivers together.

Because of you going on holiday, the fact that the next z2m release is around 1st of Sept, and because I have finally made a start on my receiver, I think there is no rush to submit a PR to support your device just yet. I think it would be better if i can submit a better version of it in one go. Does that work with you?

steveuk23, 2021-08-01:
@robertalexa 
Yeah sure i don't mind testing both.
I'm presuming it's a SLR2b at this point unless i take it off, i'm just guessing at this point i know it's the latest dual channel.
Would i be able to test the SLR2b while its still attached to the boiler ? i'm guessing so lol
I would have tried setting up my working hive originally but i was worried i wouldn't be able to get it working back with Hive if i was unable to get it working locally and i couldn't understand how climates worked in HA and also find a flow for the heating schedule .
Also the wiring is slightly different between the single and dual channel hives and i always worry about blowing up my boiler even though i have the wiring diagram in front of me ha ha 
i started off by reading this but couldn't get his node red flow to load
https://github.com/roadsnail/Hive-SLR2-SLT2-Zigbee2MQTT-with-node-RED

robertalexa, 2021-08-01:
It is either an SLR2 or a SLR2b, god knows what the difference is.

Yes you should be able to test it while connected. None of the data goes to the boiler. The boiler only heat the water or not (in a simplified explanation). I do not know the extend of our testing yet. It depends on what i can get working :)

You can always put the devices back into pairing mode, get onto the Hive app, add device and voila.

I am guessing your boiler is a combi one? With the boiler boiling water for both heating and running water at once? Cause only that will work with SLR1 (basically single channel). I am no expert in wiring them so take my advice wiht a pinch of salt, but if your boiler is a combi then you probably have no wires connected to the water side of things on your SLR2b? Which basically means you transfer your existing wires onto the SLR1? Should be simple?

I am familiar with that link, but in my approach I am not using that. The reason for it is because i want my dashboard in HA, not in Node Red. Have a look here: https://www.reddit.com/r/homeassistant/comments/l1hzuv/recommendations_for_an_editable_schedule/

This is a person that wanted to move away from Hive too, but didn't fully do it. Feel free to have a read through it, the guy also provided me with code snippets to form an idea for his approach. But still ended up pretty much doing my own thing. I do not mind sharing some idea with you on this, but I am not sure this thread is the best place, cause this would be very off-topic. I did promise that guy that i will share with him my setup once it is ready (and i just need to do the main thermostat), so maybe we can talk about it in thhat Reddit thread?

steveuk23, 2021-08-01:
@robertalexa Thanks for the link that look good my reddit name is the same as here if you want to tag me to arrange testing through there.
I've just undone my working Hive and it is a SLR2b and it's just a case of swapping 1 wire as i have a combi boiler like you thought,
from random websites i've read i think the model number change was because of branding as they were moving away from using a company called "alert.me" 
I've actually just read your line on there about not requiring it to be "Granny Friendly" ha ha never heard that before Wife friendly yeah...

robertalexa, 2021-08-01:
Before you take things apart, take photos with your phone. You will be grateful to yourself later in case things don't go well. Write on a piece of paper what colour cable you took out of what hole. Then it should be easy to swap over.

Thanks for the information about versioning :)

Yeah, Wife Friendly in my household means she can see a pretty dashboard, press a button to turn the heating on. The rest of the magic happens with my automations, wife doesn't understand nor care probably. And scheduling would also go above her head. Heck it gave me a really hard time and i work as a developer so... yeah... granny friendly :)

PS: i am currently reviewing all the features, all the possible controls etc and I am trying to filter out the things that should not be exposed, or require a more complex process that will not work through a simple expose. So those things are out of the equation. I would however want to document them, so users can learn what commands they need to run for example to boost their heating "natively" etc. We shall see how it goes. Heck i mind realise that not a lot of these can or should be exposed. As they will make big boo boos if you use them wrong.

PPS: I have create a PR for that image already, so it will be ready when you need it

steveuk23, 2021-08-01:
@robertalexa 
Thanks yeah I just took pictures then when I took it off it looks a lot simpler then I thought just moving 1 wire across.
With you saying that I probably don't really use the schedule as much as I think as my wife and son are constantly manually changing it.
That also reminded me of another worry I had about how I would recreate the boost function 
On my Lovelace at the moment I have 3 buttons for 30,45 and 60 mins boost that run a script that's linked to the boost sensor thing that provides a timer...I would have no idea how to create that.

robertalexa, 2021-08-01:
Patience young grass hopper :) All in due time. With good documentation you will understand how to do it. And I will be sorting that out, and you will also participate in testing 2 different devices, so you will get a chance to practice.

Did you get an update on the pi_heating_demand yet?

robertalexa, 2021-08-01:
Oh by the way, if you are worried about testing while wired, you can first test on the SLR1, once we are done you get that fitted, we test on the SLR2b on your desk. So realistically nothing horribly wrong can happen.

robertalexa, 2021-08-01:
@steveuk23 You wanna hear something funny? My receiver is not a SLR1b, it's actually and SLR1, just like yours. 

I just tried pairing it and taaa daaa surprise surprise.

So i will be doing all the work and testing on this. Go enjoy your holiday :)

steveuk23, 2021-08-01:
@robertalexa 
No it's still reporting N/A I think it's been over a hour now can't remember exactly when I set it lol
Hopefully I'll learn some more from doing this.
Yeah I might do that what you said about swapping them over to work on the desk.
I'm getting further along each day with most things, them 3 buttons I mentioned turn red when the boost is active but because all 3 buttons are linked to one sensor they all turn red no matter which is pressed.
Anyway that's a different topic :)

steveuk23, 2021-08-01:
> @steveuk23 You wanna hear something funny? My receiver is not a SLR1b, it's actually and SLR1, just like yours.
> 
> I just tried pairing it and taaa daaa surprise surprise.
> 
> So i will be doing all the work and testing on this. Go enjoy your holiday :)

Ha ha that's a coincidence least you've done all the leg work already.

robertalexa, 2021-08-01:
> Ha ha that's a coincidence least you've done all the leg work already.

Yes but it could also have been sooo much faster without the ping pong. Well, it happens

steveuk23, 2021-08-01:
@robertalexa well yeah suppose.
Have you had your hive a few years then ?
my wife thinks i'm daft even trying saying whats the point it works and i tell her...... but no it goes down no and again and the local part etc

robertalexa, 2021-08-01:
I have bought the house last year and it had it already.

1st - she doesn't need to understand, you probably know better. See my Wife Friendly point above.
And then now all the legit reasons
- I have both Thermostat and TRVs on all the radiators. I bought the TRVs, spent quite some money.
- Heating on Demand via Hive is a POS. The boiler is pretty much on because if the temp of a room goes 0.1 degrees lower, it asks the boiler for heat. That is stuuuupid logic. 0.1 of a degree is nothing
- You want to see graphs of your temps and stuff? You have to pay. I already paid for the devices, no thank you
- The Hive integration with HA keep crapping the bed. Not because of the integration, but because Hive doesn not have a public API, like any decent company. So the community hacked it. But it can break anyday, if they change it.
- Hive does not publicise their specs, hence the crappy or missing features from devices - that is stupid corporate greed. If your device can work not only on your platform, but on open source ones too, then people will buy your product to do whatever tf they want, and you still make money.
- If everything works well, and I have control over my heating, i save money on bills. That is probably the most important one. With that money i can do sth else.

Does that help to convince your wife? :)

steveuk23, 2021-08-01:
@robertalexa 
Some great points :)
TRV's are probably my next purchase i've been looking through different brands already but must say have read some really bad feedback on the HIve ones.

robertalexa, 2021-08-01:
Now that i moved them away from Hive and into Z2M, and added a truckload of support for them, I am super pleased with them. I am not trying to say buy these ones in particular, you can probably get TuYa ones for less money. But i bought them before i started my own HA and yeah, i ended up supporting them properly. They are not even Hive products, they are actually Danfoss Ally trvs, and Danfoss is one of the best "heating" companies out there, so the actual device is pretty solid, just Hive is bad :))

steveuk23, 2021-08-01:
@robertalexa 
i'll have another look at then then thanks.

peterguy04, 2021-08-02:
@steveuk23 
> Yeah sure i don't mind testing both.
> I'm presuming it's a SLR2b at this point unless i take it off, i'm just guessing at this point i know it's the latest dual channel.

hey, the difference visually between SLR2 and SLR2b is text vs icons, so on SLR2 its text (hot water / central heating ) and SLR2b is icons (a tap and a flame )  i dont know what changes where actually made but thats how you can tell them apart from my experience 

also @robertalexa i thought id message ya here seen as i am here, i have a SLT2

robertalexa, 2021-08-02:
@peterguy04 Really appreciate you offering to help.

Allow me to put together some code for you to test with, i am currently trying to make sure it works before i give it to you.

1. TRVs - i have already added the most support I was able to a few months ago. The TRVs now expose a lot more attributes, and more importantly they change temperature instantly skipping the PID

2. SLR2 - can you confirm that system_mode does not update when you change it from the physical thermostat? As in in order to get an update you need to refresh it manually? If so, i can make a fix and have that report the change and i can also guide you so you can test it fairly quickly. I will also change the temperature based on the info you gave me.

3, SLT2 - give me a few minutes and i will reply with some instructions

robertalexa, 2021-08-02:
@peterguy04 

Regarding my 2nd point, if my above thinking is correct, using the Frontend, can you please go on the receiver, click on Reporting, and set a new reporting using the following values?

Endpoint 5 is for heating
[image]

Endpoint 6 is for water
You can try to set the same thing but on that endpoint to honestly i have no idea how the 2 channels receivers work.

You will need to receive a successful binding message before you can attempt to change system_mode via the physical thermostat, and then check if the mode changed in z2m

peterguy04, 2021-08-02:
so just a bit of info, when we first pair these to z2m and try set them to a temp it jumps back to 0, this is fixed by physically setting the Tstat to manual mode . it only needs to be done once , then we can change modes etc from there and temp values stick

SLR2 - your right you have to manually hit refresh for the change to happen, this is true for both heat/manual and off. but auto is non functional 

adding the reporting as you suggested to EP5 worked 
EP6 we have 4 modes on thermostat ON , OFF, Boost and Auto
setting ON and auto both turned to HEAT in z2m, off worked as expected but Boost made all become unselected.

the water endpoint has a fixed temp reading of 21c and a fixed "setheatpoint" as 22c. so this relay really is just a on/off switch

robertalexa, 2021-08-02:
So just to make sure i understand correctly, you have set reporting on both endpoint 5 and enpoint 6 and you have seen improvements right?

peterguy04, 2021-08-02:
reporting works on both endpoints yes after manually adding it like you suggested. before that it did nothing. but i havent been able to properly use the hive stuff on z2m because it wont work with my home assistant properly . cant control it. i paired it back up to help you here 

[image]

robertalexa, 2021-08-02:
Auto means Scheduler, so the internal scheduler of the thermostat. There is kinda some support for it, but i feel that there are some bits missing. And also this would be done via commands only, not via interface. This would also make it impossible to expose to HA. But equally, people that plan on using a scheduler will do their own thing, and not rely on the internal one. See my comments above when talking to Steve about scheduling.

With regards to boost, yes, there is a 4th value called emergency_heating. But this is not supported by z2m. I can create support for it, but it might not be a good idea. The reason for it is cause it requires more than 1 call to action it, you need a temperature and a duration. When i do my PR i will discuss this with Koen and see what he suggests. Nonetheless, i plan on adding it to the documentation when i am done.

Once you put the receiver in emergency_heating (boost on the Thermostat), you have to send a new command to stop the boost. So that explains why you see them as unresponsive.

I have not tried to test my receiver in HA, i will handle that part of it once i finish changing these devices. Maybe I can help you with your device too.

I will send another message soon to test the battery reporting, though from what i can gather i am 99% sure it will work without testing.

peterguy04, 2021-08-02:
@robertalexa sure, ye i do my boost and scheduling between HA and nodered anyway, aswell as my own automation for "heat on demand" for the TRVs as hives was very in-efficient 

i helped create support in deconz and sniffed the zigbee packets . so theres some info here on the different modes that i got if its helpful for anything. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2978

robertalexa, 2021-08-02:
> @robertalexa sure, ye i do my boost and scheduling between HA and nodered anyway, aswell as my own automation for "heat on demand" for the TRVs as hives was very in-efficient

THANK YOU!!! This is exactly how i ended up supporting these devices, cause i want to do my own thing precisely because of heat on demand! Maybe you can share with me your flow logic for heat on demand in node red? or even an export if it makes it easier?

Here is what to do to test the battery reporting:
1. stop the addon
2. edit your database.db file and change the model of your remote from SLT2 to SLT2A (this way it becomes unsupported)
in my case with an SLT3 it looks like this
[image]

3. create a file called SLT2A.js in the same place as your configuration.yaml
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const e = exposes.presets;

const definition = {
    zigbeeModel: ['SLT2A'],
    model: 'SLT2A',
    vendor: 'Hive',
    description: 'Heating thermostat remote control',
    fromZigbee: [fz.battery],
    toZigbee: [],
    exposes: [e.battery()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(9);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
        await reporting.batteryPercentageRemaining(endpoint, {min: 60, max: 120, change: 1});
    },
};

module.exports = definition;
```
4. edit your configuration.yaml file and add
```
external_converters:
  - SLT2A.js
```
5. restart the addon and check the frontend. it should look like this
[image]

Thanks a lot!

peterguy04, 2021-08-02:
so for some reason am having a hard time adding the line to the yaml, its either giving me invalid yaml OR when i restart the addon its deleting the line.... this is using the HA addon btw 

also i notice EP6 has no function on SLR2 i already said that it has fixed values so acts more like a on/off switch. but changing from heat to off, or off to heat does nothing

robertalexa, 2021-08-02:
Hi @peterguy04 

That error is probably because you already have the external_converters: [] line in your configuration. Replace that with the one i sent you :)

I kinda expected exactly that about the EP6, but because i do not have it i find it impossible to understand how it operates on its own, let apart what should z2m have. So how would you go about turning water on and off from z2m on the water endpoint?

peterguy04, 2021-08-03:
sorry for the delay, was having abit of a drama between my VM and my coordinator so the whole thing came crashing down.
anyway ive fired it up in a docker container now rather than HA addon so its easier for me to work with. i have a few things to be on with today but ill try that battery reporting on the SLT when i get chance

so heres the image of the SLR2 endpoint, so focusing on water both the "occupied_heating_setpoint" and the "local_temperature" are hardcoded to the device at 21c and 22c . these are not settable by the user. it turns it into a ON/OFF switch based on the mode OFF/HEAT . so the state is reported from SLT in Z2M correctly but when we send the command from Z2M it does nothing. 

i know when i sniffed the zigbee packets they needed multiple commands at once , but i dont know how this translates to z2m but should be similar to EP5 i would of thought 
```for "on - manual"

attribute 0x001C > 0x04 = "heat"
attribute 0x0023 > 0x01 = "Setpoint Hold On"

For "off"

attribute 0x001C > 0x00 = "off"
attribute 0x0023 > 0x00 = "Setpoint Hold Off"```
![image](https://user-images.githubusercontent.com/45408070/127998299-d2122f6d-3a40-4c1b-a6cd-8ae2e6c72699.png)

robertalexa, 2021-08-03:
@peterguy04 

Thank you for the above information. I think i kinda understand.

You are saying that nothing is happening when you change system_mode OFF and HEAT. Does that mean the receiver doesn't do anything? Does the remote show that you did that action? What happens if you press the blue refresh button (for occupied, local, system and running state) after running this action (reporting currently is a bit wrong, so that could be why it wouldnt update without manually refreshing - but i have already changed this in my improvements)?

Now onto "improvements":
- I am correct in thinking that AUTO should be removed from the water endpoint?
- I am correct in thinking that the temperature slider should be lock to a range between 21 and 22 rather than the current 7 to 30?
- I am correct in thinking that pi_heating_demand is never showing a value for either of the endpoints and should be removed?

Because this has to act as a climate device, and it should link correctly in HA, as much as it is counter intuitive, we can't remove the occupied heating setpoint and local temperature, cause they all contribute to the climate entity. But i we restrict it to only correct values than that would make it work better generally.

Let me know about the above and I can make the relevant changes. 
Thanks

robertalexa, 2021-08-03:
Can you also confirm if for the heating endpoint you can set the temperature on the physical thermostat in 0.5 degrees increments rather than 1 degree as z2m does? If that is the case, i can fix z2m steps

peterguy04, 2021-08-03:
> Can you also confirm if for the heating endpoint you can set the temperature on the physical thermostat in 0.5 degrees increments rather than 1 degree as z2m does? If that is the case, i can fix z2m steps

yes it goes up in .5 increments on the SLT , we can set that by manually typing the value but not on slider in z2m and setting a value with .5 does update to the SLT
[image]

[189 more replies]

closed 2021-08-08 by steveuk23