Koenkk/zigbee2mqtt #8190: YRD226HA2619 PIN code support

zeiphon, 2021-07-31:
I noticed that the Yale lock YRD226/246 TSDB currently has support for PIN code set/get/clear as well as user status:
`        fromZigbee: [fz.lock, fz.battery, fz.lock_operation_event, fz.lock_programming_event, fz.lock_pin_code_response,
            fz.lock_user_status_response],
        toZigbee: [tz.lock, tz.pincode_lock, tz.lock_userstatus],`

I have a Yale YRD226HA2619 which is a very similar lock using the same zigbee module I believe, and which also supports these functions/clusters. Would it be possible to implement support for this in zigbee2mqtt by mostly copying that implementation across to the YRD226HA2619 model block?

Koenkk, 2021-07-31:
Added (for all Yale locks).

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-07-31 by Koenkk

zeiphon, 2021-07-31:
Brilliant! Thank you

keito, 2021-10-10:
@Koenkk, I think the commit above may have introduced a bug into the battery percentage calculation for some of the Yale models; before the change, only some of the models set `dontDividePercentage` to true, but `lockExtend` sets it to true for all models. For instance, I have YRD226/246 TSDB and now it is reporting > 100% battery percentage.

Koenkk, 2021-10-10:
@keito you are right, fixed this now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

keito, 2021-10-10:
Thanks for the quick fix, @Koenkk!

pedroxns, 2021-11-20:
Hello, I have the same lock and I'm trying to understand how to manage the codes.
I can create/edit codes without any issues but I need to create one code with weekday schedule, and can't figure out how to program this, any helps here?

homeseer666, 2022-01-22:
I am about ot purchase a Yale YRD216-HA2-619 as I see that Pin code support as been added to Zigbee2MQTT;
Which was one of my requirements.
Documentation is clear in regards to what should be published and available for control via MQTT.

However something is not clear to me: If the door is unlocked via the keypad or the physical key, is the associated data published ? (lock/unlocked status, pin code used, etc.)

Thank you!

pedroxns, 2022-01-22:
YRD226 publishes codes about the way the lock was locked/unlocked and, in case the unlocking method was password, the user id that did the operation.

I can't figured out how to proper setup the week_day_schedule user option.

Can't remember if there is any publishing when the wrong password is entered.

homeseer666, 2022-01-22:
Thank you, sounds good; 

havent seen documentation on the week_day_schedule user option either, would be good to have it.

 maybe you can ask/bump your existing question on the forum ?