Koenkk/zigbee2mqtt #8200: Xiaomi DJT11LM strength is sent as "string" instead of numeric.

MiguelAngelLV, 2021-08-01:
## What happened

The strength signal is sent as string to Home Assistant, so, is very hard read it

[image]

## What did you expect to happen

The values may sent as numeric to and home assistant show them in time series chart.

## How to reproduce it (minimal and precise)

Use DTJL11LM with Home Assistant, move it and show the values.

## Debug info
Zigbee2MQTT version: 1.20
Adapter hardware: CC2531
Adapter firmware version:  20201127

Koenkk, 2021-08-01:
Can you provide the zigbee2mqtt debug log when a new `strength` is published?

See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging.

MiguelAngelLV, 2021-08-03:
```
Debug Received Zigbee message from 'Puerta garaje', type 'attributeReport', cluster 'genBasic', data '{"65282":[{"elmType":16,"elmVal":0},{"elmType":33,"elmVal":3045},{"elmType":33,"elmVal":5032},{"elmType":36,"elmVal":[0,3]},{"elmType":33,"elmVal":252},{"elmType":32,"elmVal":85}]}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Puerta garaje', payload '{"battery":100,"contact":true,"linkquality":44,"voltage":3045}'
Debug Received Zigbee message from 'Movimiento sótano', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Movimiento sótano', payload '{"battery":100,"linkquality":42,"occupancy":true,"voltage":3035}'
Debug Received Zigbee message from 'Movimiento despensa', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Movimiento despensa', payload '{"battery":100,"linkquality":42,"occupancy":true,"voltage":3045}'
Debug Received Zigbee message from 'Movimiento entrada', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Movimiento entrada', payload '{"battery":100,"linkquality":47,"occupancy":true,"voltage":3065}'
Debug Received Zigbee message from 'Lavadora', type 'attributeReport', cluster 'closuresDoorLock', data '{"85":1}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Lavadora', payload '{"action":"vibration","angle":9,"angle_x":-3,"angle_x_absolute":93,"angle_y":78,"angle_y_absolute":12,"angle_z":12,"battery":86,"linkquality":31,"sensitivity":"high","strength":8,"voltage":2975}'
Info MQTT publish: topic 'zigbee2mqtt/Lavadora', payload '{"action":"","angle":9,"angle_x":-3,"angle_x_absolute":93,"angle_y":78,"angle_y_absolute":12,"angle_z":12,"battery":86,"linkquality":31,"sensitivity":"high","strength":8,"voltage":2975}'
Info MQTT publish: topic 'zigbee2mqtt/Lavadora/action', payload 'vibration'
Debug Received Zigbee message from 'Estufa Baño', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/Estufa Baño', payload '{"linkquality":31,"state":"OFF"}'

```
In the payload is show as number, but Home Assistant show it as string...

I removed all states of devices from Home Assistant database, but it did not resolve the problem.

Koenkk, 2021-08-03:
@emontnemery do you know why this happens? The value is send as a number but displayed as a string in HA.

MiguelAngelLV, 2021-08-03:
I checked and in the mqtt payload is sent as numeric too.

[image]

So, the problem is HA... Maybe HA require explicit numeric definition? Any attribute (state class, measurement...) ?

The angles are numeric and they are show as string...

[image]

Battery and link quality they are show as numeric in a chart.

closed 2021-09-11 by github-actions[bot]