Koenkk/zigbee2mqtt #8201: Energy marked as Unknown in HA since #8187

ALaDoffe, 2021-08-01:
## What happened

Hello 
Since PR #8187 all my energy sensors show "unknown" in HA.
Maybe I missed something in configuration.
HA Version : 2021.7.4

## What did you expect to happen

Work as before

## Debug info
Zigbee2MQTT version: latest dev 
Adapter hardware: CC2652P
Adapter firmware version: latest

frenck, 2021-08-02:
Anything in the logs?

Mr-Groch, 2021-08-02:
I can confirm that. HA Version : 2021.7.4 and Z2M 1.21.0 commit: unknown.

[image]

I've tried restarting Z2M and HA. No change...

In Z2M reported energy looks OK:

[image]

frenck, 2021-08-02:
Hmm interesting, the properties `last_reset` and `state_class` properties set in #8187 appear in Home Assistant as expected, the `energy` value is there. 

I wonder why Home Assistant qualifies it as Unknown.

Can you dump the discovery and state topic for this device from MQTT?

Mr-Groch, 2021-08-02:
Yes I can, discovery topic `homeassistant/sensor/0x842e14fffe186b71/energy/config`:

```json
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state"
    }
  ],
  "device": {
    "identifiers": [
      "zigbee2mqtt_0x842e14fffe186b71"
    ],
    "manufacturer": "TuYa",
    "model": "10A UK or 16A EU smart plug (TS0121_plug)",
    "name": "Gniazdko z miernikiem - ogród",
    "sw_version": "Zigbee2MQTT 1.21.0"
  },
  "device_class": "energy",
  "json_attributes_topic": "zigbee2mqtt/Gniazdko z miernikiem - ogród",
  "last_reset_topic": "zigbee2mqtt/Gniazdko z miernikiem - ogród",
  "last_reset_value_template": "1970-01-01T00:00:00+00:00",
  "name": "Gniazdko z miernikiem - ogród energy",
  "state_class": "measurement",
  "state_topic": "zigbee2mqtt/Gniazdko z miernikiem - ogród",
  "unique_id": "0x842e14fffe186b71_energy_zigbee2mqtt",
  "unit_of_measurement": "kWh",
  "value_template": "{{ value_json.energy }}"
}
```

state topic `zigbee2mqtt/Gniazdko z miernikiem - ogród`:

`{"current":0.3,"energy":426.62,"last_seen":"2021-08-02T11:51:12+02:00","linkquality":36,"power":36,"power_outage_memory":"restore","state":"ON","voltage":234}`

frenck, 2021-08-02:
Thanks! I will ensure to analyze this further today 👍

ALaDoffe, 2021-08-02:
Same as @Mr-Groch 

Value OK in frontend 
[image]

OK Also in the log 

```
debug 2021-08-02 19:35:36: Received Zigbee message from '0x00158d0006327572', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":0,"101":0,"110":0,"111":0,"148":3,"149":6.333125114440918,"150":2309.987060546875,"151":0.0035429000854492188,"152":0.030308961868286133,"155":0,"3":38,"5":162,"7":"0x0000000000000000","8":4644,"9":1024}}' from endpoint 1 with groupID 0
info  2021-08-02 19:35:36: MQTT publish: topic 'zigbee2mqtt/0x00158d0006327572', payload '{"consumption":6.33,"energy":6.33,"interlock":false,"linkquality":12,"power":0.03,"state_l1":"OFF","state_l2":"OFF","temperature":38,"voltage":231}'
```

[image]

mcfly2283-ai, 2021-08-03:
i have the same issue with two Blitzwolf BW-SHP13. The Energy-Entity has the value "unknow" after updateing.

frenck, 2021-08-03:
So I have tested this with the data provided by @Mr-Groch:

[image]

I wonder if this might be caused by a HA version, which is what I'll research next.

frenck, 2021-08-03:
Found the issue!

This is not a Zigbee2MQTT issue, but a bug in Home Assistant.
The good news is, it has been fixed: <https://github.com/home-assistant/core/pull/53677>

This fix is expected to land in the Home Assistant release of tomorrow (Wednesday 4 August 2021).

Koenkk, 2021-08-03:
Thanks @frenck! Since it's not a z2m issue I will close this.

closed 2021-08-03 by Koenkk

amy92000, 2021-08-18:
I found the solution you have to go to Z2M selection not taken and you go to the report you will have to add commands as in the photo and then you work.
[image]

amy92000, 2021-08-18:
you just have to add up to line in z2m reports and you would have the value Energy
[image]