Koenkk/zigbee2mqtt #8205: User Extension cannot request state changes over MQTT (freezes device on Z2M side)

froogl, 2021-08-01:
## What happened

I am trying to implement some basic rules for one of my ZGP devices within zigbee2mqtt. I am doing that with a custom user extension within `onPublishEntityState`:

```javascript
const stringify = require('json-stable-stringify-without-jsonify');

class MyExampleExtension {
    constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
        this.zigbee = zigbee;
        this.mqtt = mqtt;
        this.state = state;
        this.publishEntityState = publishEntityState;
        this.eventBus = eventBus;
        this.settings = settings;
        this.logger = logger;

        logger.info('Loaded  MyExampleExtension');
        this.eventBus.on('publishEntityState', (data) => this.onPublishEntityState(data));
    }

    async onPublishEntityState(data) {         
         if (data.entity.name == "sw_test_2") {
             if (data.payload["action"] == "recall_scene_0") {
                 logger.info("TEST: publish toggle message");
                 await this.mqtt.publish('light_test_1/set', stringify({"state": "TOGGLE"}));
             }            
         }
    }

    stop() {
        this.eventBus.removeListenersExtension(this.constructor.name);
        
    }
}

module.exports = MyExampleExtension;
```

Using `this.mqtt.publish` on a light (or probably any device) introduces multiple issues:

1. It does not work. The message is published, but Zigbee2Mqtt does not react to it.
2. The worst part is that the light (`light_test_1` in my example) cannot be controlled through Z2M any more until node.js is restarted. This is actually a pretty big issue.

Note: In the long-run I'd like to create a more generic rule engine with the approach from above.

## What did you expect to happen

The light should react to the MQTT message from the user extension.

## How to reproduce it (minimal and precise)

Place the code from above into a user extension and adjust events/lights.

## Debug info
Zigbee2MQTT version: 1.20.0
Adapter hardware: not applicable
Adapter firmware version: not applicable

Koenkk, 2021-08-01:
Instead of `await this.mqtt.publish('light_test_1/set', stringify({"state": "TOGGLE"}));` you can do 
 `this.mqtt.onMessage('zigbee2mqtt/light_test_1/set', stringify({"state": "TOGGLE"}));`, this will also prevent the message from published from MQTT (so less delay)

froogl, 2021-08-02:
@Koenkk Cool, this is working. Thanks. :+1: 

However this still does not explain why calling `this.mqtt.publish` from within that context blocks up all further communication to that topic. This is not relevant for me personally, as you have shown me an even better approach. However I am keeping this issue open so you can decide whether you are considerung this as a bug or not. So you can close if you want. ;)

Koenkk, 2021-08-02:
@froogl this is expected, if you set `skipReceive=false` this behaviour will be disabled: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/mqtt.js#L111

closed 2021-08-02 by Koenkk