Koenkk/zigbee2mqtt #8208: "tsc not found" in v1.21.0 start

11csive, 2021-08-01:
## What happened
npm start fails with "tsc not found"

## What did you expect to happen
zigbee2mqtt to start without errors

## How to reproduce it (minimal and precise)
Follow https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html until the "npm start" step.

## Debug info
Zigbee2MQTT version: 1.21.0

Having had a quick look at the new version, I assume this is caused by Typescript being set as a devDependency, and the documentation instructs users to use the --production argument, which causes the devDependencies to not be installed.

Running npm ci without the --production argument makes it start as normal. :)

11csive, 2021-08-01:
Nevermind, I just noticed that the documentation was updated to include this. I guess we are waiting for the cdn to update the website.. :)

Koenkk, 2021-08-02:
Fixed with https://github.com/Koenkk/zigbee2mqtt.io/pull/844

closed 2021-08-02 by Koenkk