Koenkk/zigbee2mqtt #8212: Tuya door sensor TS0203 set is unsupported in 1.21.0

enigmaspb, 2021-08-02:
## What happened
Then update from 1.20.0 to 1.21.0 all my door sensor is unsupported 
Door sensor in 1.20.0 detected is Tuya TS0203  ( https://aliexpress.ru/item/1005002198797695.html ) 

## How to reproduce it (minimal and precise)
Install 1.20.0 , paring , update to 1.21.0

## Debug info
Zigbee2MQTT version: 1.21.0 (commit #70891eec) zigbee-herdsman (0.13.124)
Adapter hardware: Slaesh's CC2652RB stick
Adapter firmware version: 
first test old firmware "maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200715,"transportrev":2},"type":"zStack3x0"
second test update firmware  'maintrel': 1, 'majorrel': 2, 'minorrel': 7, 'product': 1, 'revision': 20210708, 'transportrev': 2}, 'type': 'zStack3x0'

warn  2021-08-02 07:52:52: Received message from unsupported device with Zigbee model 'TS0203' and manufacturer name '_TZ3000_psqjayrd'
warn  2021-08-02 07:52:52: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
info  2021-08-02 07:52:52: MQTT publish: topic 'zigbee2mqtt/door_sensors_1', payload '{"battery_low":false,"contact":true,"last_seen":"2021-08-02T04:52:52.833Z","linkquality":78,"tamper":false}'
info  2021-08-02 07:52:52: MQTT publish: topic 'zigbee2mqtt/door_sensors_1/contact', payload 'true'
info  2021-08-02 07:52:52: MQTT publish: topic 'zigbee2mqtt/door_sensors_1/tamper', payload 'false'
info  2021-08-02 07:52:52: MQTT publish: topic 'zigbee2mqtt/door_sensors_1/battery_low', payload 'false'
info  2021-08-02 07:52:52: MQTT publish: topic 'zigbee2mqtt/door_sensors_1/last_seen', payload '2021-08-02T04:52:52.833Z'
info  2021-08-02 07:52:52: MQTT publish: topic 'zigbee2mqtt/door_sensors_1/linkquality', payload '78'

vjiggi, 2021-08-02:
same experience.  CR Smart Home Door sensor (TS0203) not recognized after upgrading from 1.20.0 to 1.21.0

adrianofox, 2021-08-02:
Ditto. CR Smart Home Door sensor (TS0203) not recognized after upgrading from 1.20.0 to 1.21.0.

The sensor disappears from recognized devices on HA even though the status tab data on z2m is complete.

I returned to previous z2m, so ok.

Koenkk, 2021-08-02:
Added back `_TZ3000_psqjayrd`.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-08-02 by Koenkk

enigmaspb, 2021-09-02:
In new last dev branch and 1.21.1 is now unsupported  !